espressif/esp-idf #11040: Reading ADC data during deep-sleep from ULP-RISCV

Walid-Adel, 2023-03-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm using ESP32-S2
I'm trying to read ADC data from ULP-RISCV :

-    i created an infinite loop in the ULP main function as following:

```
int main(void)
{
	for(;;)
	{
		if(once)
		{
			raw_adc_data = ulp_riscv_adc_read_channel(ADC_UNIT_1, ADC_CHANNEL_1);
			once = 0;
		}
		if(is_sleep)
		{
		    if(cycle_count == 1000)
		    {
		    	server_time++;
		    	cycle_count = 0;
                        raw_adc_data2 = ulp_riscv_adc_read_channel(ADC_UNIT_1, ADC_CHANNEL_1);
		        if((raw_adc_data2 < 150))
		        {
		        	ulp_riscv_wakeup_main_processor();
		        	is_sleep = 0;
		        }
		    }
		   cycle_count++;
		}
		ulp_riscv_delay_cycles(ULP_RISCV_CYCLES_PER_MS);
	}

	return 0;
}
```
-----> in this method i was able to read adc data from ulp but in active state (not in deep sleep) and it was reading correct values. but when going to deep sleep it always read 8191.
by the way, if i want to read adc data during active state (not deep sleep) then i just set the variable `once` to 1 and ulp will execute its if block. And also if i want to execute the ulp during deep-sleep then i just set the variable `is_sleep` to 1 and ulp will execute its if block.

In short, if i want to read adc data during active state then:
1. Inside the main CPU, i do the following: `ulp_once = 1`
2. Inside the ULP, it will execute the following if block directly:
```
if(once)
{
	raw_adc_data = ulp_riscv_adc_read_channel(ADC_UNIT_1, ADC_CHANNEL_1);
	once = 0;
}
```

And if i want to read adc data during deep-sleep then:
1. Inside the main CPU, i do the following: `ulp_is_sleep = 1` and then call enter the deep-sleep mode
2. Inside the ULP, it will execute the following if block directly:
	```
	if(is_sleep)
		{
		    if(cycle_count == 1000)
		    {
		    	server_time++;
		    	cycle_count = 0;
                        raw_adc_data2 = ulp_riscv_adc_read_channel(ADC_UNIT_1, ADC_CHANNEL_1);
		        if((raw_adc_data2 < 150))
		        {
		        	ulp_riscv_wakeup_main_processor();
		        	is_sleep = 0;
		        }
		    }
		   cycle_count++;
		}
		ulp_riscv_delay_cycles(ULP_RISCV_CYCLES_PER_MS);
	}
```
```
===> but again it was reading fine during active state (when executing the if block of variable `once`). But on the other hand, it always reading 8191 during deep-sleep (when executing the if block of variable `is_sleep`)
so, how can i solve it so that it reads correct adc data during deep sleep ?
```

ESP-Marius, 2023-03-27:
Hi,

There is an in-progress issue internally for a known problem with S2 ULP ADC in deep sleep related to some power configurations (it gives the wrong results when reading). I'll remind them that we still have users asking for support with this...

Walid-Adel, 2023-03-31:
Thanks for your response
So, is there any way that i can read from the ADC data from ULP without having wrong results ?

Walid-Adel, 2023-04-07:
@ESP-Marius  I tried the code with ESP32S3 chip and now it's working fine and it's more stable than S2 board.
it seems like the ULP is working better with S3 board

ESP-Marius, 2023-04-10:
@Walid-Adel Yes, S3 supports using the ADC from the ULP RISCV

closed 2023-04-18 by espressif-bot

mickeyl, 2024-01-02:
@ESP-Marius I reckon this is raw voltage. Is there any way to tap into the calibration curves when the main cores are sleeping or any other way to take the calibration into account?

ESP-Marius, 2024-01-05:
Currently there is no convenient way of accessing these calibration curves (they are stored in internal RAM, which gets powered down). But I think it should be possible to do some updates to the adc calibration API to allow it be allocated to RTC RAM, as well as make the convert API callable from the ULP. 

I'll make an internal ticket for this.

mickeyl, 2025-02-17:
@ESP-Marius Yearly reminder for this feature ;)