espressif/esp-idf #11043: During the OTA process,Task watchdog got triggered.

casitelwithdean, 2023-03-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi, my chip is esp32s3wroomn8r8 .I am using OTA in my project.And maybe because my bin size is a little big(1700KB), when OTA begins,Task watchdog got triggered.However after several task_watchdog information displaying on the cmd,it will successfully finish OTA.
So i wanna know if the task_watchdog has some bad effects  on OTA process?
if so,is there a way to sovle this  problem?
Thanks a lot.

hmalpani, 2023-04-05:
Hello @casitelwithdean 
The default behavior of task watchdog timeout is to print a warning and backtrace before continuing running the app. Unless `CONFIG_ESP_TASK_WDT_PANIC` is selected in the menuconfig, the application will run normally. You can refer to [Task Watchdog Timer](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/wdts.html#task-watchdog-timer-twdt) documentation for more information. I would still recommend you solve this issue. Can you try [simple_ota_example](https://github.com/espressif/esp-idf/tree/master/examples/system/ota/simple_ota_example) to update with your binary and check if you get the same warnings in that example as well?

AxelLin, 2023-04-07:
@hmalpani 
The point is WDT should not be triggered during OTA.
I have reported similar issue at https://github.com/espressif/esp-idf/issues/7335 since **2021**, but this seems never get fix.

hmalpani, 2023-04-10:
Hello @casitelwithdean 
I tried the simple_ota example present in esp-idf to download a binary image of size ~1750 KB. I didn't get any task watchdog warning. Can you please try the simple_ota_example to update with your binary and see if the watchdog gets triggered?

casitelwithdean, 2023-04-10:
> Hello @casitelwithdean I tried the simple_ota example present in esp-idf to download a binary image of size ~1775 KB. I didn't get any task watchdog warning. Can you please try the simple_ota_example to update with your binary and see if the watchdog gets triggered?

Hi, sorry for replying so late.I will test it immediately!

casitelwithdean, 2023-04-10:
> Hello @casitelwithdean I tried the simple_ota example present in esp-idf to download a binary image of size ~1775 KB. I didn't get any task watchdog warning. Can you please try the simple_ota_example to update with your binary and see if the watchdog gets triggered?

@hmalpani   Hi,I tested it just now,there is no  task watchdog triggered in simple_ota_example . So I guess maybe because  I ran a lot of tasks in my project..

hmalpani, 2023-04-14:
Closing this issue. If you could reproduce the issue the please drop a comment and we will reopen the ticket.

closed 2023-04-14 by hmalpani