espressif/esp-idf #11044: Stack Overflow and PortEnterCriticalTimeout in NMEA Parser Example Code

violinoleum, 2023-03-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Board: ESP32-S3 WROOM-2
ESP-IDF Version: 5.0
Using Espressif VS Code Extension
GPS Model: Neo-6M GT-U7

Hello, I made a project template using the NMEA Parser Example, and I ran the code but I'm running into a stack overflow issue and then a Port Critical Timeout Issue. My log file is pasted below. Do you know what could be causing the issue? I turned off statement support for all except GLL, but the same issue occurs. I also increased my NMEA Parser Ring Buffer Size from 1024 to 2048, and there is still the same issue. I also tried the "Detect flash size when flashing bootloader" option in the configuration, but it also didn't fix the issue;

I (209) cpu_start: Application information:
I (211) cpu_start: Project name:     GPSTest
I (216) cpu_start: App version:      1
I (221) cpu_start: Compile time:     Mar 23 2023 15:25:51
I (227) cpu_start: ELF file SHA256:  50dd0e75474fda06...
I (233) cpu_start: ESP-IDF:          v5.0.1
I (238) cpu_start: Min chip rev:     v0.0
I (242) cpu_start: Max chip rev:     v0.99 
I (247) cpu_start: Chip rev:         v0.1
I (252) heap_init: Initializing. RAM available for dynamic allocation:
I (259) heap_init: At 3FC95078 len 00054698 (337 KiB): D/IRAM
I (266) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (272) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (278) heap_init: At 600FE010 len 00001FF0 (7 KiB): RTCRAM
I (286) spi_flash: detected chip: mxic
W (289) spi_flash: Detected flash size > 16 MB, but access beyond 16 MB is not supported for this flash model yet.
I (300) spi_flash: flash io: dio
I (305) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (325) uart: queue free spaces: 16
I (325) nmea_parser: NMEA Parser init OK
W (345) gps_demo: Unknown statement:PGLL,,,,,,V,N*64

W (1215) gps_demo: Unknown statement:$GPRMC,,V,,,,,,,,,,N*53

W (1265) gps_demo: Unknown statement:$GPVTG,,,,,,,,,N*30

W (1315) gps_demo: Unknown statement:$GPGGA,,,,,,0,00,99.99,,,,,,*48

W (1365) gps_demo: Unknown statement:$GPGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99*30

I (1415) gps_demo: 2000/0/0 63:87:52 => 
                                                latitude   = 0.00000°N
                                                longitude = 0.00000°E
                                                altitude   = 0.00m
                                                speed      = 0.000000m/s
W (2215) gps_demo: Unknown statement:$GPRMC,,V,,,,,,,,,,N*53

W (2265) gps_demo: Unknown statement:$GPVTG,,,,,,,,,N*30

W (2315) gps_demo: Unknown statement:$GPGGA,,,,,,0,00,99.99,,,,,,*48

W (2365) gps_demo: Unknown statement:$GPGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99*30

I (2415) gps_demo: 2000/0/0 63:87:52 => 
                                                latitude   = 0.00000°N
                                                longitude = 0.00000°E
                                                altitude   = 0.00m
                                                speed      = 0.000000m/s
W (3215) gps_demo: Unknown statement:$GPRMC,,V,,,,,,,,,,N*53

W (3265) gps_demo: Unknown statement:$GPVTG,,,,,,,,,N*30

W (3315) gps_demo: Unknown statement:$GPGGA,,,,,,0,00,99.99,,,,,,*48

[... 288 lines cut ...]

A2      : 0x00000000  A3      : 0x0000abab  A4      : 0x8037db5e  A5      : 0x3fcf1360
A6      : 0x00060023  A7      : 0x00000000  A8      : 0x8200c0d6  A9      : 0x3fcf4550
A10     : 0x00000000  A11     : 0x00000003  A12     : 0x00000008  A13     : 0x00000101
A14     : 0x0000000c  A15     : 0xfffffff6  SAR     : 0x00000000  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace: 0x40378b43:0x3fcf4590 0x42002466:0x3fcf45b0 0x4037c181:0x3fcf45d0 0x4037d861:0x3fcf45f0
0x40378b43: xt_utils_wait_for_intr at C:/Users/User/esp/esp-idf/components/xtensa/include/xt_utils.h:81
 (inlined by) esp_cpu_wait_for_intr at C:/Users/User/esp/esp-idf/components/esp_hw_support/cpu.c:101

0x42002466: esp_vApplicationIdleHook at C:/Users/User/esp/esp-idf/components/esp_system/freertos_hooks.c:59

0x4037c181: prvIdleTask at C:/Users/User/esp/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:4157 (discriminator 1)

0x4037d861: vPortTaskWrapper at C:/Users/User/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:154

ELF file SHA256: 50dd0e75474fda06

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375729
0x40375729: esp_restart_noos_dig at C:/Users/User/esp/esp-idf/components/esp_system/esp_system.c:64 (discriminator 1)

ESP-Marius, 2023-03-24:
Did you try increasing `CONFIG_NMEA_PARSER_TASK_STACK_SIZE` in menuconfig? The default of 2048 bytes does seem a bit small..

violinoleum, 2023-03-24:
Thank you for your advice. I did try to increase the stack size, but it still crashed. It seems that there is also a limit to how large the stack can be, according to the code in Kconfig.projbuild:

config NMEA_PARSER_TASK_STACK_SIZE
        int "NMEA Parser Task Stack Size"
        range 0 4096
        default 2048
        help
            Stack size of NMEA Parser task.

The max value is 4096 if I'm interpreting it correctly.

Also I did some more debugging, and it seems that the issue only happens after 10 seconds, once the code reaches the nmea_parser_remove_handler() and nmea_parser_deinit() in app_main(). So I "fixed" the issue by replacing 
vTaskDelay(10000 / portTICK_PERIOD_MS); in app_main() with 

while(1){
        vTaskDelay(1000 / portTICK_PERIOD_MS);
 }

so that it does not reach the nmea_parser_remove_handler(). This likely does not solve the underlying issue, but it is enough for the purpose that I'm using the code for.

Thank you for your help with resolving this issue!

ESP-Marius, 2023-03-24:
I just did a quick test locally, the example does indeed have the potential for issues if ran with the default stack size of 2048 as we are right on the stack border.

I'll increase the default value.

Are you sure you are still having issues if you change the size to 4000 (and no other changes to the example)? At least it seems to work fine for me.

violinoleum, 2023-03-24:
Hi, I followed your advice and changed the stack size to 4000, and it fixed the issue. I think when I thought I changed the stack size before, I had really just changed the NMEA Parser Ring Buffer Size instead, and I misremembered. With a stack size of 4000, it consistently works now, even when testing the vTaskDelay as 30 seconds instead of the original 10 seconds. Thank you so much for your help!

closed 2023-03-24 by violinoleum

ESP-Marius, 2023-03-24:
No problem. I've also submitted an MR internally to increase the default to prevent other people from running into the same issue.