espressif/esp-idf #11048: Error in creating the spiffs storage in Encrypted flash

YaswanthKumarSaravanaKumar, 2023-03-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.1

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP-32 Wrover B|custom board

### Power Supply used.

USB

### What is the expected behavior?

I have implemented spiffs storage and moved my webserver files into the spiffs storage. after flashing into the device, I will able to access my webserver as well but now there is continuous panic happening.  and one more thing the device is flash encrypted is it possible to create and access spiff storage in the flash-encrypted device? I also go through some forums they mentioned it is not possible to use spiffs in case of flash-encrypted devices. is there any possible configuration needed to access files in spiffs storage .

### What is the actual behavior?

without moving the web server into the spiffs storage the webserver works fine we are able to access it and modify it.

### Steps to reproduce.

Create a spiffs storage and access it in file-encrypted devices.

### Debug Logs.

```plain
The panic code says the following.

[0;31mE (89725) SPIFFS: spiffs can not run on encrypted partition

abort() was called at PC 0x400eb40c on core 0<\r><\n>

Backtrace:0x40081a8d:0x3ffd3cf00x4008e9c9:0x3ffd3d10 0x40095f82:0x3ffd3d30 0x400eb40c:0x3ffd3da0 0x400ec665:0x3ffd3e10 0x400e4485:0x3ffd3e80 0x401a1e4e:0x3ffd3ef0 0x401a1881:0x3ffd3f20 0x401a1961:0x3ffd3f60 0x40092409:0x3ffd3f80

ELF file SHA256: e9c7b90bb7308e36
Rebooting...
the
```

igrr, 2023-03-24:
> [0;31mE (89725) SPIFFS: spiffs can not run on encrypted partition

@YaswanthKumarSaravanaKumar  As the error message says, it is not possible to use SPIFFS on a partition with hardware flash encryption enabled. This is a limitation due to the way SPIFFS performs flash operations. SPIFFS relies on the "NAND write" property of NOR flash, where the additional bits can be cleared in a flash page without performing an erase operation. With hardware flash encryption enabled, clearing a bit in plaintext results also in changes 0 ->1 in ciphertext, but changing a bit in Flash from 0 to 1 can only be done via an erase operation. SPIFFS would lose it's wear levelling properties if these additional erase operations were performed. Put shortly, this is not something that would be implemented.

Please consider using FATFS for webserver files as an alternative. FATFS can be used also in encrypted flash partitions.

For storage of data which can be modified at run time we also recommend NVS.  Compared to FAT, NVS partitions are more tolerant of unexpected power loss,.which in case of FAT can cause data loss if you were writing to a partition. Read-only FAT partitions should be fine, though.

YaswanthKumarSaravanaKumar, 2023-03-24:
@igrr Thanks for the reply actually the FATFS you referring to Storing the HTML in an external sd card? actually, the real problem statement of my project is a storage issue the HTML file is large and the total bin size is around 3MB so in during OTA the device crashed due to insufficient storage in the partition table the device is also flash encrypted i also followed the reducing the bin size steps you espressif mentioned their document even though the HTML is larger so I need to move that HTML to any other storage. so I picked SPIFFS but it is not possible due to flash encryption. the FATFS needs to access an external sd card but my hardware doesn't have does sd card configuration. so is there any method to store those HTML? otherwise is there any FATFS capability to use internal storage other than using sd card functionality?

Thanks in advance.

igrr, 2023-03-24:
FATFS can be used in internal flash as well, not only in SD cards.

Starting from IDF v5.0,  `examples/storage/fatfsgen/` example is provided in IDF, it shows how to automatically create an upload files to a FAT filesystem in internal flash.

The feature which this example demonstrates (generation of a FAT filesystem on the host side) is not available in IDF v4.4.1 which you are using, though. You can still use FATFS in internal flash in IDF v4.4.1 (using examples/storage/wear_levelling) but you won't have access to the tool to create FATFS images on the host.

You can consider either upgrading to IDF v5.0.1 to get this feature.

YaswanthKumarSaravanaKumar, 2023-03-24:
Thanks for the feedback the above points are so useful for me, I will check it out and come back to you if any further changes.

closed 2023-03-31 by YaswanthKumarSaravanaKumar