espressif/esp-idf #11054: org.eclipse.ui.navigator java.lang.IllegalArgumentException: newPosition > limit: (973669377 > 176322)

RJSDevel, 2023-03-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp-idf-master-v5.1

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32C6 DIY

### Power Supply used.

USB

### What is the expected behavior?

Run debigging

### What is the actual behavior?

[image]
[image]
[log.log](https://github.com/espressif/esp-idf/files/11071461/log.log)

### Steps to reproduce.

1. Setup esp32c6 toolchans
2. Run debug
3. Get the error
4. If priceed it than debug doesn't catch breakpoints, just flash and terminate

### Debug Logs.

```plain
in What is the actual behavior?
```

### More Information.

#Sun Mar 26 16:21:55 YEKT 2023
org.eclipse.core.runtime=2
org.eclipse.platform=4.25.0.v20220831-1800

ESP-Marius, 2023-03-27:
@RJSDevel for IDF eclipse plugin related issues you might be better of reporting it directly at https://github.com/espressif/idf-eclipse-plugin

RJSDevel, 2023-03-28:
Could you move it? I can't see any opportunity

ESP-Marius, 2023-03-28:
Sorry, I dont have access to do it either. Think you'll just have to close this one and open a new one.