espressif/esp-idf #11055: esp32 compiles helloworld example on windows and reports ninja error

i-jaffer, 2023-03-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.1

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

I wish i could compile esp idf project with powershell

### What is the actual behavior?

report ninja error！
```bash
Executing action: all (aliases: build)
Running ninja in directory e:\esp-idf\esp-idf\examples\get-started\hello_world\build
Executing "ninja all"...
ninja: error: 'E:/esp-idf/esp-idf/components/esp_wifi/lib/esp32/libcoexist.a', needed by 'hello_world.elf', missing and no known rule to make it
ninja failed with exit code 1, output of the command is in the e:\esp-idf\esp-idf\examples\get-started\hello_world\build\log\idf_py_stderr_output_7424 and e:\esp-idf\esp-idf\examples\get-started\hello_world\build\log\idf_py_stdout_output_7424
PS E:\esp-idf\esp-idf\examples\get-started\hello_world> idf.py build
Executing action: all (aliases: build)
Running ninja in directory e:\esp-idf\esp-idf\examples\get-started\hello_world\build
Executing "ninja all"...
ninja: error: 'E:/esp-idf/esp-idf/components/esp_wifi/lib/esp32/libcoexist.a', needed by 'hello_world.elf', missing and no known rule to make it
ninja failed with exit code 1, output of the command is in the e:\esp-idf\esp-idf\examples\get-started\hello_world\build\log\idf_py_stderr_output_25140 and e:\esp-idf\esp-idf\examples\get-started\hello_world\build\log\idf_py_stdout_output_25140
```

### Steps to reproduce.

I just configured the esp32 compilation environment on windows with powershell, and then tested it;

My compilation environment configuration is as follows:

1. git clone esp-idf project
2. git checkout v5.0.1
3. Open powershell and run ./install.bat
4. Run ./export.ps1 to import environment variables
5. Enter the helloworld example directory, the relative path is "esp-idf\examples\get-started\hello_world"
6. After running idf.py build, ninja reports an error, and the error message is as follows:

```bash
Executing action: all (aliases: build)
Running ninja in directory e:\esp-idf\esp-idf\examples\get-started\hello_world\build
Executing "ninja all"...
ninja: error: 'E:/esp-idf/esp-idf/components/esp_wifi/lib/esp32/libcoexist.a', needed by 'hello_world.elf', missing and no known rule to make it
ninja failed with exit code 1, output of the command is in the e:\esp-idf\esp-idf\examples\get-started\hello_world\build\log\idf_py_stderr_output_7424 and e:\esp-idf\esp-idf\examples\get-started\hello_world\build\log\idf_py_stdout_output_7424
PS E:\esp-idf\esp-idf\examples\get-started\hello_world> idf.py build
Executing action: all (aliases: build)
Running ninja in directory e:\esp-idf\esp-idf\examples\get-started\hello_world\build
Executing "ninja all"...
ninja: error: 'E:/esp-idf/esp-idf/components/esp_wifi/lib/esp32/libcoexist.a', needed by 'hello_world.elf', missing and no known rule to make it
ninja failed with exit code 1, output of the command is in the e:\esp-idf\esp-idf\examples\get-started\hello_world\build\log\idf_py_stderr_output_25140 and e:\esp-idf\esp-idf\examples\get-started\hello_world\build\log\idf_py_stdout_output_25140
```

[... 265 lines cut ...]
Requirement already satisfied: six in c:\users\mr. jaffer\.espressif\python_env\idf5.0_py3.10_env\lib\site-packages (from idf-component-manager->-r E:\esp-idf\esp-idf\tools\requirements\requirements.core.txt (line 10)) (1.16.0)
Requirement already satisfied: pyyaml>5.2 in c:\users\mr. jaffer\.espressif\python_env\idf5.0_py3.10_env\lib\site-packages (from idf-component-manager->-r E:\esp-idf\esp-idf\tools\requirements\requirements.core.txt (line 10)) (6.0)
Requirement already satisfied: cachecontrol[filecache]>0.12.6 in c:\users\mr. jaffer\.espressif\python_env\idf5.0_py3.10_env\lib\site-packages (from idf-component-manager->-r E:\esp-idf\esp-idf\tools\requirements\requirements.core.txt (line 10)) (0.12.11)
Requirement already satisfied: requests-file in c:\users\mr. jaffer\.espressif\python_env\idf5.0_py3.10_env\lib\site-packages (from idf-component-manager->-r E:\esp-idf\esp-idf\tools\requirements\requirements.core.txt (line 10)) (1.5.1)
Requirement already satisfied: schema in c:\users\mr. jaffer\.espressif\python_env\idf5.0_py3.10_env\lib\site-packages (from idf-component-manager->-r E:\esp-idf\esp-idf\tools\requirements\requirements.core.txt (line 10)) (0.7.5)
Requirement already satisfied: packaging in c:\users\mr. jaffer\.espressif\python_env\idf5.0_py3.10_env\lib\site-packages (from idf-component-manager->-r E:\esp-idf\esp-idf\tools\requirements\requirements.core.txt (line 10)) (23.0)
Requirement already satisfied: requests<3 in c:\users\mr. jaffer\.espressif\python_env\idf5.0_py3.10_env\lib\site-packages (from idf-component-manager->-r E:\esp-idf\esp-idf\tools\requirements\requirements.core.txt (line 10)) (2.28.2)
Requirement already satisfied: contextlib2>0.6.0 in c:\users\mr. jaffer\.espressif\python_env\idf5.0_py3.10_env\lib\site-packages (from idf-component-manager->-r E:\esp-idf\esp-idf\tools\requirements\requirements.core.txt (line 10)) (21.6.0)
Requirement already satisfied: requests-toolbelt in c:\users\mr. jaffer\.espressif\python_env\idf5.0_py3.10_env\lib\site-packages (from idf-component-manager->-r E:\esp-idf\esp-idf\tools\requirements\requirements.core.txt (line 10)) (0.10.1)
Requirement already satisfied: tqdm<5 in c:\users\mr. jaffer\.espressif\python_env\idf5.0_py3.10_env\lib\site-packages (from idf-component-manager->-r E:\esp-idf\esp-idf\tools\requirements\requirements.core.txt (line 10)) (4.65.0)
Requirement already satisfied: construct~=2.10 in c:\users\mr. jaffer\.espressif\python_env\idf5.0_py3.10_env\lib\site-packages (from esp-coredump->-r E:\esp-idf\esp-idf\tools\requirements\requirements.core.txt (line 11)) (2.10.68)
Requirement already satisfied: pygdbmi>=0.9.0.2 in c:\users\mr. jaffer\.espressif\python_env\idf5.0_py3.10_env\lib\site-packages (from esp-coredump->-r E:\esp-idf\esp-idf\tools\requirements\requirements.core.txt (line 11)) (0.9.0.2)
Requirement already satisfied: bitstring>=3.1.6 in c:\users\mr. jaffer\.espressif\python_env\idf5.0_py3.10_env\lib\site-packages (from esptool->-r E:\esp-idf\esp-idf\tools\requirements\requirements.core.txt (line 12)) (4.0.1)
Requirement already satisfied: ecdsa>=0.16.0 in c:\users\mr. jaffer\.espressif\python_env\idf5.0_py3.10_env\lib\site-packages (from esptool->-r E:\esp-idf\esp-idf\tools\requirements\requirements.core.txt (line 12)) (0.18.0)
Requirement already satisfied: reedsolo<=1.6.0,>=1.5.3 in c:\users\mr. jaffer\.espressif\python_env\idf5.0_py3.10_env\lib\site-packages (from esptool->-r E:\esp-idf\esp-idf\tools\requirements\requirements.core.txt (line 12)) (1.6.0)
Requirement already satisfied: msgpack>=0.5.2 in c:\users\mr. jaffer\.espressif\python_env\idf5.0_py3.10_env\lib\site-packages (from cachecontrol[filecache]>0.12.6->idf-component-manager->-r E:\esp-idf\esp-idf\tools\requirements\requirements.core.txt (line 10)) (1.0.5)
Requirement already satisfied: lockfile>=0.9 in c:\users\mr. jaffer\.espressif\python_env\idf5.0_py3.10_env\lib\site-packages (from cachecontrol[filecache]>0.12.6->idf-component-manager->-r E:\esp-idf\esp-idf\tools\requirements\requirements.core.txt (line 10)) (0.12.2)
Requirement already satisfied: pycparser in c:\users\mr. jaffer\.espressif\python_env\idf5.0_py3.10_env\lib\site-packages (from cffi>=1.12->cryptography->-r E:\esp-idf\esp-idf\tools\requirements\requirements.core.txt (line 7)) (2.21)
Requirement already satisfied: charset-normalizer<4,>=2 in c:\users\mr. jaffer\.espressif\python_env\idf5.0_py3.10_env\lib\site-packages (from requests<3->idf-component-manager->-r E:\esp-idf\esp-idf\tools\requirements\requirements.core.txt (line 10)) (3.1.0)
Requirement already satisfied: idna<4,>=2.5 in c:\users\mr. jaffer\.espressif\python_env\idf5.0_py3.10_env\lib\site-packages (from requests<3->idf-component-manager->-r E:\esp-idf\esp-idf\tools\requirements\requirements.core.txt (line 10)) (3.4)
Requirement already satisfied: certifi>=2017.4.17 in c:\users\mr. jaffer\.espressif\python_env\idf5.0_py3.10_env\lib\site-packages (from requests<3->idf-component-manager->-r E:\esp-idf\esp-idf\tools\requirements\requirements.core.txt (line 10)) (2022.12.7)
Requirement already satisfied: urllib3<1.27,>=1.21.1 in c:\users\mr. jaffer\.espressif\python_env\idf5.0_py3.10_env\lib\site-packages (from requests<3->idf-component-manager->-r E:\esp-idf\esp-idf\tools\requirements\requirements.core.txt (line 10)) (1.26.15)
All done! You can now run:
   export.bat
```

igrr, 2023-03-26:
Hi @i-jaffer, i would suggest performing the step which is given in this warning:

```
Git submodule components/bt/controller/lib_esp32c3_family is out of date.
  Run 'git submodule update --init --recursive' to fix.
```

Please try running `git submodule update --init --recursive` in esp-idf directory.

i-jaffer, 2023-03-27:
Thank you for your reply.

After switching to `v5.0.1` branch, I did not run `git status` to check whether the partition was clean. 

I missed the unsynchronized submodule, which caused the problem.

I re-ran `git submodule update --init --recursive` and solved it

The full details of the environment have been compiled for a blog to share with others who need to refer to it: 

[ESP-IDF + (PowerShell / Vscode) ESP32开发环境搭建（windows)](https://blog.csdn.net/qq_43332314/article/details/129776889)