espressif/esp-idf #11056: Need example code for sending data to Bluetooth LE devices with separated WRITE and NOTIFY UUIDs

prazsmarid, 2023-03-26:
Hello,
Can somebody send me an example for sending data to a BLE server which is using separated characteristic UUID for the NOTIFY and WRITE properties?
In all of the examples that I have found the NOTIFY and WRITE properties are in the characteristic UUID.
Thanks in advance.
[image]
[image]

prazsmarid, 2023-04-06:
Finally I have searched a group of programmers, after paying to them they solved the problem. It is very disappointing that nobody helped from the official support.

xiewenxiang, 2023-04-18:
Hi @prazsmarid ,

You can refer to the definition of this service:

```c
static const uint8_t char_prop_write               = ESP_GATT_CHAR_PROP_BIT_WRITE;
static const uint8_t char_prop_notify   =  ESP_GATT_CHAR_PROP_BIT_NOTIFY;
static const uint8_t test_ccc[2]      = {0x00, 0x00};
static const uint8_t test_char_value[4]                 = {0x11, 0x22, 0x33, 0x44};

/* Full Database Description - Used to add attributes into the database */
static const esp_gatts_attr_db_t test_gatt_db[6] =
{
    // Service Declaration
    [0]        =
    {{ESP_GATT_AUTO_RSP}, {ESP_UUID_LEN_16, (uint8_t *)&primary_service_uuid, ESP_GATT_PERM_READ,
      sizeof(uint16_t), sizeof(GATTS_SERVICE_UUID_TEST), (uint8_t *)&GATTS_SERVICE_UUID_TEST}},

    /* Characteristic Declaration */
    [1]     =
    {{ESP_GATT_AUTO_RSP}, {ESP_UUID_LEN_16, (uint8_t *)&character_declaration_uuid, ESP_GATT_PERM_READ,
      CHAR_DECLARATION_SIZE, CHAR_DECLARATION_SIZE, (uint8_t *)&char_prop_notify}},

    /* Characteristic Value */
    [2] =
    {{ESP_GATT_AUTO_RSP}, {ESP_UUID_LEN_16, (uint8_t *)&GATTS_CHAR_UUID_TEST_A, ESP_GATT_PERM_READ,
      GATTS_DEMO_CHAR_VAL_LEN_MAX, sizeof(test_char_value), (uint8_t *)test_char_value}},

    /* Client Characteristic Configuration Descriptor */
    [3]  =
    {{ESP_GATT_AUTO_RSP}, {ESP_UUID_LEN_16, (uint8_t *)&character_client_config_uuid, ESP_GATT_PERM_READ | ESP_GATT_PERM_WRITE,
      sizeof(uint16_t), sizeof(test_ccc), (uint8_t *)test_ccc}},

    /* Characteristic Declaration */
    [4]      =
    {{ESP_GATT_AUTO_RSP}, {ESP_UUID_LEN_16, (uint8_t *)&character_declaration_uuid, ESP_GATT_PERM_READ,
      CHAR_DECLARATION_SIZE, CHAR_DECLARATION_SIZE, (uint8_t *)&char_prop_write}},

    /* Characteristic Value */
    [5]  =
    {{ESP_GATT_AUTO_RSP}, {ESP_UUID_LEN_16, (uint8_t *)&GATTS_CHAR_UUID_TEST_C, ESP_GATT_PERM_READ | ESP_GATT_PERM_WRITE,
      GATTS_DEMO_CHAR_VAL_LEN_MAX, sizeof(test_char_value), (uint8_t *)test_char_value}},

};
```

prazsmarid, 2023-04-18:
Hello,
Thank you, but it was resolved some weeks ago.

closed 2023-10-25 by Alvin1Zhang