espressif/esp-idf #11058: spi_flash function and type are undefined when I am building blink example

waruqi, 2023-03-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-4051-g4f0769d2ed

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I added esp-sr component to `main/idf_component.yml` file in get-started/blink example project.

```
$ cat main/idf_component.yml
dependencies:
  espressif/led_strip: "^2.0.0"
  espressif/esp-sr: "^1.2.0"
```

then run 

```
idf.py build
```

It will build fails.

### What is the actual behavior?

build ok

### Steps to reproduce.

```
1. added `espressif/esp-sr: "^1.2.0"` to `main/idf_component.yml` file in get-started/blink example project.
2. run `rm -rf build/; rm -rf managed_components/; idf.py set-target esp32s3; idf.py build`
```

### Build or installation Logs.

```plain
In file included from /test/managed_components/espressif__esp-sr/src/model_path.c:5:
/test/managed_components/espressif__esp-sr/src/include/model_path.h:21:5: error: unknown type name 'spi_flash_mmap_handle_t'
   21 |     spi_flash_mmap_handle_t mmap_handle; // mmap_handle if using esp_partition_mmap else NULL;
      |     ^~~~~~~~~~~~~~~~~~~~~~~
/test/managed_components/espressif__esp-sr/src/model_path.c: In function 'srmodel_mmap_init':
/test/managed_components/espressif__esp-sr/src/model_path.c:248:59: error: 'SPI_FLASH_MMAP_DATA' undeclared (first use in this function)
  248 |     esp_err_t err=esp_partition_mmap(part, 0, part->size, SPI_FLASH_MMAP_DATA, &root, &models->mmap_handle);
      |                                                           ^~~~~~~~~~~~~~~~~~~
/test/managed_components/espressif__esp-sr/src/model_path.c:248:59: note: each undeclared identifier is reported only once for each function it appears in
/test/managed_components/espressif__esp-sr/src/model_path.c:248:87: warning: passing argument 6 of 'esp_partition_mmap' from incompatible pointer type [-Winc
ompatible-pointer-types]
  248 |     esp_err_t err=esp_partition_mmap(part, 0, part->size, SPI_FLASH_MMAP_DATA, &root, &models->mmap_handle);
      |                                                                                       ^~~~~~~~~~~~~~~~~~~~
      |                                                                                       |
      |                                                                                       int *
[... 91 lines cut ...]
      |                                     ^~~~~~~~~~
/Users/ruki/projects/personal/esp-idf/components/log/include/esp_log.h:414:86: note: in expansion of macro 'LOG_FORMAT'
  414 |         else                                { esp_log_write(ESP_LOG_INFO,       tag, LOG_FORMAT(I, format), esp_log_timestamp(), tag, ##__VA_ARGS__); } \
      |                                                                                      ^~~~~~~~~~
/Users/ruki/projects/personal/esp-idf/components/log/include/esp_log.h:432:41: note: in expansion of macro 'ESP_LOG_LEVEL'
  432 |         if ( LOG_LOCAL_LEVEL >= level ) ESP_LOG_LEVEL(level, tag, format, ##__VA_ARGS__); \
      |                                         ^~~~~~~~~~~~~
/Users/ruki/projects/personal/esp-idf/components/log/include/esp_log.h:342:38: note: in expansion of macro 'ESP_LOG_LEVEL_LOCAL'
  342 | #define ESP_LOGI( tag, format, ... ) ESP_LOG_LEVEL_LOCAL(ESP_LOG_INFO,    tag, format, ##__VA_ARGS__)
      |                                      ^~~~~~~~~~~~~~~~~~~
/test/managed_components/espressif__esp-sr/src/model_path.c:253:9: note: in expansion of macro 'ESP_LOGI'
  253 |         ESP_LOGI(TAG, "partition %s size: %d by mmap", part->label, part->size);
      |         ^~~~~~~~
/test/managed_components/espressif__esp-sr/src/model_path.c: In function 'srmodel_mmap_deinit':
/test/managed_components/espressif__esp-sr/src/model_path.c:307:9: error: implicit declaration of function 'spi_flash_munmap' [-Werror=implicit-function-decl
aration]
  307 |         spi_flash_munmap(models->mmap_handle);
      |         ^~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
[1027/1037] Building CXX object esp-idf/espressif__esp-dsp/CMakeFiles/__idf_espressif__esp-dsp.dir/modules/kalman/ekf/common/ekf.cpp.objninja: build stopped: subcommand failed.
HINT: The issue is better to resolve by replacing format specifiers to 'PRI'-family macros (include <inttypes.h> header file).
ninja failed with exit code 1, output of the command is in the /test/build/log/idf_py_stderr_output_12182 and /test
/build/log/idf_py_stdout_output_12182
```
```

ESP-Marius, 2023-03-27:
I think at the moment `esp-sr` do not support IDF v5

ESP-Marius, 2023-03-27:
I see people are working on getting all the libraries working on IDF 5.0 but for now I think most are still only compatible with v4.4.

waruqi, 2023-03-27:
ok, I will try 4.4. thanks!

waruqi, 2023-03-27:
But I switch to v4.4, it will error when I run `. export.sh` or `./install.sh`

```
$ ./install.sh 
Detecting the Python interpreter
Checking "python" ...
Checking "python3" ...
Python 3.10.8
"python3" has been detected
Installing ESP-IDF tools
Traceback (most recent call last):
  File "/Users/ruki/projects/personal/esp-idf/tools/idf_tools.py", line 1828, in <module>
    main(sys.argv[1:])
  File "/Users/ruki/projects/personal/esp-idf/tools/idf_tools.py", line 1824, in main
    action_func(args)
  File "/Users/ruki/projects/personal/esp-idf/tools/idf_tools.py", line 1420, in action_install
    targets = clean_targets(args.targets)
  File "/Users/ruki/projects/personal/esp-idf/tools/idf_tools.py", line 1083, in clean_targets
    export_targets_to_idf_env_json(targets_from_tools_json)
  File "/Users/ruki/projects/personal/esp-idf/tools/idf_tools.py", line 1051, in export_targets_to_idf_env_json
    targets = list(set(targets + get_user_defined_targets()))
  File "/Users/ruki/projects/personal/esp-idf/tools/idf_tools.py", line 1097, in get_user_defined_targets
    if env == idf_env_json['idfSelectedId']:
KeyError: 'idfSelectedId'
ruki-2:esp-idf ruki$ . export.sh 
Detecting the Python interpreter
Checking "python" ...
Checking "python3" ...
Python 3.10.8
"python3" has been detected
Adding ESP-IDF tools to PATH...
Traceback (most recent call last):
  File "/Users/ruki/projects/personal/esp-idf/tools/idf_tools.py", line 1828, in <module>
    main(sys.argv[1:])
  File "/Users/ruki/projects/personal/esp-idf/tools/idf_tools.py", line 1824, in main
    action_func(args)
  File "/Users/ruki/projects/personal/esp-idf/tools/idf_tools.py", line 1175, in action_export
    tools_info = filter_tools_info(tools_info)
  File "/Users/ruki/projects/personal/esp-idf/tools/idf_tools.py", line 1117, in filter_tools_info
    targets = get_user_defined_targets()
  File "/Users/ruki/projects/personal/esp-idf/tools/idf_tools.py", line 1097, in get_user_defined_targets
    if env == idf_env_json['idfSelectedId']:
KeyError: 'idfSelectedId'
```

feizi, 2023-03-27:
Yes, currently esp-sr(master and releases) are still only compatible with IDF v4.4.  
I am updating the mmap interface to support IDF v5.0.

waruqi, 2023-03-27:
> Yes, currently esp-sr(master and releases) are still only compatible with IDF v4.4. I am updating the mmap interface to support IDF v5.0.

Can I ask how long it will take for idf 5.0 to support esp-sr? If it's a long wait, I'm going to try 4.4 first

dobairoland, 2023-03-27:
> But I switch to v4.4, it will error ...

v4.4 was suggested but use the latest bugfix release of it, ie. 4.4.4 at the moment, or the latest version of the `release/v4.4` branch. Otherwise you might experience any of the issues already fixed since the initial release of v4.4.

waruqi, 2023-03-27:
> > But I switch to v4.4, it will error ...
> 
> v4.4 was suggested but use the latest bugfix release of it, ie. 4.4.4 at the moment, or the latest version of the branch. Otherwise you might experience any of the issues already fixed since the initial release of v4.4.`release/v4.4`

it works for 4.4.4, thanks!

feizi, 2023-03-27:
The master of esp-sr has support IDF v5.0.  
But for now we recommend to use IDF v4.4 because the avaliable free storage is less than the left free flash space on IDF v5.0. This bug is being resolved.

waruqi, 2023-03-27:
> The master of esp-sr has support IDF v5.0. But for now we recommend to use IDF v4.4 because the avaliable free storage is less than the left free flash space on IDF v5.0. This bug is being resolved.

Thanks, but I tried [esp-skanet](https://github.com/espressif/esp-skainet/tree/master/examples/wake_word_detection) example, it will crash in esp-sr/srmodel_mmap_init() on esp32s3 with esp-idf 4.4.4

crash logs: https://github.com/espressif/esp-sr/issues/61

waruqi, 2023-03-27:
and If I use newest commit from esp-sr for idf 4.4.4, it will build error.

```
/Users/ruki/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /Users/ruki/projects/persona
l/hellokiki/ottobot/components/esp-sr/lib/esp32s3/libesp_audio_processor.a(checks.cc.obj):(.literal._ZN3rtc12FatalMessageD2Ev+0x4): undefined reference to `std::__cxx11::basic_stringbuf<char, std::c
har_traits<char>, std::allocator<char> >::str() const &'
/Users/ruki/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /Users/ruki/projects/persona
l/hellokiki/ottobot/components/esp-sr/lib/esp32s3/libesp_audio_processor.a(checks.cc.obj):(.literal.rtc_FatalMessage+0x0): undefined reference to `std::__cxx11::basic_ostringstream<char, std::char_t
raits<char>, std::allocator<char> >::basic_ostringstream()'
/Users/ruki/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /Users/ruki/projects/persona
l/hellokiki/ottobot/components/esp-sr/lib/esp32s3/libesp_audio_processor.a(checks.cc.obj): in function `rtc::FatalMessage::~FatalMessage()':
/home/sunxiangyu/workspace/esp_sr_lib_without_hw/build/../components/esp_audio_processor/rtc_base/checks.cc:82: undefined reference to `std::__cxx11::basic_stringbuf<char, std::char_traits<char>, st
d::allocator<char> >::str() const &'
/Users/ruki/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /Users/ruki/projects/persona
l/hellokiki/ottobot/components/esp-sr/lib/esp32s3/libesp_audio_processor.a(checks.cc.obj): in function `std::ctype<char> const& std::__check_facet<std::ctype<char> >(std::ctype<char> const*)':
/home/sunxiangyu/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/xtensa-esp32s3-elf/include/c++/11.2.0/bits/basic_ios.h:49: undefined reference to `std::__cxx11::basic_ostri
ngstream<char, std::char_traits<char>, std::allocator<char> >::basic_ostringstream()'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

formatBCE, 2023-03-29:
So no luck using esp32s3 then?