espressif/esp-idf #11064: Support for SSD1983 (parallel) in examples/peripherals/lcd

Designer-Rick, 2023-03-27:
### Describe the solution you'd like.

Is it on the horizon to support SSD1963 within IDF5.x examples for lcd peripherals?

suda-morris, 2023-03-28:
hi @Designer-Rick this is a BSP feature request, please consider asking for help at https://github.com/espressif/esp-bsp/issues

BTW, you can find many SSD1983 drivers in GitHub already: https://github.com/search?q=ssd1963 What you need to do is to adapt them to the driver framework used by esp-idf. There're some LCD drivers that can be used for reference: https://github.com/espressif/esp-bsp/tree/master/components/lcd

Designer-Rick, 2023-03-29:
Thank you for the response.  Driving the SSD1963 is not what I have a problem with, its how to get data through the i2s in 8 bit parallel mode "Fast".

tore-espressif, 2023-03-30:
Hi @Designer-Rick , if you are interested only in the I2S parallel part, you can check our lcd examples. The I2S configuration part starts here https://github.com/espressif/esp-idf/blob/master/examples/peripherals/lcd/i80_controller/main/i80_controller_example_main.c#L156

Providing a driver for SSD1963 is on our roadmap, but ATM, we can't provide you a specific deadline. More info about LCD drivers working with `esp_lcd` component is kept [here](https://github.com/espressif/esp-bsp/blob/master/LCD.md)

Designer-Rick, 2023-03-30:
It is more driving the I2S in parallel I am interested in, thank you.