espressif/esp-idf #11065: aliyun demo bug

king592, 2023-03-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

V5.0.1

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

目前我使用的是ESP-IDF5.0.1开发环境，然后我编译阿里云例程是时候报错，这个是例程链接https://github.com/espressif/esp-aliyun 

### What is the actual behavior?

编译提示没有 system_event_cb_t 这个类型的定义，是因为版本不兼容吗

### Steps to reproduce.

下载阿里云代码之后，编译发现esp_event_loop.h好像版本更替了，然后我换成esp_event.h，编译的时候提示报错如下
[image]

### Build or installation Logs.

```plain
unknown type name 'system_event_cb_t'
unknown type name 'system_event_t'
unknown type name 'system_event_cb_t'
implicit declaration of function 'tcpip_adapter_init' [-Werror=implicit-function-declaration]
```

wujiangang, 2023-03-28:
esp-aliyun 是在 idf v4.2 版本上适配的，此仓库当前已不再维护。之后的 idf 版本对一些功能就行了重构，所以导致编译问题。如需使用 v5.0，请参考 IDF 编程手册进行相关接口的适配。比如 system_event 部分可参考 https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/esp_event.html，tcpip_adapter_init 可参考 https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_netif.html 接口。

Alvin1Zhang, 2023-03-31:
Thanks for reporting, feel free to reopen.

closed 2023-03-31 by Alvin1Zhang

LHYYSY, 2023-04-12:
[image]
Version idf5.0.1 fixes errors in Aliyun