espressif/esp-idf #11066: External 32khz crystal not working with ESP BLE MESH application

Paramesh19, 2023-03-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

@xiongweichao  Hi all,
I am using ESP-32C3DevKitC-2 for BLE Mesh Application, Here my main Concern is which power saving I need to operate the 
 BLE nodes with Li-ion Battery for minimum 20 days, but now the node is consuming 90 mill amps of current , I need to reduce it to less than 10 Milli amp, I tried by soldering external 32.768khz crystal as RTC clock for modem sleep and made the required configuration settings in menu config but I didn't see any improvement, my question is it possible to reduce power with external crystal in mesh network or shall i need to follow another approach. Here I  am attaching sdk config file 
Kindly help me with this.
Thank you in advance.

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/11088212/sdkconfig.txt)

xiongweichao, 2023-03-28:
Hi @Paramesh19 ,

Please verify whether the ESP32C3 will enter light sleep mode.You can add printing [here](https://github.com/espressif/esp-idf/blob/81e1e6599553a646a689ad51e32a5d48b34cfec5/components/esp_hw_support/sleep_modes.c#L844) or use other methods.

Thanks

Paramesh19, 2023-03-29:
@xiongweichao Hi external clock was enabled for modem sleep but still current consumption was not reduced ,is there anything i need to add in my main code ,i am attaching the logs with crystal detected and not detected , my doubt is this external crystal concept works for ble mesh applications or not
[log1.txt](https://github.com/espressif/esp-idf/files/11097614/log1.txt)
[log2.txt](https://github.com/espressif/esp-idf/files/11097615/log2.txt)
log1 is without external crystal,log2 is with external crystal

xiongweichao, 2023-03-29:
Hi @Paramesh19 ,

If you need to reduce the current, you can try to increase the scan interval.

Thanks

Paramesh19, 2023-03-29:
@xiongweichao just confirm me one thing In BLE Mesh Concept this external crystal concept supports or not, is it possible to reduce power consumption in mesh application with this concept, if it is possible how much we can reduce?

closed 2023-04-12 by Paramesh19