espressif/esp-idf #11067: BLE-HID-Device Re-Connection is not possible

DerPicknicker, 2023-03-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello, 

I bought an BLE-HID Remote which I successfully paired with the esp32-s3 - for that I used your example: [https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/esp_hid_host](url)

Everything works as expected, Connection is done, all Information from the Device is visible in the Serial Monitor.

#### Now coming to my Problem: 

- IF the Battery dies the ESP32-S3 is not capable to re-connect to the Remote. I tried to bring the Remote back to the Paring mode but the ESP32-S3 is not able to connect again to the Remote. 
- Is this a Problem occurred by the Code or just not implemented? 

If this is not implemented yet, could you give me a hint, which Functions are used for the Re-Connection handling and what Informations need's to be stored to re-connect to that device? 

Best regards

esp-cjh, 2023-03-31:
@DerPicknicker There is no implement for reconnecting to remote device after disconnected. You must reboot the ESP32S3 and then it will reconnect to your device.

You can implement it just refer to this patch.
[0001-reconnect-to-remote-after-disconnected.patch](https://github.com/espressif/esp-idf/files/11121131/0001-reconnect-to-remote-after-disconnected.patch)

esp-cjh, 2023-04-01:
@DerPicknicker The example code is what you mentioned above `example/Bluetooth/esp_hid_host`.

DerPicknicker, 2023-04-01:
@esp-cjh .. Yes but I don’t get your point how to implement a reconnection? 
do you which functions are the most important for the BLE-Stack?

DerPicknicker, 2023-04-04:
Hi @esp-cjh .. 

I will try that thank you!

closed 2023-05-23 by Alvin1Zhang