espressif/esp-idf #11068: May I ask which channel estimation algorithm is used by ESP32-WROOM-32U to obtain channel state information(CSI)？

ogleede, 2023-03-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

May I ask which channel estimation algorithm is used by ESP32-WROOM-32U to obtain channel state information(CSI)？

CC-MASTER05, 2023-03-31:
Please refer https://www.esp32.com/viewtopic.php?t=9733

ogleede, 2023-04-04:
> Please refer https://www.esp32.com/viewtopic.php?t=9733

thank you！Sorry I didn't explain my problem clearly. What I want to ask is, what algorithm is used for channel measurement, LS or MMSE?

MaxwellAlan, 2024-04-03:
Hi @ogleede 

We use LS :)

closed 2024-06-07 by Sherry616