espressif/esp-idf #11072: rmt stepper motor is not a real s-curve

watershade, 2023-03-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

the example of stepper motor(using rmt to fultill it) is not a s-curve. 
`static float convert_to_smooth_freq(uint32_t freq1, uint32_t freq2, uint32_t freqx)
{
    float normalize_x = ((float)(freqx - freq1)) / (freq2 - freq1);
    // third-order "smoothstep" function: https://en.wikipedia.org/wiki/Smoothstep
    float smooth_x = normalize_x * normalize_x * (3 - 2 * normalize_x);
    return smooth_x * (freq2 - freq1) + freq1;
}`
It looks like you are try to use "3X^2 - 2X^3".  This fomula is right just when x and time axis is linear. because duration is changed. Every time the new square wave have a different duration. 

In acceleration phase, the duration become short and the acceleration will become faster. 
If  there is a way to change duration for every spicific time it will be right.

suda-morris, 2023-03-29:
If you want to use another formula or "change duration for every specific time", I'm afraid you may need to customize the RMT encoder. Refer to https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/rmt.html#rmt-encoder for how to.

watershade, 2023-04-07:
> If you want to use another formula or "change duration for every specific time", I'm afraid you may need to customize the RMT encoder. Refer to https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/rmt.html#rmt-encoder for how to.

Thanks for your information. My meaning is when you use this fomula, you need to care about time-linear. When you get a new period, you need to care about when it is the time. It is not so simple to use y = 3x^2 - 2x^3  (x and time are not linear)

 But anyway, lit's close it.

closed 2023-04-07 by watershade