espressif/esp-idf #11073: nvs_flash_init failed, ESP_ERR_NOT_FOUND

waruqi, 2023-03-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.4

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-WROOM-1 (ESP32-S3R8)

### Power Supply used.

USB

### What is the expected behavior?

I tried wifi/get-started/softap example. it works.

But I ported it's code to my project, it does not work and nvs_flash_init will return failure.

```c
    // initialize NVS
    esp_err_t ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
      ESP_ERROR_CHECK(nvs_flash_erase());
      ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK(ret);    <------ crashed
```

I use custom partition.csv

```
$ cat partitions.csv
# Espressif ESP32 Partition Table
# Name,  Type, SubType, Offset,  Size
factory, app,  factory, 0x010000, 2500k
model,  data, spiffs,         , 5168K,
```

```
sdkconfig
142:CONFIG_PARTITION_TABLE_OFFSET=0x8000
```

And my board has 8M flash

### What is the actual behavior?

work

### Steps to reproduce.

```
idf.py flash
idf.py monitor
```

### Debug Logs.

```plain
I (0) cpu_start: Starting scheduler on APP CPU.
I (810) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (810) I2S: DMA Malloc info, datalen=blocksize=640, dma_buf_count=6
I (820) board: init ok
I (820) network: starting wifi softap ..
ESP_ERROR_CHECK failed: esp_err_t 0x105 (ESP_ERR_NOT_FOUND) at 0x40381a4c
0x40381a4c: _esp_error_check_failed at /Users/ruki/projects/personal/esp-idf/components/esp_system/esp_err.c:42

file: "./main/network.c" line 62
func: ot_network_wifi_softap_init
expression: ret

abort() was called at PC 0x40381a4f on core 0
0x40381a4f: _esp_error_check_failed at /Users/ruki/projects/personal/esp-idf/components/esp_system/esp_err.c:43

Backtrace: 0x40379ebe:0x3fcac800 0x40381a59:0x3fcac820 0x4038945e:0x3fcac840 0x40381a4f:0x3fcac8b0 0x42007f64:0x3fcac8d0 0x42008127:0x3fcaca60 0x42007d72:0x3fcaca80 0x42007c37:0x3fcacaa0 0x420dc254:
0x3fcacac0 0x40384e21:0x3fcacae0
0x40379ebe: panic_abort at /Users/ruki/projects/personal/esp-idf/components/esp_system/panic.c:408

0x40381a59: esp_system_abort at /Users/ruki/projects/personal/esp-idf/components/esp_system/esp_system.c:137

0x4038945e: abort at /Users/ruki/projects/personal/esp-idf/components/newlib/abort.c:46

0x40381a4f: _esp_error_check_failed at /Users/ruki/projects/personal/esp-idf/components/esp_system/esp_err.c:43

0x42007f64: ot_network_wifi_softap_init at /Users/ruki/projects/personal/hellokiki/ottobot/main/network.c:62 (discriminator 1)

0x42008127: ot_network_init at /Users/ruki/projects/personal/hellokiki/ottobot/main/network.c:110

0x42007d72: ot_app_main at /Users/ruki/projects/personal/hellokiki/ottobot/main/app_main.c:100

0x42007c37: app_main at /Users/ruki/projects/personal/hellokiki/ottobot/main/main.c:40

0x420dc254: main_task at /Users/ruki/projects/personal/esp-idf/components/freertos/port/port_common.c:141 (discriminator 2)

0x40384e21: vPortTaskWrapper at /Users/ruki/projects/personal/esp-idf/components/freertos/port/xtensa/port.c:142

ELF file SHA256: 18e2cb83aaa77af8

```
```

igrr, 2023-03-29:
> factory, app,  factory, 0x010000, 2500k
> model,  data, spiffs,         , 5168K,
 
I think the error makes sense, you don't have any NVS partitions in the partition table.

Please try adding:
```
nvs, data, nvs, , 64k
```
(You may need to adjust the partition sizes to fit this.)

waruqi, 2023-03-29:
It works fine. thanks!

closed 2023-03-29 by waruqi