espressif/esp-idf #11074: Delay between Command and DMA transfer start in panel_io_i80_tx_color

Designer-Rick, 2023-03-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am developing code for ssd1963 through 8 bit parallel bus using the i2s in parallel mode.

The lcd is 800 x 480 and at 3 bytes per pixel I can get a 2.4 FSP throughput.  If it was just a bit faster it would be great.

Looking at the timings there is a 172uS delay from the color data being placed on the que and the DMA transmissing starting.  Is there any way to reduce this time.

[image]

suda-morris, 2023-03-30:
which ESP target? esp32  or esp32s2? The delay may come from `i2s_lcd_prepare_cmd_buffer` and `i2s_lcd_prepare_color_buffer` because we need to change the data format, by software.

Designer-Rick, 2023-03-30:
Yes that is exactly where the delay is, if I shorten the conversion loop the delay time decreases.
I am using an ESP32, is the ESP32s2 better?
I was looking to see if I could prepare the data before sending it especially for clear screen etc.
Is the parallel SPI a better option for DMA / Speed?
Thanks

suda-morris, 2023-03-30:
`SOC_I2S_TRANS_SIZE_ALIGN_WORD` is only defined on esp32, we use that to express the hardware limitation. ESP32S2 and ESP32 share the same source file "esp_lcd_panel_io_i2s.c", because `SOC_I2S_TRANS_SIZE_ALIGN_WORD` is not defined on esp32s2, so the "data prepare path" must be shorter than esp32. 

> Is the parallel SPI a better option for DMA / Speed?
> Will the ESP32s3 be quicker?

Yes, because we don't see such a long prepare loop in the driver. But there still be some small buffer (for LCD command, parameters) prepare, usually several bytes.

Designer-Rick, 2023-03-30:
That is ace, I have an ESP32s3 here I will rework the pre-production hardware for that and test it .  Thanks for your help.

closed 2023-03-30 by Designer-Rick