espressif/esp-idf #11075: Performing OTA-update while driving driving RGB-LCD causing glitches

DCSBL, 2023-03-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have a 800x480RGB display that is driven using [`esp_lcd` in RGB mode](https://github.com/espressif/esp-idf/blob/master/components/esp_lcd/include/esp_lcd_panel_rgb.h). When I perform an OTA-update the display is showing random glitches (shifting content at random places). I use external PSRAM for a full-frame buffer.

My goal is to have _something_ on screen during OTA, it does not have to change. So some content in a frozen state is fine.

**My question:** Is it possible to achieve this goal?

**I tried:**
  - Lower the PCLK speed with esp_lcd_rgb_panel_set_pclk() to the lowest speed the display accepts, comfirmed lower speed but no changes
  - De-init the driver, display turns off (as expected)
  - Fixed PCLK, DE or enable to see if display accepts that.

suda-morris, 2023-03-30:
According to https://github.com/espressif/esp-idf/tree/master/examples/peripherals/lcd/rgb_panel#troubleshooting, have you tried "Enable CONFIG_SPIRAM_FETCH_INSTRUCTIONS and CONFIG_SPIRAM_RODATA"?

suda-morris, 2023-03-30:
also hope this FAQ page can help you: https://docs.espressif.com/projects/espressif-esp-faq/en/latest/software-framework/peripherals/lcd.html#lcd

DCSBL, 2023-04-05:
Thank you @suda-morris. I've configured all the recommended options, but it seems that our display just can't accept a lower pclk speed. I will look for other options!

closed 2023-04-05 by DCSBL