espressif/esp-idf #11077: esp-x509-crt-bundle: Failed to verify certificate

kotyara12, 2023-03-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp-idf: 4.4.3 / 5.0.0 / 5.0.1

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

PlatformIO, package Espressif 32

### Power Supply used.

External 5V

### What is the expected behavior?

I'm trying to check the networking of a server for an HTTPS connection using esp_crt_bundle_attach.
Server: http://open-monitoring.online/
I've tried using both the built-in package and adding the certificates from the folder - to no avail.
If you transfer the certificate to the connection directly (via a buffer) or the global storage, there are no problems, the connection is established. Through a package of certificates - no.
sdkconfig.h
```
#define CONFIG_MBEDTLS_INTERNAL_MEM_ALLOC 1
#define CONFIG_MBEDTLS_ASYMMETRIC_CONTENT_LEN 1
#define CONFIG_MBEDTLS_SSL_IN_CONTENT_LEN 16384
#define CONFIG_MBEDTLS_SSL_OUT_CONTENT_LEN 4096
#define CONFIG_MBEDTLS_SSL_KEEP_PEER_CERTIFICATE 1
#define CONFIG_MBEDTLS_CERTIFICATE_BUNDLE 1
#define CONFIG_MBEDTLS_CERTIFICATE_BUNDLE_DEFAULT_NONE 1
#define CONFIG_MBEDTLS_CUSTOM_CERTIFICATE_BUNDLE 1
#define CONFIG_MBEDTLS_CUSTOM_CERTIFICATE_BUNDLE_PATH "certs"
#define CONFIG_MBEDTLS_CERTIFICATE_BUNDLE_MAX_CERTS 200
```

### What is the actual behavior?

Log:
```
I (528554) HTTP: Send HTTP request: https://open-monitoring.online/get?cid=9999&key=xxxxxxx&p1=296720&p2=62104&p3=60928&p4=TLS%20bundle&p5=5.0.1&p6=1&p7=0&p8=0&p9=0&p10=1&p11=7
E (528964) esp-x509-crt-bundle: Failed to verify certificate
E (528964) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x3000
E (528964) esp-tls: Failed to open new connection
E (528974) transport_base: Failed to open a new connection
E (528984) HTTP_CLIENT: Connection failed, sock < 0
```

### Steps to reproduce.

```
    esp_http_client_config_t request;
    memset(&request, 0, sizeof(request));
  
    request.method = HTTP_METHOD_GET;
    request.transport_type = HTTP_TRANSPORT_OVER_SSL;
    request.skip_cert_common_name_check = false;
    #if CONFIG_MBEDTLS_CERTIFICATE_BUNDLE
      request.crt_bundle_attach = esp_crt_bundle_attach;
    #else
```

### Debug Logs.

```plain
E (528964) esp-x509-crt-bundle: Failed to verify certificate
E (528964) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x3000
E (528964) esp-tls: Failed to open new connection
E (528974) transport_base: Failed to open a new connection
E (528984) HTTP_CLIENT: Connection failed, sock < 0
I (528984) HTTP: esp_http_client_get_status_code = 0
```

kotyara12, 2023-03-30:
Yes, it seems to get help from the developers is more difficult and longer than from aliens...
Hey, do you exist?

mahavirj, 2023-03-31:
@kotyara12 

- Can you please share your application code and sdkconfig here? I was not able to recreate the issue by modifying the URL in the `examples/protocols/https_request` IDF application. [Reference](https://github.com/espressif/esp-idf/blob/81e1e6599553a646a689ad51e32a5d48b34cfec5/examples/protocols/https_request/main/https_request_example_main.c#L174)
- The endpoint that you specified uses the ISRG Root X1 cross signed by DST Root CA X3 ([Reference](https://letsencrypt.org/certificates/)) and we already have this scenario covered in our examples with the test endpoint `https://howsmyssl.com` which uses same certificate. If there was any issue in the certificate bundle feature then the `examples/protocols/https_request` would have failed too.
- It is possible that the issue could be memory related, did you track free heap and minimum free heap in your application?

kotyara12, 2023-03-31:
@mahavirj

You can download my project from the [link](url)https://github.com/kotyara12/dzen/tree/master/https_requests
Now it is just configured to work with the bundle

Yesterday I dug into the sources of esp_crt_bundle, and indeed, I found that when searching for a certificate in the list, the search goes to DST Root CA X3, although in all other cases it is a normal ISRG Root X1
I believe that this is fundamentally wrong, since I, as a user, have not been aware of any cross-certificates until today. It is your job to use them correctly.
Moreover, in the buffer and global storage mode, the check is carried out correctly.

This project is prepared for an article that allows users to better understand the mechanisms of HTTPS, and I'm afraid I will be forced to write that the bundle does not work properly

Harshal5, 2023-03-31:
Hello @kotyara12 ,
I tried reproducing the scenario using the above-linked project for ESP-IDF `v5.0.1` but the certificate gets validated in my case.

Following are the logs for the same:
>I (9506) example_common: - IPv4 address: 192.168.20.202,
I (9516) example_common: - IPv6 address: fe80:0000:0000:0000:ea31:cdff:fec2:ffd0, type: ESP_IP6_ADDR_IS_LINK_LOCAL
I (74526) RAM: Heap total: 296696, usage: 59340
I (74526) HTTP: Send HTTP request: https://open-monitoring.online/get?cid=2481&key=rR3LH9&p1=296696&p2=59340&p3=59340&p4=TLS%20bundle&p5=v5.0.1&p6=1&p7=0&p8=0&p9=0&p10=0&p11=0
I (75846) esp-x509-crt-bundle: Certificate validated
I (77336) HTTP: esp_http_client_get_status_code = 200

mahavirj, 2023-03-31:
@kotyara12 

Most likely your application is somehow missing on including the [DST Root CA X3](https://github.com/espressif/esp-idf/blob/master/components/mbedtls/esp_crt_bundle/cacrt_local.pem) in the certificate bundle. This is handled by the build system [here](https://github.com/espressif/esp-idf/blob/81e1e6599553a646a689ad51e32a5d48b34cfec5/components/mbedtls/CMakeLists.txt#L42). 

- You may check if you had any local changes related to this part
- Also I would recommend trying full project cleanup and then rebuild once

kotyara12, 2023-03-31:
PlatformIO can get in the way?
In normal mode (via buffer), you have to specify all external files in PlatfromIO.ini too...
Although I added "manually" only one certificate to the bundle yesterday and it still didn't work

mr258876, 2023-03-31:
Same issue with [Open-Meteo](https://open-meteo.com/en/docs), using PlatformIO `v6.1.6` + esp-idf `v4.4.4` + arduino-as-component `v2.0.7`. Tried to move content in [DST Root CA X3](https://github.com/espressif/esp-idf/blob/master/components/mbedtls/esp_crt_bundle/cacrt_local.pem) into `cacrt_all.pem` but no luck.

```
I (15102) HTTP: GET https://api.open-meteo.com/v1/forecast?latitude=22.55&longitude=114.07&hourly=temperature_2m
D (16190) esp-x509-crt-bundle: 137 certificates in bundle
E (16190) esp-x509-crt-bundle: Failed to verify certificate
E (16192) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x3000
E (16198) esp-tls: Failed to open new connection
E (16203) TRANSPORT_BASE: Failed to open a new connection
E (16211) HTTP_CLIENT: Connection failed, sock < 0
```

kotyara12, 2023-04-01:
@mahavirj 

This solved the problem, thank you so much!

closed 2023-04-01 by kotyara12