espressif/esp-idf #11078: Start http server failed, SPIFFS: mount failed, -10025

waruqi, 2023-03-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.4

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-WROOM-1

### Power Supply used.

USB

### What is the expected behavior?

it works

### What is the actual behavior?

I tried get-started/http_server example, it works. But I ported to my project. it does not work.

```c
#define OT_HTTP_SERVER_WEB_MOUNT_POINT "/web"

static ot_bool_t ot_http_server_init_fs() {
    esp_vfs_spiffs_conf_t conf = {
        .base_path = OT_HTTP_SERVER_WEB_MOUNT_POINT,
        .partition_label = ot_null,
        .max_files = 50,
        .format_if_mount_failed = false
    };
    esp_err_t ret = esp_vfs_spiffs_register(&conf);
    if (ret != ESP_OK) {
        if (ret == ESP_FAIL) { <---  failed
            ot_trace_e("failed to mount or format filesystem");
        } else if (ret == ESP_ERR_NOT_FOUND) {
            ot_trace_e("failed to find SPIFFS partition");
        } else {
            ot_trace_e("failed to initialize SPIFFS (%s)", esp_err_to_name(ret));
        }
        return ot_false;
    }

    size_t total = 0, used = 0;
    ret = esp_spiffs_info(ot_null, &total, &used);
    if (ret != ESP_OK) {
        ot_trace_e("failed to get SPIFFS partition information (%s)", esp_err_to_name(ret));
    } else {
        ot_trace_i("partition size: total: %d, used: %d", total, used);
    }
    return ot_true;
}
```

I use custom partition.csv

```
# Espressif ESP32 Partition Table
# Name,  Type, SubType, Offset,  Size
factory, app,  factory, 0x010000, 2048k
model,   data, spiffs,          , 4096K,
nvs,     data, nvs,             , 64k,
web,     data, spiffs,          , 700k
```

```

set(WEB_SRC_DIR "${CMAKE_CURRENT_SOURCE_DIR}/../assets/web")
spiffs_create_partition_image(web ${WEB_SRC_DIR} FLASH_IN_PROJECT)
```

### Steps to reproduce.

```
idf.py build
idf.py flash
idf.py monitor
```

### Debug Logs.

```plain
I (1298) wifi:new:<6,0>, old:<1,0>, ap:<255,255>, sta:<6,0>, prof:1
I (1298) wifi:state: init -> auth (b0)
I (1308) wifi:state: auth -> assoc (0)
I (1318) wifi:state: assoc -> run (10)
I (1338) wifi:connected with ruki-v20, aid = 11, channel 6, BW20, bssid = c2:25:23:85:0d:f2
I (1348) wifi:security: WPA2-PSK, phy: bgn, rssi: -37
I (1348) wifi:pm start, type: 1

I (1348) wifi:set rx beacon pti, rx_bcn_pti: 0, bcn_timeout: 0, mt_pti: 25000, mt_time: 10000
I (1358) wifi:AP's beacon interval = 102400 us, DTIM period = 2
I (1858) wifi:state: run -> init (3c0)
I (1858) wifi:pm stop, total sleep time: 405880 us / 512947 us

I (1858) wifi:new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
I (1868) network: retry to connect to the AP
I (1868) network: connect to the AP fail
I (1868) wifi:new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
I (1878) wifi:state: init -> auth (b0)
I (1888) wifi:state: auth -> assoc (0)
I (1898) wifi:state: assoc -> run (10)
I (1918) wifi:connected with ruki-v20, aid = 6, channel 6, BW20, bssid = c2:25:23:85:0d:f2
I (1918) wifi:security: WPA2-PSK, phy: bgn, rssi: -37
I (1928) wifi:pm start, type: 1

I (1928) wifi:set rx beacon pti, rx_bcn_pti: 0, bcn_timeout: 0, mt_pti: 25000, mt_time: 10000
I (1938) wifi:state: run -> init (3c0)
I (1938) wifi:pm stop, total sleep time: 0 us / 11816 us

I (1938) wifi:new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
I (1948) network: retry to connect to the AP
I (1958) network: connect to the AP fail
I (1968) wifi:new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
I (2008) wifi:connected with ruki-v20, aid = 1, channel 6, BW20, bssid = c2:25:23:85:0d:f2
I (2008) wifi:security: WPA2-PSK, phy: bgn, rssi: -37
I (2018) wifi:pm start, type: 1

I (2018) wifi:set rx beacon pti, rx_bcn_pti: 0, bcn_timeout: 0, mt_pti: 25000, mt_time: 10000
I (2078) wifi:AP's beacon interval = 102400 us, DTIM period = 2
I (3028) esp_netif_handlers: sta ip: 192.168.43.110, mask: 255.255.255.0, gw: 192.168.43.1
I (3028) network: got ip:192.168.43.110
I (3028) network: connected to ap SSID: ruki-v20 password: 31415926
E (3038) SPIFFS: mount failed, -10025
E (3038) http_server: failed to mount or format filesystem
```
```

mahavirj, 2023-03-30:
@waruqi 

This error indicates an invalid file-system at the given flash offset and hence an error in mounting it. You may try erasing the complete flash first (`idf.py erase-flash). Alternatively, you may also modify the code:

```
.format_if_mount_failed = true
```

and see if that helps.

waruqi, 2023-03-30:
I tried it. format op is too slow and I cannot get any file from http server.

BTW, Why does it work when I am using the http server example without formatting

```
W (3275) SPIFFS: mount failed, -10025. formatting...
I (58925) http_server: partition size: total: 3848081, used: 0
I (58925) esp-rest: Starting HTTP Server
E (81595) esp-rest: Failed to open file : /www/index.html
W (81595) httpd_txrx: httpd_resp_send_err: 500 Internal Server Error - Failed to read existing file
W (81605) httpd_uri: httpd_uri: uri handler execution failed
```

waruqi, 2023-03-30:
> You may try erasing the complete flash first (`idf.py erase-flash).

I tried this, It does not work too.

waruqi, 2023-03-30:
When I set the partition_label explicitly, it worked.

```c
    esp_vfs_spiffs_conf_t conf = {
        .base_path = OT_HTTP_SERVER_WEB_MOUNT_POINT,
        .partition_label = "web",    <------- here
        .max_files = 50,
        .format_if_mount_failed = false
    };
```

But what I don't understand is why the http_server example works even though it is not set.

https://github.com/espressif/esp-idf/blob/81e1e6599553a646a689ad51e32a5d48b34cfec5/examples/protocols/http_server/restful_server/main/esp_rest_main.c#L98

igrr, 2023-03-30:
> But what I don't understand is why the http_server example works even though it is not set.

This is because in the http_server/restful_server example, there is only one SPIFFS partition. In your project, there are two:

```
model,   data, spiffs,          , 4096K,
...
web,     data, spiffs,          , 700k
```

If the name is not specified, we will mount the first SPIFFS partition in the partition table. So you need to specify the partition name explicitly if you have more than one.

waruqi, 2023-03-30:
Got it, thanks!

closed 2023-03-30 by waruqi