espressif/esp-idf #11080: ESP32S3 with LVGL

ArvinHou, 2023-03-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

IDF 5.0.1
LVGL 8.3

I have init sdcard finish,it works OK.
but when I show a png in sdcard ,It runs error. The log is bellow?how to fix it?

code:

```
uint8_t sd_mount(void)
{
    esp_err_t ret;

    esp_vfs_fat_sdmmc_mount_config_t mount_config = {

        .format_if_mount_failed = true,

        .max_files = 5,
        .allocation_unit_size = 1024
    };
    sdmmc_card_t *card;
    const char mount_point[] = MOUNT_POINT;
    ESP_LOGI(TAG, "Initializing SD card");

    ESP_LOGI(TAG, "Using SDMMC peripheral");
    sdmmc_host_t host = SDMMC_HOST_DEFAULT();
    host.max_freq_khz = SDMMC_FREQ_DEFAULT;

    sdmmc_slot_config_t slot_config = SDMMC_SLOT_CONFIG_DEFAULT();

    slot_config.width = 4;

    slot_config.clk = 13;
    slot_config.cmd = 14;
    slot_config.d0 = 12;

    slot_config.d1 = 11;
    slot_config.d2 = 47;
    slot_config.d3 = 21;

    // Enable internal pullups on enabled pins. The internal pullups
    // are insufficient however, please make sure 10k external pullups are
    // connected on the bus. This is for debug / example purpose only.
    slot_config.flags |= SDMMC_SLOT_FLAG_INTERNAL_PULLUP;

    ESP_LOGI(TAG, "Mounting filesystem");
    ret = esp_vfs_fat_sdmmc_mount(mount_point, &host, &slot_config, &mount_config, &card);

    if (ret != ESP_OK) {
        if (ret == ESP_FAIL) {
            ESP_LOGE(TAG, "Failed to mount filesystem. "
                     "If you want the card to be formatted, set the EXAMPLE_FORMAT_IF_MOUNT_FAILED menuconfig option.");
        } else {
            ESP_LOGE(TAG, "Failed to initialize the card (%s). "
                     "Make sure SD card lines have pull-up resistors in place.", esp_err_to_name(ret));
        }
        return -1;
    }
    ESP_LOGI(TAG, "Filesystem mounted");

    // Card has been initialized, print its properties
    sdmmc_card_print_info(stdout, card);

  return 0;
}

lv_obj_t * img;
img = lv_img_create(lv_scr_act());
lv_img_set_src( img,  "S:/root/logo.png");
lv_obj_align(img, LV_ALIGN_DEFAULT, 0, 0);
```

log:

```
[0;32mI (1111) sdcard: Initializing SD card[0m
[0;32mI (1115) sdcard: Using SDMMC peripheral[0m
[0;32mI (1120) sdcard: Mounting filesystem[0m
D (1124) sdmmc_periph: peripheral version 5342270a, hardware config 03c44c83[0m
D (1131) intr_alloc: Connected src 30 to int 17 (cpu 0)[0m
D (1136) sdmmc_periph: using GPIO13 as clk pin[0m
[0;32mI (1141) gpio: GPIO[13]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [0m
D (1150) sdmmc_periph: using GPIO14 as cmd pin[0m
[0;32mI (1155) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [0m
D (1164) sdmmc_periph: using GPIO12 as d0 pin[0m
[0;32mI (1169) gpio: GPIO[12]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [0m
D (1178) sdmmc_periph: using GPIO11 as d1 pin[0m
[0;32mI (1182) gpio: GPIO[11]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [0m
D (1192) sdmmc_periph: using GPIO47 as d2 pin[0m
[0;32mI (1196) gpio: GPIO[47]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [0m
[0;32mI (1206) gpio: GPIO[21]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
D (1215) sdmmc_periph: slot=1 host_div=10 card_div=20 freq=400kHz[0m
D (1221) sdmmc_periph: slot=1 width=1[0m
D (1226) sdmmc_req: process_command_response: error 0x107  (status=00000100)[0m
D (1252) sdmmc_sd: SDHC/SDXC card[0m
D (1253) sdmmc_req: process_command_response: error 0x107  (status=00000100)[0m
D (1253) sdmmc_io: sdmmc_init_io: io_send_op_cond (1) returned 0x107; not IO card[0m
D (1270) sdmmc_common: host_ocr=0x40ff8000 card_ocr=0xc0ff8000[0m
D (1270) sdmmc_common: sdmmc_card_init: host_ocr=40ff8000, card_ocr=c0ff8000[0m
D (1272) sdmmc_init: sdmmc_card_init: card type is SD[0m
D (1281) sdmmc_common: sdmmc_init_host_bus_width: using 4-bit bus[0m
D (1283) sdmmc_periph: using GPIO21 as d3 pin[0m
[0;32mI (1288) gpio: GPIO[21]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [0m
D (1297) sdmmc_periph: slot=1 width=4[0m
D (1302) sdmmc_common: sdmmc_init_host_frequency: using 20000 kHz bus frequency[0m
D (1308) sdmmc_periph: slot=1 host_div=8 card_div=0 freq=20000kHz[0m
D (1315) vfs_fat_sdmmc: using pdrv=0[0m
[0;32mI (1323) sdcard: Filesystem mounted[0m
Name: EZSD1
Type: SDHC/SDXC
Speed: 20 MHz
Size: 30145MB
CSD: ver=2, sector_size=512, capacity=61736960 read_bl_len=9
SSR: bus_width=4
[0;32mI (1336) main: Turn off LCD backlight[0m
[0;32mI (1339) gpio: GPIO[14]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
[0;32mI (1348) main: Initialize Intel 8080 bus
[0;32mI (1533) main: Turn on LCD backlight[0m
[0;32mI (1533) main: Initialize LVGL library[0m
[0;32mI (1534) main: buf1@0x3c0ac508, buf2@0x3c0d1d0c[0m
[0;32mI (1537) main: Register display driver to LVGL[0m
[0;32mI (1546) main: Install LVGL tick timer[0m
D (1550) sdmmc_req: process_command_response: error 0x108  (status=00000006)[0m
D (1553) sdmmc_req: process_command_response: error 0x108  (status=00000006)[0m
[0;31mE (1560) sdmmc_cmd: sdmmc_read_sectors_dma: sdmmc_send_cmd returned 0x108[0m
D (1568) sdmmc_cmd: sdmmc_read_sectors: error 0x108 writing block 32216+0[0m
[0;31mE (1574) diskio_sdmmc: sdmmc_read_blocks failed (264)[0m
D (1651) sdmmc_req: process_command_response: error 0x108  (status=0000020e)[0m
D (1651) sdmmc_req: process_command_response: error 0x108  (status=0000020e)[0m
[0;31mE (1655) sdmmc_cmd: sdmmc_read_sectors_dma: sdmmc_send_cmd returned 0x108[0m
D (1662) sdmmc_cmd: sdmmc_read_sectors: error 0x108 writing block 32216+0[0m
[0;31mE (1669) diskio_sdmmc: sdmmc_read_blocks failed (264)
```

ArvinHou, 2023-03-30:
@igrr  Thank you for reply.The connection is not long and all pin have 10k pull-up resistance .
I can read txt file success,but in lvgl still error.
[image]

ArvinHou, 2023-03-30:
[image]

ArvinHou, 2023-03-30:
[image]

igrr, 2023-03-30:
>  slot_config.cmd = 14;

and 

[image]

seem to be in conflict?

ArvinHou, 2023-03-31:
I carefully reconfigured the pins of sdmmc and lvgl, and the problem has been resolved. Thanks @igrr .This is the price of my carelessness.

closed 2023-03-31 by ArvinHou