espressif/esp-idf #11082: RMT LED stops working once TWAI is working

ramiws, 2023-03-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.0 stable 

### Operating System used.

macOS

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3 Devkit from espressif

### Power Supply used.

External 5V

### What is the expected behavior?

The RMT should still be driving the LED strip (30 LEDs)  regardless of TWAI status 

### What is the actual behavior?

LED turns off when the TWAI communication is active 

### Steps to reproduce.

1. Get RMT to drive a LED strip. 
2. Start TWAI communication.

igrr, 2023-03-30:
Hi @ramiws, thanks for reporting the issue. As it is, there is unfortunately not enough information for us to reproduce this problem. Could you please share the code you are running as well as describe the testing procedure? (For example, do any other devices have to be attached to the CAN bus? Or simply initializing CAN is enough?)

atanisoft, 2023-03-30:
@ramiws do you have any code that reproduces this? I'm currently using an ESP32-S3-WROOM-1-N8R2 module with both CAN and RMT active simultaneously without issues. I'm wondering if it is possibly an issue in how you are initializing the peripherals / drivers.

ramiws, 2023-04-01:
Hi, thank you for testing, I stripped down the code and issue seems to be that the gpio that is controlling the led on/off was floating and when there is CAN communication is somehow effect it. I set a pulldown to the pin and now it works.

closed 2023-04-01 by ramiws