espressif/esp-idf #11083: error parameter in spi_bus_initialize when using example connect

gl-chenxiaosheng, 2023-03-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

IDF Version: master(https://github.com/espressif/esp-idf/commit/ab63aaa4a24a05904da2862d627f3987ecbeafd0)

I use esp32s3 and eth module DM9051 to run example [ot_br](https://github.com/espressif/esp-idf/tree/ab63aaa4a24a05904da2862d627f3987ecbeafd0/examples/openthread/ot_br).
And it  couldn't work as expect, log as follow:
```shell
I (571) app_start: Starting scheduler on CPU0

I (575) app_start: Starting scheduler on CPU1

I (575) main_task: Started on CPU0

I (585) main_task: Calling app_main()

E (605) spi: spi_bus_initialize(774): invalid dma channel, chip only support spi dma channel auto-alloc

ESP_ERROR_CHECK failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x4200afc7

0x4200afc7: eth_start at /home/kelly/esp/esp-idf/examples/common_components/protocol_examples_common/eth_connect.c:125 (discriminator 1)

file: "/IDF/examples/common_components/protocol_examples_common/eth_connect.c" line 125

func: eth_start

expression: spi_bus_initialize(CONFIG_EXAMPLE_ETH_SPI_HOST, &buscfg, 1)

abort() was called at PC 0x4037f207 on core 0

```
So I modify the param in spi_bus_initialize, it work.
I hope this configuration can be fix and merge into the branch, thanks.

xieqinan, 2023-03-31:
OK, I will confirm the problem and fix it.

jjsch-dev, 2023-04-18:
Hi, I have the same problem with an esp32-s3 and a wiznet w5500.

`I (44) boot: ESP-IDF v5.1-dev-4167-gab63aaa4a2 2nd stage bootloader`
`I (45) boot: compile time Apr 18 2023 10:51:48`
`I (45) boot: chip revision: v0.1`
`I (49) qio_mode: Enabling QIO for flash chip WinBond`
`I (55) boot.esp32s3: Boot SPI Speed : 80MHz`

`E (1674) spi: spi_bus_initialize(764): invalid dma channel, chip only support spi dma channel auto-alloc
ESP_ERROR_CHECK failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x4200f4be
0x4200f4be: eth_start at /home/jjsch/esp/esp-idf/examples/common_components/protocol_examples_common/eth_connect.c:125 (discriminator 1)`

I also fixed the problem by changing the code of the function [eth_start](https://github.com/espressif/esp-idf/blob/ef64e4e5b3fb4878ed7df62a054f44ae5b131519/examples/common_components/protocol_examples_common/eth_connect.c#L125) in the eth_connect.c file.

 ```C 
Fail
ESP_ERROR_CHECK(spi_bus_initialize(CONFIG_EXAMPLE_ETH_SPI_HOST, &buscfg, 1));

Success
ESP_ERROR_CHECK(spi_bus_initialize(CONFIG_EXAMPLE_ETH_SPI_HOST, &buscfg, SPI_DMA_CH_AUTO));
`

xieqinan, 2023-04-19:
The error has already been fixed, and it will be synced to [esp-idf](https://github.com/espressif/esp-idf) today or tomorrow.

jjsch-dev, 2023-04-19:
@xieqinan thanks for reply, when you make the change in the master, I update the IDF and compile my project to confirm that the problem is fixed.

closed 2023-04-20 by espressif-bot

jjsch-dev, 2023-04-20:
@xieqinan I did the pull to the master and tried it with my motherboard that has a PHY W5500 and it works fine, thank you very much for the work.

xieqinan, 2023-04-21:
@jjsch-dev Happy to help you.