espressif/esp-idf #11087: [mesh] Mesh component triggering multiple times the same event

KonssnoK, 2023-03-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Using modified code from
https://github.com/KonssnoK/esp-idf/tree/bug/mesh_restart/examples/mesh/ip_internal_network

- switch on the device with hotspot on
- device will connect
- switch off hotspot
- wait for a while
- press key button
- switch on hotspot

the device will most of the times reconnect, but the number of reported events will increase.

Probably some callback not correctly de-initialized ? 

```
W (14:42:10.826) mesh_main: Start checking DYNAMIC ROOT delayed tasks
I (913823) mesh: [RXQ]<max:32 = cfg:32 + extra:0>self:0, <max:32 = cfg:32 + extra:0>tods:0
W (14:42:10.844) mesh_main: Tell children we have connection
I (913843) mesh: <MESH_NWK_ROOT_TODS_STATE>toDS:1
I (14:42:10.858) mesh_main: <MESH_EVENT_PARENT_CONNECTED>layer:1-->1, parent:d6:92:7d:c3:02:1e<ROOT>, ID:77:77:77:77:77:76
I (14:42:10.874) mesh_netif: Already wifi station, no need to do anything
W (14:42:10.882) mesh_main: Tell children we have connection
I (14:42:10.888) mesh_main: <MESH_EVENT_PARENT_CONNECTED>layer:1-->1, parent:d6:92:7d:c3:02:1e<ROOT>, ID:77:77:77:77:77:76
I (14:42:10.900) mesh_netif: Already wifi station, no need to do anything
W (14:42:10.907) mesh_main: Tell children we have connection
I (14:42:10.914) mesh_main: <MESH_EVENT_PARENT_CONNECTED>layer:1-->1, parent:d6:92:7d:c3:02:1e<ROOT>, ID:77:77:77:77:77:76
I (14:42:10.925) mesh_netif: Already wifi station, no need to do anything
W (14:42:10.933) mesh_main: Tell children we have connection
W (14:42:10.939) mesh_main: <MESH_EVENT_TODS_REACHABLE>state:0
W (14:42:10.946) mesh_main: Forcing MQTT disconnection to avoid packet loss
I (14:42:10.953) MQTT_CLIENT: Client asked to disconnect
W (14:42:10.959) mesh_main: <MESH_EVENT_TODS_REACHABLE>state:0
W (14:42:10.966) mesh_main: Forcing MQTT disconnection to avoid packet loss
I (14:42:10.973) MQTT_CLIENT: Client asked to disconnect
W (14:42:10.979) mesh_main: <MESH_EVENT_TODS_REACHABLE>state:0
W (14:42:10.986) mesh_main: Forcing MQTT disconnection to avoid packet loss
I (14:42:10.994) MQTT_CLIENT: Client asked to disconnect
W (14:42:10.999) mesh_main: <MESH_EVENT_TODS_REACHABLE>state:0
W (14:42:11.006) mesh_main: Forcing MQTT disconnection to avoid packet loss
I (14:42:11.014) MQTT_CLIENT: Client asked to disconnect
```

KonssnoK, 2023-03-31:
This feels a lot like https://github.com/espressif/esp-idf/issues/11020
where the issue was that the dhcps stop function was not called before releasing the netif.
This made the UDP pcbs not to be updated, so they kept getting triggered
if there is an equivalent in the mesh, that's probably the issue.

KonssnoK, 2023-03-31:
we also tried adding `esp_mesh_deinit` after `esp_mesh_stop` but no change

zhangyanjiaoesp, 2023-04-11:
@KonssnoK 
after changing the code as follows, " triggering multiple times the same event " issue can be solved.
[image]

KonssnoK, 2023-04-12:
@zhangyanjiaoesp thanks, i didn't think about that.

closed 2023-04-12 by KonssnoK