espressif/esp-idf #11088: Unable to notify/indicate data of more than 20 bytes

TheodoreBardy, 2023-03-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.1

### Operating System used.

Linux

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32-devkitc

### Power Supply used.

USB

### What is the expected behavior?

I am developing a BLE service on my ESP32, the ESP32 is running as a server and the client is a mobile application. This service needs to notify data longer than 20 bytes, to do this I use the `esp_ble_gatts_send_indicate` function in the ESP-IDF Bluedroid stack.
The warning message `BT_GATT: attribute value too long, to be truncated to 20`, is printed and the exchange has received the first 20 bytes of the message.

1. I know that only the client can send an MTU request but I have no control over the central part (since it is an official mobile application).
2. It is impossible to notify more than 20 bytes in several packets because no offset is available in the ESP-IDF API. Each packet is therefore considered as a new and complete message by the client.

I also know that the client has an MTU size greater than 23 because the same mobile application works fine with an nRF52 as a server.
So I wonder if it is possible to force the ESP-IDF BLE stack not to truncate the data?

### What is the actual behavior?

.

### Steps to reproduce.

.

xiewenxiang, 2023-04-07:
If you want to exchange MTU size when ESP32 is the server role, you need to call the API `esp_err_t esp_ble_gattc_open(esp_gatt_if_t gattc_if, esp_bd_addr_t remote_bda, esp_ble_addr_type_t remote_addr_type, bool is_direct)` in ESP32 side after the connection is established. At this point you can initiate MTU updates on ESP32.

GTCLive, 2023-05-11:
@xiewenxiang 
I'm experiencing the same behavior with the "**gatt_server_service_table**" example. I can the update and read the characteristic buffer up to mtu's 500 from the server side without any problem, however if esp_ble_gatts_send_indicate() is used to achieve the same, so to notify the connected client, the error "_BT_GATT: attribute value too long, to be truncated to 20_" is logged. 

I'm unsure if this is a bug or as per design. I'd be grateful if you could confirm this so I can either report this or look for a way around this limitation. 

ps: the advised "esp_ble_gatts_open()" did not work in my case. 

Thanks.

zhp0406, 2023-08-24:
The size of indicate_data[] should be less than the MTU size. The esp_ble_gattc_send_mtu_req API can be used to modify the MTU.

closed 2023-10-25 by Alvin1Zhang