espressif/esp-idf #11089: RF Test Tool - Certification binaries - UART0 Not accesible

MiguelBailonS, 2023-03-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello

I have the following issue. 
We want to reproduce the WiFi/BLE test described in your documentation of ESP_RF_Test_EN. Unfortunately, we dont have access to the UART0 interface while our product is in its normal operation. As your documentation states that trough UART0 the commands are sent. Is it possible to generate a binary file to use a different UART configuration to send/recieve the commands and results? e.g. UART1 - GPIO 15(TX) / GPIO 14(RX)

Please, Let me know if its possible.

igrr, 2023-03-30:
Hi @MiguelBailonS,

RF certification binaries are not maintained as part of the esp-idf project. Please use the technical support form on Espressif website for this problem instead of this issue tracker.

In the upcoming IDF v5.1, the certification test API will be supported in IDF as well, so it will be possible to integrate this functionality into a custom application. This feature is only available on esp32-c3 and esp32-s3 for the time being.
https://github.com/espressif/esp-idf/tree/master/examples/phy/cert_test

MiguelBailonS, 2023-03-30:
Understood, thank you!

closed 2023-03-30 by MiguelBailonS