espressif/esp-idf #11090: are there examples or libraries about ota with tcp?

Inverter-Jo, 2023-03-31:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Now, i can apply ota with https on my tcp comm code.
but it dosen't look like clean code.

i want to apply ota with tls(tcp_communication) code on my tcp communication code.
is there any ota example with tls?

please let me know this.

CC-MASTER05, 2023-03-31:
check rainmaker application they have OTA facility

Inverter-Jo, 2023-04-03:
@CC-MASTER05 
Thank you!

when i checked it, they used https communication to do ota.
are you mean that i should replace esp_https_ota_begin() code with tcp(tls) code?

is there any ota example with tcp(tls) communcation you know?

if i replace that code, it would be long time to finish that.

hmalpani, 2023-04-05:
Hello @Inverter-Jo 
There ain't any OTA example that uses TLS api's directly for connection to perform OTA updates. Using API from the `esp_https_ota` component won't help your case to use `esp-tls` APIs. You can check [Native OTA example](https://github.com/espressif/esp-idf/tree/master/examples/system/ota/native_ota_example) which can help you use APIs from the `app_update` component and [https_request example](https://github.com/espressif/esp-idf/tree/master/examples/protocols/https_request) for using `esp_tls` component for performing connection with the server.

Inverter-Jo, 2023-04-12:
@hmalpani 

Thanks for giving a tip to me.
But i want to change communication for OTA.
for example, tcp communication would be that communication instead of https communication.
So i try to replace this like below flow diagram.

[image]

but in the esp_ota_end code, i got a err code with esp_image_verify() code.

[image]

[image]

[image]

what should i do?

hmalpani, 2023-04-12:
Hello @Inverter-Jo 
I think you are getting this error because of writing the wrong image in the partition. Can you check if the length of the image you are getting at the start of the process is the same as the actual OTA image hosted on the server? Writing the wrong data in the partition will result in this error. Please also check if the data you are writing in the partition is correct.

Inverter-Jo, 2023-04-12:
@hmalpani 

thanks!

i checked that length of the image is same as actual OTA image file.
but there is concerned thing that i transfer bin file with fread();
it should be fault?

hmalpani, 2023-04-12:
Hello @Inverter-Jo 
I recommend you check the TCP examples present in esp-idf for reference. I am not sure if `fread()` could be the cause of the problem. Can you share the code to debug the issue?

Inverter-Jo, 2023-04-13:
@hmalpani 

thanks!

i already checked that tcp communcation operate well!

and It's hard to share the full code. but here is code that cause this problem.

The problem occurs in the second loop in the for code of the code that named process_segments(in esp_ota_end -> esp_image_verify -> image_load -> process_segments)

verify_segment_header in process_segments function returned below log.
[image]

Inverter-Jo, 2023-04-14:
@hmalpani 

thanks for your answer

i checked that is my fault with that esptool.py read_flash
because i opened file with code called fopen("name_bin","r") on server, that makes wrong image of the file.

and then i changed that and finished my project.

thanks for your advise.

closed 2023-04-14 by hmalpani