espressif/esp-idf #11091: RGB_LCD 800*480 bug

fvstory, 2023-03-31:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4

### Operating System used.

Linux

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32_S3_spiram_oct

### Power Supply used.

USB

### What is the expected behavior?

Show the right full screen color.

### What is the actual behavior?

The left area(x:0-400) is strange, like the drift.
But the right area(x:400-800) is perfect running.
I make sure color data is right. 
When i only show the area(x:380-800), the area(x:380-400) is also strange.
I make sure RGB_LCD is no problem. (the merchant tell me it's the esp-idf's code's problem, they have the end product)
I try to fix it, but I can't make it.

### Steps to reproduce.

1.just use the rgb_lcd to bitmap the full screen color
2.it's oct spiram, also make the data in psram.

suda-morris, 2023-03-31:
1. you didn't tell how you set up the driver configuration, you know, there're some parameters that are specific to your LCD model, like HFP, VBP..
2. Have you tried your example on the master branch? If there's really a bug in the driver, we want this information to guide our backport process.
3. The code to help us reproduce your issue, and detailed sdkconfig (especially how you configure your PSRAM)

fvstory, 2023-03-31:
1.I use the v4.4, I don't try the master, because I use the micropython. I have make the lvgl_micropython_i80 perfect run.I github the code. I am testing the RGB.
2.the v4.4 LCD not many configuration, the spiram configuration i have cheeked.
3.Put it another way, the x_start=400 ,x_end=800, y_start=0, y_end=480 is OK. The area in between is also OK.
4.databus is RBG16, it show right color. 
5.sometime, I think it maybe the hal set problem?
6.it appears that the right area is a normal RGB, it divide neatly at the x=400

fvstory, 2023-04-01:
1.the lcd_drv_ic is made of two ic，so show strange.
2.I find the ic_pdf to make the configuration suit！
3.the v4.4’s rgb_lcd is not suitable for lvgl.
4.I will try to copy the v5.0 to adjust the code.

closed 2023-04-01 by fvstory