espressif/esp-idf #11092: auto-reconnect problem with bt a2dp

hadheemi, 2023-03-31:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

hello ,I have a question regarding Bluetooth Classis pairing with A2DP profile support.
Currently we encounter a lot of discrepancy depending on the device that connects, especially with regard to auto connection when the device has already been connected.
With some devices there is a connection automatically without re-entering the PIN code, on the other hand for others we always have to "forget" the bluetooth network of the esp32 to be able to reconnect to it, do you have any leads concerning this issue ? thank you in advance

amaroudis-hatch, 2023-06-04:
@hodhm Did you ever find a resolution for this ? I have encountered something similar here.  What are your arguments for the `esp_bt_gap_set_scan_mode()` call ?

xiongweichao, 2024-02-22:
Hi @hodhm @amaroudis-hatch ， Sorry for replying you so late. Could you please tell me if this issue still exists?

Sherry616, 2024-04-02:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more updates.

closed 2024-04-02 by Sherry616