espressif/esp-idf #11093: Interrupt based mcpwm example

CC-MASTER05, 2023-03-31:
### Is your feature request related to a problem?

I'm frustrated when i interface my ac fan using zero-cross circuit once i add intrupt pin in gpio it start continues rebooting and once i try diffrent way to interface it didn't generate any pwm signal.

As i have tested many way and still i am in truble to find proper example for mcpwm library example those have any kind of the intrup from ISRAM.

I have faced Timer Issue, Memory Issue many time when i try to make example.

### Describe the solution you'd like.

Need to generate PWM as per interrupt signal.

### Describe alternatives you've considered.

As we have already one another library ledc but i have found this not able to handle high frequency and that's why we need to use mcpwm library so, as per ledc library we have a fading example same we need want in mcpwm library also

CC-MASTER05, 2023-03-31:
Hi  @suda-morris can you please add this kind of example of can you guide me so i can write example send it to you

suda-morris, 2023-04-03:
I can't seem to understand your request. MCPWM can generate many interrupts, like "capture interrupt", "comparator interrupt", "brake interrupt", "fault interrupt", etc. What kind of interrupt event do you want in your application? And how frequently the interrupt is expected to be fired in your app?

CC-MASTER05, 2023-04-03:
Falling Edge and Rising Edge base interrupt we get this kind of the interrupt using zero-cross detect circuit or optocouple circuits.

closed 2023-05-17 by CC-MASTER05