espressif/esp-idf #11094: Device gets panic when continous polling via Modbus TCP

YaswanthKumarSaravanaKumar, 2023-03-31:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.1

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP-32Wrover B|custom board

### Power Supply used.

USB

### What is the expected behavior?

Hi team actually we have been running the code to poll data for the desired configuration at specific intervals via modbus tcp , and also parallel the wifi AP+STA mode , STA mode used for network connection and also we use ap mode to configure some of the parameters via web server mode. the polling via modbus tcp is conducted at continuous 1sec intervals in long run device gets panic . I also track it down via core dump not able to find what is the root cause for this issue.

### What is the actual behavior?

what is actually behavior expecting is continuous polling via modbus tcp at a 1sec interval without the device get panic. 
And we use modbus rtu to read the certain register and send those values to the web server. can you help in figuring out this issue?

### Steps to reproduce.

1. Initialize AP+STA mode to connect with a desired network.
2. use modbus tcp to collect data from register and keeps polling at 1sec rate of interval.

My core dump log goes like :

[ERR ] [rooftopd_control.c:570      ] Unable to read inputs, skipping control loop
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40093fe8  PS      : 0x00060a33  A0      : 0x80094a34  A1      : 0x3ffedf80
0x40093fe8: remove_free_block at C:/Espressif/frameworks/esp-idf-v4.4.1/components/heap/heap_tlsf.c:214
 (inlined by) block_locate_free at C:/Espressif/frameworks/esp-idf-v4.4.1/components/heap/heap_tlsf.c:448
 (inlined by) tlsf_malloc at C:/Espressif/frameworks/esp-idf-v4.4.1/components/heap/heap_tlsf.c:757

A2      : 0x3ffe4364  A3      : 0x00000108  A4      : 0x3ffb3d08  A5      : 0x00060a23
A6      : 0x00060a20  A7      : 0x00000001  A8      : 0x3fff80c4  A9      : 0x0000001b
A10     : 0x83fe0300  A11     : 0x00000004  A12     : 0x3ffe4364  A13     : 0x3ffd0d50
A14     : 0x007aefcc  A15     : 0x003fffff  SAR     : 0x0000001d  EXCCAUSE: 0x0000001d
EXCVADDR: 0x83fe0308  LBEG    : 0x4008bb31  LEND    : 0x4008bb41  LCOUNT  : 0xfffffffc
0x4008bb31: strlen at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/strlen.S:84

0x4008bb41: strlen at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/strlen.S:96

Backtrace:0x40093fe5:0x3ffedf800x40094a31:0x3ffedfa0 0x40081ba1:0x3ffedfc0 0x40081bc9:0x3ffedfe0 0x40081bf5:0x3ffee000 0x400961c6:0x3ffee020 0x400961e8:0x3ffee040 0x400ef3c0:0x3ffee060 0x4009251a:0x3ffee0a0
0x40093fe5: remove_free_block at C:/Espressif/frameworks/esp-idf-v4.4.1/components/heap/heap_tlsf.c:212
 (inlined by) block_locate_free at C:/Espressif/frameworks/esp-idf-v4.4.1/components/heap/heap_tlsf.c:448
 (inlined by) tlsf_malloc at C:/Espressif/frameworks/esp-idf-v4.4.1/components/heap/heap_tlsf.c:757

0x40094a31: multi_heap_malloc_impl at C:/Espressif/frameworks/esp-idf-v4.4.1/components/heap/multi_heap.c:197
[... 470 lines cut ...]
[INFO] [network_api.c:184           ] Wifi Started
[PRI ] [network_api.c:117           ] Ethernet Stopped
[PRI ] [network_api.c:454           ] - TIME2: 72000000000

[INFO] [device.c:229                ] WifiSTA MAC:  a8:48:fa:87:9c:f0
[INFO] [device.c:235                ] SoftAP MAC:   a8:48:fa:87:9c:f1
[INFO] [device.c:241                ] BT MAC:       a8:48:fa:87:9c:f2
[INFO] [device.c:247                ] Ethernet MAC: a8:48:fa:87:9c:f3
[INFO] [device.c:252                ] Netif MAC:    a8:48:fa:87:9c:f0
[INFO] [rooftopd.c:357              ] (re)init fan operation mode 2 [= 0]
[PRI ] [rooftopd_fan_config.c:62    ] -> Config fan 001[INFO] [network_api.c:191           ] Wifi station is connected

[ERR ] [rooftopd_fan_config.c:69    ] modbus_rtu_master_write_holding_register() failed slave: 1
[PRI ] [network_api.c:255           ] ************************ STA Got IP Address
[PRI ] [network_api.c:256           ] ************************ STAIP:  192.168.137.149
[PRI ] [network_api.c:257           ] ************************ STAMASK:255.255.255.0
[PRI ] [network_api.c:258           ] ************************ STAGW:  192.168.137.1
[PRI ] [network_api.c:259           ] ************************ DNS:  192.168.137.1
[ERR ] [rooftopd_fan_config.c:69    ] modbus_rtu_master_write_holding_register() failed slave: 1
[ERR ] [rooftopd_fan_config.c:69    ] modbus_rtu_master_write_holding_register() failed slave: 1
[ERR ] [rooftopd_fan_config.c:69    ] modbus_rtu_master_write_holding_register() failed slave: 1
[ERR ] [rooftopd_fan_config.c:69    ] modbus_rtu_master_write_holding_register() failed slave: 1
[INFO] [rooftopd.c:174              ] Operating mode 1: 1
[INFO] [rooftopd.c:177              ] Presure:       min-max [0-0] step 0
[INFO] [rooftopd.c:179              ] Factor:        p,i,k [0, 1, 0]

YaswanthKumarSaravanaKumar, 2023-04-03:
Hi,  Team is there any update on this topic?

alisitsyn, 2023-04-03:
Hi @YaswanthKumarSaravanaKumar,

Thank you for the report. Could you provide more information about your project?

- Please send your sdkconfig file.
- The whole project communication log including Modbus messages (set log verbosity to debug in kconfig).
- What kind of Modbus TCP master you are using to poll for your Modbus TCP slave?
- Setup code for the AP and STA mode and Modbus initialization for RTU master and TCP slave (netif configurations). The shared project would be better.
- You are using the old freemodbus component from esp-idf. In order to proceed further I recommend to change the code to use esp-modbus v1.0.10 component instead. This can be accomplished by adding the line `set(EXCLUDE_COMPONENTS freemodbus)` to your [CMakeLists.txt file (an example)](https://github.com/alisitsyn/modbus_support/blob/modbus_support/fix_glitches_esp_modbus_master_rtu_ascii/modbus_master_rtu/CMakeLists.txt#L6), then add the [component manifest file](https://github.com/espressif/esp-idf/blob/release/v5.0/examples/protocols/modbus/tcp/mb_tcp_slave/main/idf_component.yml) to the main folder of your project. Once completed please clean the build and rebuild your project again.

The issue happens when Modbus Slave server task tries to accept connection from your Master and then error happens in `xMBTCPPortAcceptConnection()`. This should be indicated by `Fail to accept connection for client #` error message in your log.

Please follow the steps above to proceed. Thank you.

sureshm168, 2023-04-03:
Thanks Guys for your feedback. We'll update you as soon as possible.

YaswanthKumarSaravanaKumar, 2023-09-27:
Hi @alisitsyn  It's been long time ,The actually the issue is from our side , we are using the modified rtu and tcp library not the updated one , after removing the modbus tcp to rtu bridge everything works fine.

YaswanthKumarSaravanaKumar, 2023-09-27:
@alisitsyn We are currently working on implementing the modbus tcp to rtu bridge is there any api function or source code available from the esp end .

alisitsyn, 2024-03-05:
@YaswanthKumarSaravanaKumar , thank you for the respond and sorry for the delays with answer. Unfortunately currently I can not provide the complete and tested example of the bridge for now. I added it to my plan to provide the simple way for the bridge functionality in the stack and examples.

YaswanthKumarSaravanaKumar, 2024-03-15:
Thanks for the update @alisitsyn if there is any update please let me know , and if the poc has been completed please share the link with us .

alisitsyn, 2025-11-25:
The issue is closed due to lack of feedback. The esp-modbus v1.0.18 is released since that time. The functionality with the similar scenario is confirmed by the test.