espressif/esp-idf #11097: WebSocket add client when REST server retrieves files

law-ko, 2023-03-31:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Retrieving files with REST server along with WebSocket running should not open a new web socket connection.

### What is the actual behavior?

```
I (52400) WS_REST_SERVER: New client connected 56
I (52740) WS_REST_SERVER: File sending complete
I (52750) WS_REST_SERVER: New client connected 57
I (52900) WS_REST_SERVER: File sending complete
I (52910) WS_REST_SERVER: New client connected 58
I (52930) WS_REST_SERVER: File sending complete
I (52930) WS_REST_SERVER: New client connected 59
I (53010) WS_REST_SERVER: File sending complete
I (53030) WS_REST_SERVER: File sending complete
I (53100) WS_REST_SERVER: File sending complete
I (53130) WS_REST_SERVER: File sending complete
I (53400) WS_REST_SERVER: File sending complete

// after complete sending file
E (63330) wss_keep_alive: Client (fd=56) not alive!
E (63330) WS_REST_SERVER: Client not alive, closing fd 56
I (63330) WS_REST_SERVER: Client disconnected 56
I (63340) WS_REST_SERVER: New client connected 56
I (63350) WS_REST_SERVER: Handshake done, the new connection was opened
E (63360) wss_keep_alive: Client (fd=57) not alive!
E (63360) WS_REST_SERVER: Client not alive, closing fd 57
I (63360) WS_REST_SERVER: Client disconnected 57
E (63380) wss_keep_alive: Client (fd=58) not alive!
E (63380) WS_REST_SERVER: Client not alive, closing fd 58
I (63390) WS_REST_SERVER: Client disconnected 58
I (63410) WS_REST_SERVER: ws_handler: httpd_handle_t=0x3fff7cb0, sockfd=56, client_info:2
E (63420) wss_keep_alive: Client (fd=59) not alive!
E (63420) WS_REST_SERVER: Client not alive, closing fd 59

E (64570) wss_keep_alive: Client (fd=57) not alive!
E (64570) WS_REST_SERVER: Client not alive, closing fd 57
E (64570) wss_keep_alive: Client (fd=58) not alive!
E (64580) WS_REST_SERVER: Client not alive, closing fd 58
E (64590) wss_keep_alive: Client (fd=59) not alive!
E (64590) WS_REST_SERVER: Client not alive, closing fd 59
E (65590) wss_keep_alive: Client (fd=57) not alive!
E (65590) WS_REST_SERVER: Client not alive, closing fd 57
E (65590) wss_keep_alive: Client (fd=58) not alive!
E (65590) WS_REST_SERVER: Client not alive, closing fd 58
E (65600) wss_keep_alive: Client (fd=59) not alive!
E (65610) WS_REST_SERVER: Client not alive, closing fd 59
```

### Steps to reproduce.

```
// Start the httpd server
    httpd_handle_t server = NULL;
    ESP_LOGI(TAG, "Starting server");

	httpd_config_t conf = HTTPD_DEFAULT_CONFIG();
    conf.max_open_sockets = max_clients;
    conf.global_user_ctx = keep_alive;
    conf.open_fn = wss_open_fd;
    conf.close_fn = wss_close_fd;
	conf.uri_match_fn = httpd_uri_match_wildcard; // REST server

static const httpd_uri_t ws = {
        .uri        = "/ws",
        .method     = HTTP_GET,
        .handler    = ws_handler,
        .user_ctx   = NULL,
        .is_websocket = true,
        .handle_ws_control_frames = true
};

// Set URI handlers
    ESP_LOGI(TAG, "Registering URI handlers");
    httpd_register_uri_handler(server, &ws);

	/* URI handler for getting web server files */
    httpd_uri_t common_get_uri = {
        .uri = "/*",
        .method = HTTP_GET,
        .handler = rest_common_get_handler,
        .user_ctx = rest_context
    };
    httpd_register_uri_handler(server, &common_get_uri);

    wss_keep_alive_set_user_ctx(keep_alive, server);
```

Should add a check uri if statement to ensure it is actual ws connection:
```
esp_err_t wss_open_fd(httpd_handle_t hd, int sockfd)
{
    ESP_LOGI(TAG, "New client connected %d", sockfd);
    wss_keep_alive_t h = httpd_get_global_user_ctx(hd);
	// /* check uri */
	// /* if uri is /ws, then add to wsConnectionCount */
	// if (strcmp(hd->uri, "/ws") == 0) {
	// 	ESP_LOGW(TAG, "This is /ws");
	// }
	// else
	// {
	// 	ESP_LOGE(TAG, "This is not /ws");
	// }
    return wss_keep_alive_add_client(h, sockfd);
}
```

### More Information.

If `conf.open_fn` can handle what the URI is then it can be resolved by adding an if statement to check whether the URI is intended for a web socket connection.

law-ko, 2023-04-17:
Any idea how to resolve this?

ESP-YJM, 2023-04-19:
@law-ko It is difficult to get `uri` in `conf.open_fn`. Because in `conf.open_fn` callback execute, at this time, only the TCP handshake connection has been completed, and the http application layer has not yet been reached, so at this time, the link uri is not available.

law-ko, 2023-04-19:
@ESP-YJM , thank you for the information. In this case, we are combining WS and REST server together, which caused this issue. I believe this is a pretty common use case, is there any way to resolve REST server opening multiple WS connection when receiving REST file and contents?

ESP-YJM, 2023-04-20:
@law-ko I think when you new connection connect to HTTP server, it will call `wss_open_fd`, and it not mean opening multiple WS connection. It maybe still REST connection. The WS is a long connection, if the client not disconnect, it will keep the connection.

law-ko, 2023-04-20:
@ESP-YJM In my case it will open multiple WS connection and it won't be able to be automatically closed overtime.

Below is the error we get when the http receives REST server files.
```
E (64570) wss_keep_alive: Client (fd=57) not alive!
E (64570) WS_REST_SERVER: Client not alive, closing fd 57
E (64570) wss_keep_alive: Client (fd=58) not alive!
E (64580) WS_REST_SERVER: Client not alive, closing fd 58
E (64590) wss_keep_alive: Client (fd=59) not alive!
E (64590) WS_REST_SERVER: Client not alive, closing fd 59
E (65590) wss_keep_alive: Client (fd=57) not alive!
E (65590) WS_REST_SERVER: Client not alive, closing fd 57
E (65590) wss_keep_alive: Client (fd=58) not alive!
E (65590) WS_REST_SERVER: Client not alive, closing fd 58
E (65600) wss_keep_alive: Client (fd=59) not alive!
E (65610) WS_REST_SERVER: Client not alive, closing fd 59
```

ESP-YJM, 2023-04-20:
@law-ko I think you can first remove `wss_keep_alive_add_client` in `wss_open_fd`. Because keep-alive is for wss long-time connection and not for REST. If you register the ws and REST server, it will do keep-alive for both. So it will confuse you.

law-ko, 2023-04-22:
@ESP-YJM Commenting out `wss_keep_alive_add_client` does not seem to resolve the issue but making it worse. Now if the REST server attempts to receive multiple files at the same time, ESP32 will not accept to any new WS connection. Using Postman will show connecting...

Here is the code:
```
esp_err_t wss_open_fd(httpd_handle_t hd, int sockfd)
{
    ESP_LOGW(TAG, "New client connected %d", sockfd);
    //wss_keep_alive_t h = httpd_get_global_user_ctx(hd);
    //return wss_keep_alive_add_client(h, sockfd);
	return ESP_OK;
}

void wss_close_fd(httpd_handle_t hd, int sockfd)
{
    ESP_LOGW(TAG, "Client disconnected %d", sockfd);
    //wss_keep_alive_t h = httpd_get_global_user_ctx(hd);
    //wss_keep_alive_remove_client(h, sockfd);

    /* Fixes `httpd_accept_conn: error in accept (23)` - https://github.com/espressif/esp-idf/issues/9395 */
    struct linger so_linger;
    so_linger.l_onoff = true;
    so_linger.l_linger = 0;
    lwip_setsockopt(sockfd, SOL_SOCKET, SO_LINGER, &so_linger, sizeof(so_linger));
    close(sockfd);
}
```

ESP-YJM, 2023-04-23:
@law-ko `wss_keep_alive_add_client` is for ws connection and not suitable for REST connection. When you not commenting out `wss_keep_alive_add_client`, if the  REST connection coming in, i will be closed after about 10s by websocket keep-alive soltion. So you can't add `wss_keep_alive_add_client` in wss_open_fd, which will interrupt REST connection. And now you commenting out `wss_keep_alive_add_client`, it be worse now. Because REST client may create a lot zombie TCP connection, you need add TCP keep-alive in `wss_open_fd` to replace `wss_keep_alive_add_client`. Otherwise, it will run out of socket number and make new TCP connection(WS connection) not be accepted.

law-ko, 2023-04-23:
@ESP-YJM If this is the case, what would be the solution for this? How should the code look like if we want REST and WS work at the same time?

ESP-YJM, 2023-04-24:
You can follow this patch and try.
[wss_server_with_rest_server.patch](https://github.com/espressif/esp-idf/files/11307257/wss_server_with_rest_server.patch)

law-ko, 2023-04-30:
@ESP-YJM , this patch is for ESP IDF v5. We are currently on v4.4, anyway these settings can also be applied on v4.4?

ESP-YJM, 2023-05-04:
@law-ko If you use v4.4, you need first merge those two commits https://github.com/espressif/esp-idf/commit/e05d61d07563e036855a4601dfbe3b968d904cf0 and https://github.com/espressif/esp-idf/commit/a3fd6d1070312ae111003a04f50a56efbaab1769 manual and then add my patch offered before.

AxelLin, 2023-05-04:
> @law-ko If you use v4.4, you need first merge those two commits [e05d61d](https://github.com/espressif/esp-idf/commit/e05d61d07563e036855a4601dfbe3b968d904cf0) and [a3fd6d1](https://github.com/espressif/esp-idf/commit/a3fd6d1070312ae111003a04f50a56efbaab1769) manual and then add my patch offered before.

Why not backport the fix to 4.4?
https://github.com/espressif/esp-idf/issues/9848#issuecomment-1274185957 was reported against **v4.4**.

law-ko, 2023-08-08:
@ESP-YJM Tested with v5.0.1 with patches applied, but still having 2 extra clients not being disconnected after webpage is fully loaded.

```
I (8613) esp_https_server: performing session handshake
I (9683) wss_echo_server: New client connected 57
I (9713) wss_echo_server: File sending complete
I (10383) wss_echo_server: File sending complete
I (10393) esp_https_server: performing session handshake
I (11463) wss_echo_server: New client connected 58
I (11613) wss_echo_server: File sending complete
I (11623) esp_https_server: performing session handshake
I (12683) wss_echo_server: New client connected 59
I (13903) wss_echo_server: File sending complete
I (13913) esp_https_server: performing session handshake
I (14973) wss_echo_server: New client connected 60
I (15523) wss_echo_server: File sending complete
I (15523) wss_echo_server: Client disconnected 57
E (15523) wss_keep_alive: Cannot remove client fd:57
I (15533) esp_https_server: performing session handshake
I (16603) wss_echo_server: New client connected 57
I (16643) wss_echo_server: File sending complete
I (16653) wss_echo_server: Client disconnected 58
E (16653) wss_keep_alive: Cannot remove client fd:58
I (16663) esp_https_server: performing session handshake
I (17743) wss_echo_server: New client connected 58
I (18823) wss_echo_server: File sending complete
I (18863) wss_echo_server: File sending complete
I (19043) wss_echo_server: File sending complete
```

Taking wss_server example and merge in restful_server example project. Attached with merged project.
[wss_rest_server.zip](https://github.com/espressif/esp-idf/files/12287606/wss_rest_server.zip)

law-ko, 2023-10-05:
@ESP-YJM Any updates on this?

ESP-YJM, 2023-10-18:
@law-ko It seems the `queue_size` defined in function `wss_keep_alive_start` is too small, could you please increase it.

law-ko, 2023-10-18:
@ESP-YJM How big should the `queue_size` be? Currently it is `config->max_clients/2`

```
wss_keep_alive_t wss_keep_alive_start(wss_keep_alive_config_t *config)
{
    size_t queue_size = config->max_clients/2;
    size_t client_list_size = config->max_clients + queue_size;
    wss_keep_alive_t keep_alive_storage = calloc(1,
            sizeof(wss_keep_alive_storage_t) + client_list_size* sizeof(client_fd_action_t));
    if (keep_alive_storage == NULL) {
        return false;
    }
    keep_alive_storage->check_client_alive_cb = config->check_client_alive_cb;
    keep_alive_storage->client_not_alive_cb = config->client_not_alive_cb;
    keep_alive_storage->max_clients = config->max_clients;
    keep_alive_storage->not_alive_after_ms = config->not_alive_after_ms;
    keep_alive_storage->keep_alive_period_ms = config->keep_alive_period_ms;
    keep_alive_storage->user_ctx = config->user_ctx;
    keep_alive_storage->q =  xQueueCreate(queue_size, sizeof(client_fd_action_t));
    if (xTaskCreate(keep_alive_task, "keep_alive_task", config->task_stack_size,
                    keep_alive_storage, config->task_prio, NULL) != pdTRUE) {
        wss_keep_alive_stop(keep_alive_storage);
        return false;
    }
    return keep_alive_storage;
}
```

law-ko, 2023-10-19:
@ESP-YJM 

Upon setting to 10. It seems like it will crash due to stack overflow:
```
E (105002) wss_keep_alive: Client (fd=60) not alive!
E (105002) WS_REST_SERVER: Cl
***ERROR*** A stack overflow in task keep_alive_task has been detected.

Backtrace: 0x4008226e:0x3ffb8220 0x40093c5d:0x3ffb8240 0x40096ffe:0x3ffb8260 0x4009570d:0x3ffb82e0 0x40093d58:0x3ffb8310 0x40093d0a:0xa5a5a5a5 |<-CORRUPTED
0x4008226e: panic_abort at /home/espuser/esp/esp-idf/components/esp_system/panic.c:423

0x40093c5d: esp_system_abort at /home/espuser/esp/esp-idf/components/esp_system/esp_system.c:153

0x40096ffe: vApplicationStackOverflowHook at /home/espuser/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:391

0x4009570d: vTaskSwitchContext at /home/espuser/esp/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:3648

0x40093d58: _frxt_dispatch at /home/espuser/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:418

0x40093d0a: _frxt_int_exit at /home/espuser/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:213
```

ESP-YJM, 2023-10-19:
@law-ko You need also increase task `keep_alive_task ` task size.

law-ko, 2023-10-19:
@ESP-YJM I have manually adjusted to 8192.

We want to support a maximum of 4 WS clients, what is the recommended `queue_size` and the `task size`? Also, can the `keep_alive_task` be run on top of PSRAM?

ESP-YJM, 2023-10-19:
Yes, it can run psram. `task size` adjusted to 4096 maybe enough and queue_size should be adjusted to 8

closed 2024-07-01 by gabsuren