espressif/esp-idf #11099: High Resolution Timer header file (esp_timer.h) cannot be found

enoch-abban, 2023-03-31:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.1

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

I expected it to successfully create an esp32 image and then complete the build process.

### What is the actual behavior?

Instead, the build process failed with the following output:

      fatal error: esp_timer.h: No such file or directory
         13 | #include "esp_timer.h"
            |          ^~~~~~~~~~~~~
      compilation terminated.
      ninja: build stopped: subcommand failed.
      HINT: Please make sure that the header name is correct. Also please check if you've specified all component dependencies 
      with 'idf_component_register(REQUIRES ...)'. If the component is not present then it should be added 
      by the IDF Component Manager. For more information run 'idf.py docs -sp api-guides/build-system.html'.
      Also, please check if the header file has been removed, renamed or relocated - refer to the migration guide for more 
      information.
      ninja failed with exit code 1

### Steps to reproduce.

[code-and-sdkconfig.zip](https://github.com/espressif/esp-idf/files/11127040/code-and-sdkconfig.zip)

vortex314, 2023-04-02:
I had the same issue, adding esp_timer to the components seemed to remediate 
```
REQUIRES freertos nvs_flash mqtt esp_https_ota app_update esp_adc esp_hw_support lwip esp_rom esp_timer )
```

igrr, 2023-04-03:
@vortex314 's suggestion is correct, if you are including "esp_timer.h", you need to make sure "esp_timer" component is added to the requirements list of your component (REQUIRES or PRIV_REQUIRES in CMake).

For more information, see [Component Requirements](https://docs.espressif.com/projects/esp-idf/en/v5.0.1/esp32/api-guides/build-system.html#component-requirements) in the programming guide.

vortex314, 2023-04-03:
@igrr  , it looks like this explicit component dependency is introduced when the code is migrated from 4.x to 5.x. The same is happening with 'driver' and 'esp_wifi'.

igrr, 2023-04-03:
You are right, project written with IDF v4.x need to be updated to explicitly list these components as dependencies. Here is the relevant section of the migration guide:

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/release-5.x/5.0/build-system.html#specify-component-requirements-explicitly

zikalino, 2023-05-24:
I am going to close this ticket. @enoch-abban please feel free to reopen and add more info if you're still experiencing this problem.

closed 2023-05-24 by zikalino