espressif/esp-idf #11101: Can't use the SD card reader with ESP-IDF

TheElectronicMusicProject, 2023-04-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.4-336-g4b404115e8

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-WROOM-32

### Power Supply used.

USB

### What is the expected behavior?

I made a slight variant of the [SPI SD card](https://github.com/espressif/esp-idf/tree/release/v4.4/examples/storage/sd_card/sdspi) official example. You can find it [there](https://github.com/TheElectronicMusicProject/Learn_ESP32/blob/master/I2C_SPI_UART/11_7_spi_ssd/main/main.c).

I connected the SPI terminals as follows:

* CS - GPIO5
* SCK - GPIO18
* MOSI- GPIO21
* MISO - GPIO19
* VCC - +5V
* GND - GND

and I ran the code.

I use a micro-SD card reader (also tried a SD card reader) I bought from [here](https://it.aliexpress.com/item/1005001861849866.html?spm=a2g0o.order_list.order_list_main.47.164a3696QlsT4z&gatewayAdapt=glo2ita).

With both the example and my code I expect they create files in the micro SD card etc...

### What is the actual behavior?

Instead it returns an error (same error code also in the official example).

```
I (0) cpu_start: App cpu up.
I (243) cpu_start: Pro cpu start user code
I (243) cpu_start: cpu freq: 160000000
I (243) cpu_start: Application information:
I (248) cpu_start: Project name:     11_7_spi_ssd
I (253) cpu_start: App version:      dfa940e-dirty
I (259) cpu_start: Compile time:     Mar 31 2023 20:43:00
I (265) cpu_start: ELF file SHA256:  e5727dca3bfd0fa0...
I (271) cpu_start: ESP-IDF:          -128-NOTFOUND
I (276) heap_init: Initializing. RAM available for dynamic allocation:
I (283) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (289) heap_init: At 3FFB2E50 len 0002D1B0 (180 KiB): DRAM
I (296) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (302) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (308) heap_init: At 4008C4A4 len 00013B5C (78 KiB): IRAM
I (316) spi_flash: detected chip: generic
I (319) spi_flash: flash io: dio
I (324) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
E (334) vfs_fat_sdmmc: slot init failed (0x102).
E (344) spi: Failed to initialize the card: ESP_ERR_INVALID_ARG
```

To exclude hardware problems, I also ran the official example for Arduino-ESP32 found [here](https://github.com/espressif/arduino-esp32/blob/master/libraries/SD/examples/SD_Test/SD_Test.ino) and IT WORKS withouth any problem!!!

In  order to use this last code, I connected the hardware as follows:

* CS - GPIO5
* SCK - GPIO18
* MOSI- GPIO23
* MISO - GPIO19
* VCC - +5V
* GND - GND

Finally, I tried these pins in my example but the final behaviour is with the same error code...

### Steps to reproduce.

I used a 2GB micro-SD card formatted as FAT32.

chipweinberger, 2023-04-02:
add some print statements to determine where ESP_ERR_INVALID_ARG is coming from

nopnop2002, 2023-04-03:
>VCC - +5V

Note that ESP32 is not 5V tolerant.

If you drive the SPI device with 5V, MISO will be 5V.

TheElectronicMusicProject, 2023-04-03:
> > VCC - +5V
> 
> Note that ESP32 is not 5V tolerant.
> 
> If you drive the SPI device with 5V, MISO will be 5V and in the worst case destroy the ESP32.
> 
> You need to use an SPI device that can operate at 3.3V.

This module has a converter to allow a 5V power supply that is then converted into a 3.3V

TheElectronicMusicProject, 2023-04-06:
> add some print statements to determine what causes ESP_ERR_INVALID_ARG

I'm debugging right now my code using the ESP-PROG.
I see that the inner function which seems to return the error is `sdmmc_host_init_slot()`.
Inside this function I reach the point where it tests the following:
```c
if (slot_width == SDMMC_SLOT_WIDTH_DEFAULT) {
    slot_width = slot_info->width;
}
else if (slot_width > slot_info->width) {
    return ESP_ERR_INVALID_ARG;
}
s_slot_width[slot] = slot_width;
```

It skips the `if()` and reaches the `else if()`. The debugger shows me that `slot_width` is 255 but it doesn't show the value of `slot_info->width` because it's optimized.

I don't know what is doing this function, however I clearly see:
* `slot_width` is the value returned by `slot_config->width`, which is an input of this function
* `slot_info->width` is returned by `&sdmmc_slot_info[slot]`

I suppose the invalid variable could be `slot_width`.

**Now I have a doubt: is `esp_vfs_fat_sdmmc_mount()` the right function to mount the SD card?** I use SPI communication and I noticed that it exists another function called `esp_vfs_fat_sdspi_mount()` which takes a `sdspi_device_config_t` struct as third argument.
This last struct doesn't permit the use of SPI pins though...

chipweinberger, 2023-04-06:
In your code, you are setting the width here. You should make sure it is correct.

```
    sdspi_slot_config_t slot_config = SDSPI_SLOT_CONFIG_DEFAULT();
    slot_config.gpio_miso = PIN_NUM_MISO;
    slot_config.gpio_mosi = PIN_NUM_MOSI;
    slot_config.gpio_sck = PIN_NUM_CLK;
    slot_config.gpio_cs = PIN_NUM_CS;
```

chipweinberger, 2023-04-06:
It also looks like you changed:

    sdmmc_host_t host = SDSPI_HOST_DEFAULT(); 

to

    sdmmc_host_t host = SDMMC_HOST_DEFAULT(); 

Not sure why you changed that if you want SPI communication.

TheElectronicMusicProject, 2023-04-07:
> Not sure why you changed that if you want SPI communication.
> 
> Just go back to the example code, and change fewer things.

Thank you @chipweinberger , I switched to the sdspi example. I routed my connections in order to obtain the exact pins that are used in the example. So:

* MISO - GPIO2
* MOSI - GPIO15
* CLK - GPIO14
* CS  - GPIO13

Having done that, I uploaded the code in my ESP32 but again, an **error is still present**. It isn't the same as the previous one.
```
I (0) cpu_start: App cpu up.
I (228) cpu_start: Pro cpu start user code
I (228) cpu_start: cpu freq: 160000000
I (228) cpu_start: Application information:
I (232) cpu_start: Project name:     11_7_spi_ssd
I (238) cpu_start: App version:      6ba7bbe-dirty
I (243) cpu_start: Compile time:     Apr  6 2023 20:49:11
I (249) cpu_start: ELF file SHA256:  1f2610110c6eb918...
I (255) cpu_start: ESP-IDF:          -128-NOTFOUND
I (261) cpu_start: Min chip rev:     v0.0
I (265) cpu_start: Max chip rev:     v3.99
I (270) cpu_start: Chip rev:         v1.0
I (275) heap_init: Initializing. RAM available for dynamic allocation:
I (282) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (288) heap_init: At 3FFB2668 len 0002D998 (182 KiB): DRAM
I (294) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (301) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (307) heap_init: At 4008D1DC len 00012E24 (75 KiB): IRAM
I (315) spi_flash: detected chip: generic
I (318) spi_flash: flash io: dio
I (323) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (333) spi: Initializing SD card
I (333) spi: Mounting filesystem
I (343) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (393) sdspi_transaction: cmd=5, R1 response: command not supported
E (493) sdmmc_sd: sdmmc_check_scr: send_scr returned 0x109
E (493) vfs_fat_sdmmc: sdmmc_card_init failed (0x109).
I (493) gpio: GPIO[5]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
E (503) spi: Failed to initialize the card: ESP_ERR_INVALID_CRC
```
I used the ESP-PROG to find the exact point where it fails.
It happens inside `esp_vfs_fat_sdspi_mount()` -> `sdmmc_card_init()` -> `sdmmc_check_scr()`.

Then I searched for a similar issue here on Github and I found the following [issue](https://github.com/espressif/esp-idf/issues/5561)

Following the solution, I reduced the frequency as follows:
* `host.max_freq_khz = 5000;` **WORKS!**
* `host.max_freq_khz = 10000;` **WORKS!**
* `host.max_freq_khz = 15000;` **WORKS!**
* `host.max_freq_khz = 19000;` **WORKS!**
* `host.max_freq_khz = 20000;` **NOT WORKING**

I also noticed that `host.max_freq_khz = 20000;` is what happens when I run `sdmmc_host_t host = SDSPI_HOST_DEFAULT();`, so the default value could be instable with *some* devices!!!
I didn't try other SD cards, then I don't know if the same problem could happen with them.

TheElectronicMusicProject, 2023-04-07:
I tested also the same program with an SPI SD card reader. The previous one was a micro SD card reader.
I used an adapter to insert my 2GB micro SD inside the SD reader slot. I bought it from the [same website](https://it.aliexpress.com/item/1005001861849866.html?spm=a2g0o.order_list.order_list_main.47.164a3696QlsT4z&gatewayAdapt=glo2ita).

I ran the program with `host.max_freq_khz = 20000;` and now it works also with this frequency!

Could it be an hardware issue?

chipweinberger, 2023-04-07:
Higher frequencies wont work well with long wires! Things can become unstable and unpredictable.

TheElectronicMusicProject, 2023-04-08:
> Higher frequencies wont work with long wires! Things become unstable and unpredictable.
> 
> Use as short of wires as possible, for good signal integrity.
Yes, I was using very long wires. I'm sure this is the source of the problem.
Thank you for your help, I think we can close this issue!

closed 2023-04-08 by TheElectronicMusicProject