espressif/esp-idf #11102: How to uninstall esp-idf?

ttytyper, 2023-04-02:
Installing is easy enough by just running the install.sh script. But how do you uninstall it? I may have messed up the installation, and I want to revert everything and retry from scratch.

It looks like the script primarily installs a bunch of Python packages and copies files to various directories, but it's difficult to figure out which ones without going down a deep rabbit hole of nested scripts :)

atanisoft, 2023-04-02:
Remove `$HOME/.espressif` and you should be able to reinstall from scratch.

dobairoland, 2023-04-03:
Yes, everything goes by default to the `.espressif` folder. 

This page contains the official documentation to the installer script: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-tools.html

Alvin1Zhang, 2023-04-17:
Thanks for reporting, feel free to reopen.

dobairoland, 2023-04-17:
We plan to review and update the documentation. I re-open it as a reminder.