espressif/esp-idf #11104: [question] files sync between USB MSC and FFAT

chegewara, 2023-04-03:
### Is your feature request related to a problem?

Hi,

i cant figure out how to sync USB MSC with FFAT files/folders.
Im not sure if its OS related, because even when linux is detecting MSC without cache still FFAT is not seeying changes:
```
sd 9:0:0:0: [sdc] No Caching mode page found
sd 9:0:0:0: [sdc] Assuming drive cache: write through
```
Ive been trying to add this, but no success:
```
void tud_msc_write10_complete_cb(uint8_t lun)
{
    Serial.println("write10 completed");
    if (disk_ioctl(0, CTRL_SYNC, NULL) != RES_OK)
    {
        Serial.printf("failed to sync\n");
    }
}
```

Thanks

igrr, 2023-04-03:
Are you initializing FAT on the ESP and giving the host PC access to the underlying storage simultaneously?

If yes, i don't think this will work, at least not with FATFS library. It is written under an assumption that the volume is only modified by the library itself. It doesn't have the logic to detect the changes which happened on the volume and update its internal state.
Typically you should only mount FATFS when the host ejects the MSC drive.

If you are mounting FAT on MSC eject only, but still not seeing the changes, then that's something unexpected. In that case, can you please try the provided MSC/FAT example and see if you have the same behavior?

chegewara, 2023-04-03:
> Are you initializing FAT on the ESP and giving the host PC access to the underlying storage simultaneously?

Well, i am using esp-idf vfat, either with MSC code and esp-idf ffat. 
This is the code i am using to init ffat under in msc library:
```
    esp_vfs_fat_mount_config_t mount_config = 
    {
        .format_if_mount_failed = true,
        .max_files = 5,
        .allocation_unit_size = CONFIG_WL_SECTOR_SIZE
    };

    esp_err_t err = esp_vfs_fat_spiflash_mount(path, label, &mount_config, &wl_handle);
```
and to read/write:
```
disk_write(s_pdrv, _buf, lba, 1);
disk_read(s_pdrv, _buf, lba, 1);
```
I thought this should do the job, and i think it worked only one time(randomly):
```
disk_ioctl(0, CTRL_SYNC, NULL)
```
To inform OS about changes i probably need to add some extra code, but i have to study USB MSC protocol yet.

Thanks

igrr, 2023-04-03:
> Well, i am using esp-idf vfat, either with MSC code and esp-idf ffat.

Do you mean "either .. or" (i.e. esp-idf vfat and MSC are not working at the same time) or "and" (esp-idf vfat and MSC are working at the same time)?

> I thought this should do the job, and i think it worked only one time(randomly):

`disk_ioctl` does not inform the upper layer (FATFS library) that the data on disk has been changed. (For example, by MSC.)

`disk_ioctl` only syncs the changes performed using `disk_write` to the flash partition. In fact, since all Flash storage APIs are synchronous and nothing actually gets cached, `disk_ioctl` [doesn't really do anything](https://github.com/espressif/esp-idf/blob/42261df71f12b0b995774d103ce68e40e8fba529/components/fatfs/diskio/diskio_wl.c#L69-L70).

chegewara, 2023-04-03:
> Do you mean "either .. or" (i.e. esp-idf vfat and MSC are not working at the same time) or "and" (esp-idf vfat and MSC are working at the same time)?

Yes, they are working at the same time. One initialization, from MSC library. All is working fine, just files seems not to be refreshed in FAT table.

> only syncs the changes performed using disk_write to the flash partition

My MSC write is using `disk_write` to write bytes on fat partition.
For example, if i delete file under OS (MSC drive), then esp-idf code is using `disk_write` to change FFAT partition table, but fat code still think the file is there.

Thanks

igrr, 2023-04-03:
Okay, thanks for explaining, I understand the use case now.

As I wrote in the first comment, I don't think this will work. FATFS library doesn't expect that the disk (partition) will be modified by anything other than FATFS itself.

I don't think this type of concurrent usage can be supported other than by de-initializing FATFS and re-initializing it again. Or, using some other FATFS library which has a "refresh internal state from disk" type of functionality.

In the esp_tinyusb component we make mounting of the partition by host and by VFS mutually exclusive to prevent such issues. (E.g. the partition is [mounted by VFS when it is unmounted over MSC](https://github.com/espressif/idf-extra-components/blob/65d13a0d9472804fb40b9ad1bbb3aa1f252885d2/usb/esp_tinyusb/tusb_msc_storage.c#L504-L509))

esp-saurabhbansal, 2023-04-13:
Hello,

Just for your information, We have recently released USB Device MSC driver. Please check if it is useful to you. We highly appreciate the feedback.

The driver makes ESP based device recognizable by USB-hosts as a USB Mass Storage Device.
It either allows the embedded application to access the partition or Host PC accesses the partition over USB MSC. They can't be allowed to access the partition at the same time.

The driver supports storage media of two types:
1. SPI Flash
2. SD MMC Card

Driver can be found [here](https://github.com/espressif/idf-extra-components/blob/master/usb/esp_tinyusb/tusb_msc_storage.c).
Sample application can be found [here](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/device/tusb_msc).

Thanks.

chegewara, 2023-04-15:
Thanks, but i already have this done long time ago, but its good to know espressif also did it finally.

I can use USB MSC and esp-idf API at the same time with no issue, or i dint encounter any issues yet.
I can even see changes in file system when using OS i will delete/create files. What i have problem with is to sync files when i perform files operation using esp-idf API, because OS cant see new files, changes when unlink/delete files or change files content.
Probably i am missing some tinyusb API or, like igrr explained, i have to re-mount file system.