espressif/esp-idf #11105: IDF4.4 I2S PDM issue in using PDM output microphone, clk freq cannot up to 1MHz?

edword01, 2023-04-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

### IDF version: v4.4
### PDM microphone document
[https://item.szlcsc.com/510337.html](url)
### ESP module: 
ESP32S3-1u-N8R8
### Hardware connection: 
PDM output microphone with L/R pin,  CLK pin and Data pin.
ISSUE: my microphone requires clk up to 1MHz, if clk freq less than 1MHz, it will sleep. how to config , to make esp32s3's
i2s interface can generate 1Mhz????

i2s config is showed bellow, I have try several days to config this, but all fail to do this. 
I config microphone's CLK pin to the 'pin_config.ws_io_num'. 

### i2s config:
/*------------------------------------*/
// Set the I2S configuration as PDM and 16bits per sample
    i2s_config_t i2s_config = {
        .mode = I2S_MODE_MASTER | I2S_MODE_RX | I2S_MODE_PDM,
        .sample_rate = I2S_SAMPLE_RATE, //CONFIG_EXAMPLE_SAMPLE_RATE,
        .bits_per_sample = I2S_BITS_PER_SAMPLE_32BIT,
        .channel_format = I2S_CHANNEL_FMT_ONLY_LEFT,//I2S_CHANNEL_FMT_ONLY_RIGHT,//I2S_CHANNEL_FMT_ONLY_LEFT,
        .communication_format = I2S_COMM_FORMAT_STAND_PCM_SHORT,
        .intr_alloc_flags = ESP_INTR_FLAG_LEVEL1, //ESP_INTR_FLAG_LEVEL2,
        .dma_buf_count = 8,
        .dma_buf_len = 128,
        .use_apll = 0,//0
        .fixed_mclk = 1000000,
    };
    
    // Set the pinout configuration (set using menuconfig)
    i2s_pin_config_t pin_config = {
        .mck_io_num = I2S_PIN_NO_CHANGE, //I2S_PIN_NO_CHANGE,
        .bck_io_num = I2S_PIN_NO_CHANGE, //MIC_SCK_PIN,
        .ws_io_num = MIC_SCK_PIN,//MIC_LR_PIN, //CONFIG_EXAMPLE_I2S_CLK_GPIO,
        .data_out_num = I2S_PIN_NO_CHANGE,
        .data_in_num = MIC_DAT_PIN, //CONFIG_EXAMPLE_I2S_DATA_GPIO,
        
    };

    // Call driver installation function before any I2S R/W operation.
    ESP_ERROR_CHECK( i2s_driver_install(CONFIG_I2S_CH, &i2s_config, 0, NULL) );
    ESP_ERROR_CHECK( i2s_set_pin(CONFIG_I2S_CH, &pin_config) );
/*---------------------------------------------------------------------*/

edword01, 2023-04-03:
this microphone status flow:

[image]

L-KAYA, 2023-04-03:
Hi @edword01, I've tested the PDM clock output with your configuration on ESP32S3@IDF release/v4.4, it can reach about 3MHz.
[image]

May I confirm that how you monitor the clock output? Is it monitored on `MIC_SCK_PIN` with a logic analyzer or an oscilloscope?

BTW, although your configuration can output the clock, there are still some inappropriate config:

1. Please set `bits_per_sample` to `I2S_BITS_PER_SAMPLE_16BIT`, because PDM can only support `I2S_BITS_PER_SAMPLE_16BIT`;
2. `fixed_mclk` is not the output clock of PDM mode, it is at least 4 times than the signal on `ws_io_num ` (i.e. the PDM clock output pin). And, `fixed_mclk` won't take effect because it relies on `APLL` while ESP32S3 doesn't support APLL.

edword01, 2023-04-04:
@L-KAYA thanks, now i can detected 1.4MHz clk, is it calculate PDM clock automatically?? 
1. how to configure PDM clock freq? if i want to.
2. now , PDM data signal out , only has about 40mv. very low.

edword01, 2023-04-04:
According to this :  https://github.com/espressif/esp-idf/issues/8660#issuecomment-1205737159
can i need to change wav file's sample rate to 1/2 of the value set to the i2s config ?

edword01, 2023-04-04:
[image]

edword01, 2023-04-04:
now i config :
```
// Set the I2S configuration as PDM and 16bits per sample
    i2s_config_t i2s_config = {
        .mode = I2S_MODE_MASTER | I2S_MODE_RX | I2S_MODE_PDM,
        .sample_rate = I2S_SAMPLE_RATE, //CONFIG_EXAMPLE_SAMPLE_RATE,
        .bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT,
        .channel_format = I2S_CHANNEL_FMT_ONLY_RIGHT,//I2S_CHANNEL_FMT_ONLY_RIGHT,//I2S_CHANNEL_FMT_ONLY_LEFT,
        .communication_format = I2S_COMM_FORMAT_STAND_MSB,//I2S_COMM_FORMAT_STAND_PCM_SHORT,
        .intr_alloc_flags = ESP_INTR_FLAG_LEVEL1, //ESP_INTR_FLAG_LEVEL2,
        .dma_buf_count = 8,
        .dma_buf_len = 128,
        .use_apll = 0,//0
        //.fixed_mclk = 1000000,
    };
    i2s_pcm_cfg_t pcm_config = {
      .pcm_type = I2S_PCM_A_COMPRESS,
    };
    
    //uint32_t pdm_clk = i2s_calculate_pdm_rx_clock(i2s_config.sample_rate, I2S_BITS_PER_SAMPLE_16BIT, I2S_CHANNEL_MONO, 1);
    //i2s_config.fixed_mclk = pdm_clk;
    // Set the pinout configuration (set using menuconfig)
    i2s_pin_config_t pin_config = {
        .mck_io_num = I2S_PIN_NO_CHANGE,//I2S_PIN_NO_CHANGE,
        .bck_io_num = I2S_PIN_NO_CHANGE,//MIC_LR_PIN, //MIC_SCK_PIN,
        .ws_io_num = MIC_SCK_PIN,//MIC_LR_PIN, //CONFIG_EXAMPLE_I2S_CLK_GPIO,
        .data_out_num = I2S_PIN_NO_CHANGE,
        .data_in_num = MIC_WS_DAT, //CONFIG_EXAMPLE_I2S_DATA_GPIO,
        
    };

    // Call driver installation function before any I2S R/W operation.
    ESP_ERROR_CHECK( i2s_driver_install(CONFIG_I2S_CH, &i2s_config, 0, NULL) );
    i2s_pcm_config(CONFIG_I2S_CH, &pcm_config);
    i2s_set_pdm_rx_down_sample(CONFIG_I2S_CH, I2S_PDM_DSR_16S);
    ESP_ERROR_CHECK( i2s_set_pin(CONFIG_I2S_CH, &pin_config) );
/```

Then after save the wav file, the wav file wave look like this:
![image](https://user-images.githubusercontent.com/26646966/229743897-13b966e3-0443-4a5e-b333-eafa562e0e72.png)

Is that communication_format config error?? the wave are not right when recording.

edword01, 2023-04-04:
What's the config logic of communication format??? in PDM mode? 
about I2S_COMM_FORMAT_STAND_MSB or I2S_COMM_FORMAT_STAND_I2S,
and  I2S_CHANNEL_FMT_ONLY_RIGHT or I2S_CHANNEL_FMT_ONLY_LEFT. ? 
how to select this value in PDM mode, and is may affect the logic of esp32s3 to translate PDM to PCM file.

L-KAYA, 2023-04-04:
In pdm mode, `communication format` takes no effect. The `channel format` can help you to select left / right or both PDM microphones. 

Normally, `i2s_pcm_config` and `i2s_set_pdm_rx_down_sample ` is not needed. 

The output clock is influenced by the `down_sample_rate`

edword01, 2023-04-04:
but now the wav file is not right，looks only have noises.

edword01, 2023-04-06:
@L-KAYA a little confusion here, what's the data format after using PDM mode to read data? data read from 'i2s_read' is PCM data ? and can it just combine wave header and  save as wav file?  or i need to do some conversion to the raw data ??

L-KAYA, 2023-04-06:
Yes, the date read from I2S will be transmitted to PCM format by the hardware, no need to convert it manually.

If possible, I suggest to use PDM in IDF v5.0. Limited by the legacy APIs, the PDM supporting on v4.4 is not that completed.

edword01, 2023-04-06:
now i cannot obtain the wav file correctly，wave looks only have noise，don't know what to do now

L-KAYA, 2023-04-06:
>  now i cannot obtain the wav file correctly，wave looks only have noise，don't know what to do now

You can check the issue in following step:

1. Monitor the signals on the data line, as a PDM microphone, it should output the data as long as the clock is provided appropriately.

- If there is data on line, please check if the data line is connected to the correct GPIO
- If there is no data on line, keep going to the next check

2. Provide square wave at about several MHz to the PDM Mic directly by a signal generator(you can also generate the clock by other way), so that the PDM Mic can get a stable and appropriate clock.

- If the data can be detected on line in this case, it means the I2S PDM not output an appropriate clock, we need to look into it then;
- If the data still doesn't appear, the PDM Mic might be broken;

edword01, 2023-04-07:
@L-KAYA thanks, now i can record wav successfuly, but volume very low, how to do with this:
[image]
And why the signal's horizontal axis not start from Inf  ??

L-KAYA, 2023-04-10:
As far as I know, I2S PDM doesn't support adjust the gain, the url of the Mic doc seems invalid so I don't know if the PDM Mic supports to increase the gain.

Not sure why the wave has offset, maybe you can try another software: `audacity`

xG3nesis, 2024-04-27:
Hello guys @L-KAYA, @edword01 ! 

It seems like it's been a while since this issue was initially raised! I'm chiming in now with my comment, hoping to receive some assistance and possibly provide help to others who might come across this thread. Let's dive straight into it! 👇 

I've bought an [M5 Atom Echo](https://docs.m5stack.com/en/atom/atomecho?ref=pfpqkvphmgr) on which there is an [ESP32-PICO-D4](https://m5stack.oss-cn-shenzhen.aliyuncs.com/resource/docs/datasheet/core/esp32-pico-d4_datasheet_en.pdf) with an [SPM1423](https://m5stack.oss-cn-shenzhen.aliyuncs.com/resource/docs/datasheet/core/SPM1423HM4H-B_datasheet_en.pdf) PDM microphone and a [NS4168](https://m5stack.oss-cn-shenzhen.aliyuncs.com/resource/docs/datasheet/core/NS4168_CN_datasheet.pdf) speaker. Here is the schematic available on the website of M5 stack : 

[image]

My goal was to record my voice using a microphone, transmit it via Bluetooth to my computer. However, when I attempted to implement this using PlatformIO with the Arduino ESP32 library version 2.0.15 (the latest), which is based on ESP-IDF v4.4.7, I encountered an issue. Despite setting the parameters to signed 16-bit PCM, MSB, Stereo, and a sample rate of 16000Hz (although it doesn't work when I set it to 44100Hz in my code), I couldn't recover my PCM audio when decoding it with Audacity. Here's what happens when importing the raw file:

[image]

Rather than PCM, it appears that I'm getting PDM raw data. It seems that automatic conversion from PDM to PCM is not occurring.

Here's a glimpse of my code:
```C++
#include <BLEDevice.h>
#include <BLEServer.h>
#include <BLE2902.h>
#include <driver/i2s.h>
#include <M5Atom.h>

#define MICRO_I2S_NUMBER I2S_NUM_0

#define CONFIG_I2S_BCK_PIN 19
#define CONFIG_I2S_LRCK_PIN 33
#define CONFIG_I2S_DATA_PIN 22
#define CONFIG_I2S_DATA_IN_PIN 23

#define MODE_MIC 0
#define MODE_SPK 1
#define DATA_SIZE 1024

uint8_t microphonedata0[1024 * 80];
int data_offset = 0;
int count = 0;

/*INITIALIZATION OF I2S PARAMETERS (MICROPHONE)*/
void InitI2SMic(int mode)
{
    // Configurate general I2S Driver parameters !
    esp_err_t err = ESP_OK;

    i2s_driver_uninstall(MICRO_I2S_NUMBER);
    i2s_config_t i2s_config = {
        .mode = (i2s_mode_t)I2S_MODE_MASTER,
        .sample_rate = 44100,
        .bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT, 
        .channel_format = I2S_CHANNEL_FMT_ALL_RIGHT,
        .communication_format = I2S_COMM_FORMAT_STAND_I2S,
        .intr_alloc_flags = ESP_INTR_FLAG_LEVEL1,
        .dma_buf_count = 6,
        .dma_buf_len = 60,
        .use_apll = false,
        .tx_desc_auto_clear = false,
        .fixed_mclk = 0, 
    };

    if (mode == MODE_MIC)
    {
        i2s_config.mode = (i2s_mode_t)(I2S_MODE_MASTER | I2S_MODE_RX | I2S_MODE_PDM);
    }
    else
    {
        i2s_config.mode = (i2s_mode_t)(I2S_MODE_MASTER | I2S_MODE_TX);
        i2s_config.use_apll = false;
        i2s_config.tx_desc_auto_clear = true;
    }

    err += i2s_driver_install(MICRO_I2S_NUMBER, &i2s_config, 0, NULL);
[... 168 lines cut ...]
            printf("[0] %d [1] %d [2] %d [3]%d ...\n", microphone_data[0], microphone_data[1], microphone_data[2], microphone_data[3]);
        } else {
            printf("Read Failed!\n");
        }
        vTaskDelay(100/portTICK_PERIOD_MS);
        buttonState = digitalRead(BUTTON_PIN);
        record = true;
  }
  
  if (record == true) {
    printf("Recording complete !");
  }

  i2s_channel_disable(rx_handle);
  i2s_del_channel(rx_handle);
}
```

The output doesn't appear to align with the PCM data format. Despite searching through the ESP-IDF documentation for a method to convert PDM to PCM, I couldn't find one. Here's what the output looks like on the serial monitor:

[image]

I'm having trouble understanding what I'm doing wrong. I would greatly appreciate any assistance you could provide 😃!

Thank you in advance.

mmackh, 2024-04-27:
bump - is there any update on new audio drivers PDM -> PCM conversion?

xG3nesis, 2024-04-27:
Additionnaly, I really think that i'm dealing with PDM because when i perform a manual PDM to PCM conversion by applying a low pass filter, decimation and high pass filter, i almost recover my audio in audacity.

I apologize, but I find the ESP-IDF documentation to be unclear regarding [PDM Mode (RX)](https://docs.espressif.com/projects/esp-idf/en/v5.1.2/esp32/api-reference/peripherals/i2s.html#pdm-mode-rx) in version 5.1.2. Specifically, when they mention, "PDM (Pulse-density Modulation) mode for RX channel can receive PDM-format data and convert the data into PCM format," it's ambiguous. Does this imply that it has the capability to convert, or that it automatically performs the conversion? Do we need to initialize it somehow ?

xenpac, 2024-06-05:
interesting..
The PDM input is automaticly converted to PCM in hardware.

i am currently working on PDM Mic asto basic pdm example in 5.2.

The Mics converted PCM Volume is extremely low. Is there any gain-setting ?

DDDanny, 2024-12-19:
@xG3nesis : I operate a [m5 stack core2](https://github.com/m5stack/m5-docs/blob/master/docs/en/core/core2.md]) that has also a SPM1423 mic. I'm using micropython firmware that currently does not support I2S PDM RX but involved in a FR clean implementation which is able to fetch PDM as PCM converted audio without any programm side conversion in v 5.2.2
As mentioned by @xenpac I can confirm that data is internally converted to PCM format.
And I agree that the documentation and also the driver implementation is confusing/ambious sometimes. 

I would suggest that you test in STEREO (by accepting the left channel empty) with `i2s_pdm_slot_mask_t` `I2S_PDM_SLOT_BOTH` and `i2s_slot_mode_t` `I2S_SLOT_MODE_STEREO` because I had problems with rate and mono/1-channel...

Furthermore if you have any chance of build/install for testing latest firmware you can try/build FR https://github.com/espressif/esp-idf/commit/56349e6d656e532a44ae9f8f64af0c90c1967757 (feat(i2s): support to select PDM data format Closes IDF-10511)  - if you get really PDM raw data only then it should be visible/recognizable when forcing to PDM raw data.

@xenpac: There is no internal/build-in function to volume-up. But I learned during my porting approaches from micropython driver that shift each of the PCM samples left = +6db and right = -6db -> for my SPM1423 I need a leftshift,3 so +18db - here is the link to the implementation in the micropython driver: https://github.com/micropython/micropython/blob/39538e4c9a54e44ea27bcc374e17774ac2b9796d/extmod/machine_i2s.c#L441-L501
of course it is a good idea to apply any filters e.g. noise reduction beforehand ;-)
and apply the shifting according to your 16/32bit - if interested there is the sample with +18db attached :point_down: (and yes I have a noise problem... )

@lin-k23: my recorded test MONO, right 16khz pcm does not look like yours - so it looks ok for me (although I personally have no understanding about the audacity linear (amp) scala):
[image]
As you can see from the speactrum I recorded a demo 1500hz -> 150hz signal with button playback click at start of the recording
[mono,right,4x16khz,64os-10s_1500_to_150hz.zip](https://github.com/user-attachments/files/18201548/mono.right.4x16khz.64os-10s_1500_to_150hz.zip) (recorded source: https://youtu.be/5Dc-5DD8P-0?t=301) - you can open that file with your audacity version and check if the scala is 0-based. If not I suggest to update your audacity to latest version 3.7.1

L-KAYA, 2024-12-22:
Hi @DDDanny, thank you so much for looking into these issues and did such detailed reply! You're totally correct, appreciate your great work! 

As it has been a long time since this issue posted here. I'd like to share some updates about PDM RX as well!

1. The raw PDM data format has been supported. Just as @DDDanny said, it was supported in the commit [56349e6d](https://github.com/espressif/esp-idf/commit/56349e6d656e532a44ae9f8f64af0c90c1967757), and you can refer to the latest [I2S programming guide](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/i2s.html#pdm-rx-mode-in-raw-pdm-format) for the difference and usage. Generally, some chips that can not use PDM RX mode (like C3) are now able to use raw PDM mode.
2. Some new chips (P4\C5\C61) now support to set the gain for the PDM2PCM converter in hardware. The converter actually just multified the number you set in [i2s_pdm_rx_slot_config_t::amplify_num](https://github.com/espressif/esp-idf/blob/master/components/esp_driver_i2s/include/driver/i2s_pdm.h#L125).
3. And about the DC offset in the PDM mode, it might be a circuit issue, like some pins of MIC are floating or mis-connected to the GND. If you suspect it's the offset issue should on ESP side, samely, you can try to input a 50% duty
 to the PDM data input pin, and check if the average value is nearly zero. If so, it means PDM RX works correctly on ESP side.

And finally, IDF v4.4 has been end of life. Please migrate to newer version no matter you're using IDF or Arduino repo.  This issue will be closed if no more updates, thanks you all!

closed 2024-12-23 by Alvin1Zhang

xenpac, 2025-05-25:
just adding a comment:
im back on my PDM mic project and just reading this......

Sooo...now we have the option to:

- remove DC offset which includes noise removal. (see above)
  (computes an exponential moving average for the mic signal, and subtracts the average from the   actual signal to correct for it)

- adjusting the mics gain by providing a factor being multiplied to the values.

How nice;)

The DC offset is actually caused by the PDM mic itself (not always mentioned in the datasheets but sometimes).
Its just like that.