espressif/esp-idf #11106: Packet loss on IP101 after IDF v5.x

HeFeng1947, 2023-04-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

After update IDF version to 5.0, some customers have started to give feedback on packet loss on the Ethernet port.

**We did tests with three different versions of IDF and proved that there is indeed a problem in v5.0 version of IDF.**
# ETH TEST on GL-S10

Test step：

a. Make sure your PC and S10 are on the same LAN.

b. Checkout to different IDF version.

c. Program the board with the basic ethernet example. Note the relevant parameter settings.

d. Once it is running and got an IP-address, try to ping the IP with your PC.

## IDF release/v4.2

commit: c327a0016eaea7b3fd54fb00a9d88211b70b4643

firmware: [[basic](https://github.com/espressif/esp-idf/issues/%5Besp-idf/examples/ethernet/basic%20at%20release/v4.2%20%C2%B7%20espressif/esp-idf%20(github.com)%5D(https://github.com/espressif/esp-idf/tree/release/v4.2/examples/ethernet/basic))]([esp-idf/examples/ethernet/basic at release/v4.2 · espressif/esp-idf (github.com)](https://github.com/espressif/esp-idf/tree/release/v4.2/examples/ethernet/basic))

### S10 V1.0

PHY: LAN8720

No external clock crystal

#### Output RMII clock from internal

| description     | configuration |
| --------------- | ------------- |
| SMI_MDC_GPIO    | 23            |
| SMI_MDIO_GPIO   | 18            |
| PHY_Address     | 1             |
| RMII_Clock_GPIO | 17            |

- **ping test result** 

Packet size: 1024

Packet: Sent=100, Received=100, Lost=0 (0% Lost)

Estimated Round Trip Time (in milliseconds): Minimum=0ms, Maximum=1ms, Average=0ms

### S10 V2.1

PHY: IP101

#### Input RMII clock from external

| description    | configuration |
| -------------- | ------------- |
| SMI_MDC_GPIO   | 23            |
| SMI_MDIO_GPIO  | 18            |
| PHY_Reset_GPIO | 5             |
| PHY_Address    | 1             |

- **ping test result** 

Packet size: 1024

Packet: Sent=100, Received=100, Lost=0 (0% Lost)

Estimated Round Trip Time (in milliseconds): Minimum=0ms, Maximum=1ms, Average=0ms

#### Output RMII clock from internal

| description     | configuration |
| --------------- | ------------- |
| SMI_MDC_GPIO    | 23            |
| SMI_MDIO_GPIO   | 18            |
| PHY_Reset_GPIO  | 5             |
| PHY_Address     | 1             |
| RMII_Clock_GPIO | 17            |

- **ping test result** 

Packet size: 1024

Packet: Sent=100, Received=76, Lost=24 (24% Lost)

Estimated Round Trip Time (in milliseconds): Minimum=0ms, Maximum=1ms, Average=0ms

## IDF release/v4.4

commit: 3c25ecb92e2ce5cae9a68c251bd079c1245f9aee

firmware: [[basic](https://github.com/espressif/esp-idf/issues/%5Besp-idf/examples/ethernet/basic%20at%20release/v4.2%20%C2%B7%20espressif/esp-idf%20(github.com)%5D(https://github.com/espressif/esp-idf/tree/release/v4.2/examples/ethernet/basic))]([esp-idf/examples/ethernet/basic at release/v4.2 · espressif/esp-idf (github.com)](https://github.com/espressif/esp-idf/tree/release/v4.2/examples/ethernet/basic))

### S10 V1.0

PHY: LAN8720

No external clock crystal

#### Output RMII clock from internal

| description     | configuration |
| --------------- | ------------- |
| SMI_MDC_GPIO    | 23            |
| SMI_MDIO_GPIO   | 18            |
| PHY_Address     | 1             |
| RMII_Clock_GPIO | 17            |

- **ping test result** 

Packet size: 1024

Packet: Sent=100, Received=100, Lost=0 (0% Lost)

Estimated Round Trip Time (in milliseconds): Minimum=0ms, Maximum=1ms, Average=0ms

### S10 V2.1

PHY: IP101

#### Input RMII clock from external

| description    | configuration |
| -------------- | ------------- |
| SMI_MDC_GPIO   | 23            |
| SMI_MDIO_GPIO  | 18            |
| PHY_Reset_GPIO | 5             |
| PHY_Address    | 1             |

- **ping test result** 

Packet size: 1024

Packet: Sent=100, Received=100, Lost=0 (0% Lost)

Estimated Round Trip Time (in milliseconds): Minimum=0ms, Maximum=1ms, Average=0ms

#### Output RMII clock from internal

| description     | configuration |
| --------------- | ------------- |
| SMI_MDC_GPIO    | 23            |
| SMI_MDIO_GPIO   | 18            |
| PHY_Reset_GPIO  | 5             |
| PHY_Address     | 1             |
| RMII_Clock_GPIO | 17            |

- **ping test result** 

Packet size: 1024

Packet: Sent=100, Received=72, Lost=28 (28% Lost)

Estimated Round Trip Time (in milliseconds): Minimum=0ms, Maximum=1ms, Average=0ms

## IDF release/v5.0

commit: da46d216df8845b3053226f91925d26331aaff76

firmware: [[basic](https://github.com/espressif/esp-idf/issues/%5Besp-idf/examples/ethernet/basic%20at%20release/v4.2%20%C2%B7%20espressif/esp-idf%20(github.com)%5D(https://github.com/espressif/esp-idf/tree/release/v4.2/examples/ethernet/basic))]([esp-idf/examples/ethernet/basic at release/v4.2 · espressif/esp-idf (github.com)](https://github.com/espressif/esp-idf/tree/release/v4.2/examples/ethernet/basic))

### S10 V1.0

PHY: LAN8720

No external clock crystal

#### Output RMII clock from internal

| description     | configuration |
| --------------- | ------------- |
| SMI_MDC_GPIO    | 23            |
| SMI_MDIO_GPIO   | 18            |
| PHY_Address     | 1             |
| RMII_Clock_GPIO | 17            |

- **ping test result** 

Packet size: 1024

Packet: Sent=100, Received=100, Lost=0 (0% Lost)

Estimated Round Trip Time (in milliseconds): Minimum=0ms, Maximum=1ms, Average=0ms

### S10 V2.1

PHY: IP101

#### Input RMII clock from external

| description    | configuration |
| -------------- | ------------- |
| SMI_MDC_GPIO   | 23            |
| SMI_MDIO_GPIO  | 18            |
| PHY_Reset_GPIO | 5             |
| PHY_Address    | 1             |

- **ping test result** 

Packet size: 1024

Packet: Sent=100, Received=95, Lost=5 (5% Lost)

Estimated Round Trip Time (in milliseconds): Minimum=0ms, Maximum=1ms, Average=0ms

#### Output RMII clock from internal

| description     | configuration |
| --------------- | ------------- |
| SMI_MDC_GPIO    | 23            |
| SMI_MDIO_GPIO   | 18            |
| PHY_Reset_GPIO  | 5             |
| PHY_Address     | 1             |
| RMII_Clock_GPIO | 17            |

- **ping test result** 

Packet size: 1024

Packet: Sent=100, Received=52, Lost=48 (48% Lost)

Estimated Round Trip Time (in milliseconds): Minimum=0ms, Maximum=2454ms, Average=47ms

## Conclusion

- IDF release/v4.2 and IDF release/v4.4

If use Output RMII clock from internal, it will cause the package loss. And if use the Input RMII clock from external, it work as expect.

- IDF release/v5.0

what ever using Output RMII clock from internal or Input RMII clock from external, it  will cause the package loss.

HeFeng1947, 2023-04-03:
#10540

kostaond, 2023-05-12:
@HeFeng1947 have you been able to trace down the issue based on other similar discussions? We haven't observed any issues with ESP32-Ethernet-Kit which also uses IP101. [This observation](https://github.com/espressif/esp-idf/issues/10540#issuecomment-1412217948) is interesting. It's important to take care of signal integrity of the CLK on GPIO0 which is also used as boot strap pin. For example, connecting a programmer via longer wires may cause the issues too. I've observed such behavior on wESP32 board. When I used `EPS programmer` and connected it to the wESP32 by wires, Ethernet didn't work reliably and I had to buy the official programmer specifically designed for the wESP32.

HeFeng1947, 2023-05-15:
> @HeFeng1947 have you been able to trace down the issue based on other similar discussions? We haven't observed any issues with ESP32-Ethernet-Kit which also uses IP101. However, [this observation](https://github.com/espressif/esp-idf/issues/10540#issuecomment-1412217948) is interesting. It's important to take care of signal integrity of the CLK on GPIO0 which is also used as boot strap pin. For example, connecting a programmer via longer wires may cause the issues too. I've observed such behavior on wESP32 board. When I used `EPS programmer` and connected it to the wESP32 by wires, Ethernet didn't work reliably and I had to buy the official programmer specifically designed for the wESP32.

We have not tested the ESP32-Ethernet-Kit for any problem on the IDF v5.x version. However, it does vary from one IDF version to another on our devices. I think IDF did cause some errors in the 5.x upgrade, but we haven't clarified where. 
We are currently busy with other urgent projects, and when we are free, we will do the relevant tests again on the ESP32-Ethernet-Kit.

HeFeng1947, 2024-06-14:
> @HeFeng1947 do you still observe the issue?

closed 2024-06-14 by HeFeng1947