espressif/esp-idf #11107: How to safely install multiple IDF vesions in Linux which doesn't impact projects running with existing IDF versions ?

aggaddam, 2023-04-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

release/v5.1 

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Unable to use both IDF versions v4.x.x and v5.x.x for different projects

### What is the actual behavior?

Unable to use both IDF versions v4.x.x and v5.x.x for different projects

### Steps to reproduce.

idf.py set-target esp32c3

### Build or installation Logs.

```plain
Detecting the Python interpreter Checking "python" ... Python 3.10.6 "python" has been detected Adding ESP-IDF tools to PATH... Traceback (most recent call last): File "/home/agaddam/Documents/irb_v207_new_fs/targets/simplifi/irb/li_esp32_sdk/esp-idf/tools/idf_tools.py", line 1860, in <module> main(sys.argv[1:]) File "/home/agaddam/Documents/irb_v207_new_fs/targets/simplifi/irb/li_esp32_sdk/esp-idf/tools/idf_tools.py", line 1856, in main action_func(args) File "/home/agaddam/Documents/irb_v207_new_fs/targets/simplifi/irb/li_esp32_sdk/esp-idf/tools/idf_tools.py", line 1177, in action_export tools_info = filter_tools_info(tools_info) File "/home/agaddam/Documents/irb_v207_new_fs/targets/simplifi/irb/li_esp32_sdk/esp-idf/tools/idf_tools.py", line 1119, in filter_tools_info targets = get_user_defined_targets() File "/home/agaddam/Documents/irb_v207_new_fs/targets/simplifi/irb/li_esp32_sdk/esp-idf/tools/idf_tools.py", line 1099, in get_user_defined_targets if env == idf_env_json['idfSelectedId']: KeyError: 'idfSelectedId'
```

dobairoland, 2023-04-03:
With all due respect, you have checked the following two items:

> I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
> I have searched the issue tracker for a similar issue and not found a similar issue.

This issue has been reported several times. You must use the latest bugfix version (or latest version of the release branch) of whatever release you are switching to.

closed 2023-04-03 by dobairoland

aggaddam, 2023-04-14:
@dobairoland 
Let's say I have two projects A and B.
Project A uses ESP-IDF v5.1
Project B uses ESP-IDF v4.4.2

For Project A - I installed tools for IDF v5.1 using ./install.sh  esp32c3
For Project B - I tried to install tools for IDF v4.4.3 using ./install.sh  esp32c3 but I got below error 
```

Detecting the Python interpreter
Checking "python" ...
Python 3.10.6
"python" has been detected
Installing ESP-IDF tools
Current system platform: linux-amd64
Traceback (most recent call last):
  File "/home/agaddam/Documents/irb_v207_new_fs/targets/simplifi/irb/li_esp32_sdk/esp-idf/tools/idf_tools.py", line 1860, in <module>
    main(sys.argv[1:])
  File "/home/agaddam/Documents/irb_v207_new_fs/targets/simplifi/irb/li_esp32_sdk/esp-idf/tools/idf_tools.py", line 1856, in main
    action_func(args)
  File "/home/agaddam/Documents/irb_v207_new_fs/targets/simplifi/irb/li_esp32_sdk/esp-idf/tools/idf_tools.py", line 1423, in action_install
    targets = clean_targets(args.targets)
  File "/home/agaddam/Documents/irb_v207_new_fs/targets/simplifi/irb/li_esp32_sdk/esp-idf/tools/idf_tools.py", line 1083, in clean_targets
    export_targets_to_idf_env_json(targets)
  File "/home/agaddam/Documents/irb_v207_new_fs/targets/simplifi/irb/li_esp32_sdk/esp-idf/tools/idf_tools.py", line 1053, in export_targets_to_idf_env_json
    targets = list(set(targets + get_user_defined_targets()))
  File "/home/agaddam/Documents/irb_v207_new_fs/targets/simplifi/irb/li_esp32_sdk/esp-idf/tools/idf_tools.py", line 1099, in get_user_defined_targets
    if env == idf_env_json['idfSelectedId']:
KeyError: 'idfSelectedId'
```

I just want to understand - is it possible to use the tool chain for esp idf v5.1 and v4.4.2 in Linux (Ubuntu) ? If yes, what is the recommended solution provided by espressif ?

dobairoland, 2023-04-14:
You can remove `~/.espressif/idf-env.json` as a workaround. But I highly recommend to use the latest bugfix release - v4.4.4 at this moment. Otherwise, you might keep encounter issues already fixed in v4.4.3 and v4.4.4.

dobairoland, 2023-04-14:
This is the duplicate issue: https://github.com/espressif/esp-idf/issues/9837

igrr, 2023-04-14:
Hi @aggaddam, The recommended solution is to use v4.4.4 (latest bugfix release) instead of v4.4.2. 
v4.4.4 contains the fix for this `KeyError: 'idfSelectedId'` issue and it's installation will complete successfully.

aggaddam, 2023-04-14:
@dobairoland  Thanks for the update