espressif/esp-idf #11110: ESP32S3 USB MSC speed lower than expected

zhongpang, 2023-04-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Chip: ESP32S3
IDF version: 5.0.1
USB Stack: TinyUSB from example folder
We implemented a usb mass storage on ESP32S3, the storage backend is a Nand Flash, everything works fine now except the copy speed of file is lower than expected: around 8KBytes/S on Mac and Windows.
We logged the tud_msc_read10_cb() function, and found Nand Flash read operation only cost less than 1MS/4096 Bytes, but the tud_msc_read10_cb() been called every 5MS, which close to the data copy speed we tested: 8KBytes/S = 4K * 1000/5.
From the document, it says S3 USB 2.0 speed could be 12Mbit/S, which means 1.5MBytes/S.
Is there any way to enhance the copy speed from 8KBytes/S to close to 1.5MBytes/S? Thanks very much!

tore-espressif, 2023-04-17:
Hello @zhongpang , Thanks for the bug report and letting us know about this. I'll need some more information in order to help you debug this

Could you please share what Flash chip you used, how is it connected to ESP32s3 and how do you access it from the user code?

Hinalikp, 2024-02-12:
Any update? I am also getting slow speed(0.155 MBytes/s) instead of 1.5MBytes/S. I used this example(esp-idf/examples/peripherals/usb/device/tusb_msc/) on esp32s3. 
IDF version: 5.1.2

kraftwerk-zone, 2024-03-18:
Another raised hand for improved MSC speed. Max I can get is 120kB/sec

FloHomi, 2025-01-10:
I have the same Issue, with the ESP32-S3-DevkitC-1 and the Example tusb_msc code from V5.4. Using internal Flash.
Write speed 88KB/s (thats what windows says) with a 1953KB Bin file. In Reality It takes 4:50 minutes and only writes in ~20% increments with looooong stops in between. The Writing part could be 88KB/s.
From the ESP to the PC the same file takes ~3Sec...

FloHomi, 2025-01-31:
Hi there :)
you you have any idea how long it could take to fix this bug?
I urgently need it for a product deployment and do not know how to fix it myself... :(

tore-espressif, 2025-01-31:
Hi everyone,
 
We’ve been investigating the slow write issues, and here are our initial findings:
 
When implementing a Mass Storage Class (MSC) device on ESP32 chips, using the internal flash as the storage medium is strongly discouraged. Our example uses internal flash only to demonstrate the functionality without requiring extra components.
 
The core limitation is that the program memory and storage reside on the same flash chip. During a flash write operation, program execution from flash is suspended, and interrupts that are not loaded into IRAM cannot run. As a result, USB transfers are severely slowed down.
 
Our recommended approach is to use a dedicated storage medium (e.g., an external SD card or flash) for the MSC disk. This setup can achieve write speeds of around 1 MB/s.
 
We will update both the example and our official documentation to better reflect this recommendation. If you have any further questions or feedback, please let us know.

FloHomi, 2025-02-01:
Thank you very much for your investigation!
I have an AtomS3 with display where i need to load data to it via msc and then do a ota update of other devices. 
I sadly have therefore no other options, than using the internal flash. Can you suggest me maybe a solution on how to improve the situation? If 2MB is transfered in at least 10sec that would be fine with me too...
Kind regards and best wishes!

tore-espressif, 2025-04-17:
@FloHomi unfortunately, we have not found an easy way to speed this up for internal flash, due to the reasons mentioned above.
We updated the docs so it is more obvious for users that the program flash should not be used for storage

For SD cards, we improved the transfer speed + ~25%.

I'll close this issue, but feel free to reopen if you have any more questions

closed 2025-04-17 by tore-espressif