espressif/esp-idf #11114: tlsf_free issue , tlsf_free heap_tlsf.c:872 (!block_is_free(block) && "block already marked as free")

Ramakrishna247, 2023-04-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

IDF 5.0

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32S3 N16R8 

### Power Supply used.

External 3.3V

### What is the expected behavior?

When esp_http_client_cleanup is called , it should clear all resources and close https connection , but it is causing issue . below is the  message showing on the terminal .
tlsf_free heap_tlsf.c:872 (!block_is_free(block) && "block already marked as free")

I have already gone through https://github.com/espressif/esp-idf/issues/8346 , i have made changes as per issue# 8346 but still facing the issue 

### What is the actual behavior?

When esp_http_client_cleanup is called  in code  below issue is happening.

assert failed: tlsf_free heap_tlsf.c:872 (!block_is_free(block) && "block already marked as free")

Backtrace:0x40379ffe:0x3fcf49900x403821cd:0x3fcf49b0 0x40388c0e:0x3fcf49d0 0x40387c06:0x3fcf4af0 0x403882da:0x3fcf4b10 0x4037a215:0x3fcf4b30 0x40388c5d:0x3fcf4b50 0x42042cf9:0x3fcf4b70 0x4204113e:0x3fcf4b90 0x420443ff:0x3fcf4bb0 0x4200d273:0x3fcf4bd0 0x42009f96:0x3fcf4e10 0x42008ad3:0x3fcf4e40 0x420a53a9:0x3fcf4ec0
0x40379ffe: panic_abort at C:/esp/Espressif/frameworks/esp-idf-v4.4/components/esp_system/panic.c:402

0x403821cd: esp_system_abort at C:/esp/Espressif/frameworks/esp-idf-v4.4/components/esp_system/esp_system.c:121

0x40388c0e: __assert_func at C:/esp/Espressif/frameworks/esp-idf-v4.4/components/newlib/assert.c:85

0x40387c06: tlsf_free at C:/esp/Espressif/frameworks/esp-idf-v4.4/components/heap/heap_tlsf.c:872 (discriminator 1)

0x403882da: multi_heap_free_impl at C:/esp/Espressif/frameworks/esp-idf-v4.4/components/heap/multi_heap.c:222
 (inlined by) multi_heap_free_impl at C:/esp/Espressif/frameworks/esp-idf-v4.4/components/heap/multi_heap.c:210

0x4037a215: heap_caps_free at C:/esp/Espressif/frameworks/esp-idf-v4.4/components/heap/heap_caps.c:305

0x40388c5d: free at C:/esp/Espressif/frameworks/esp-idf-v4.4/components/newlib/heap.c:39

0x42042cf9: esp_transport_esp_tls_destroy at C:/esp/Espressif/frameworks/esp-idf-v4.4/components/tcp_transport/transport_ssl.c:458

0x4204113e: esp_transport_destroy_foundation_transport at C:/esp/Espressif/frameworks/esp-idf-v4.4/components/tcp_transport/transport.c:65
 (inlined by) esp_transport_list_destroy at C:/esp/Espressif/frameworks/esp-idf-v4.4/components/tcp_transport/transport.c:124

0x420443ff: esp_http_client_cleanup at C:/esp/Espressif/frameworks/esp-idf-v4.4/components/esp_http_client/esp_http_client.c:735

0x4200d273: get_stm_image_from_cloud at d:\git_codes\espressif_idf_s3\build/../main/src/http_client_support.c:777 (discriminator 13)

0x42009f96: perform_update_stm at d:\git_codes\espressif_idf_s3\build/../main/src/ota_common.c:120

0x42008ad3: app_main at d:\git_codes\espressif_idf_s3\build/../main/src/main.c:115

0x420a53a9: main_task at C:/esp/Espressif/frameworks/esp-idf-v4.4/components/freertos/port/port_common.c:129

### Steps to reproduce.

1, I'am trying to download a binary for some other controller which is on same custom board which has ESP32 S3 , So ESP32 S3 will download the binary  in the chunks. 
2. After completely  downloading  this binary , iam calling  esp_http_client_cleanup, which is causing above issue.

### Debug Logs.

```plain
Stm bin downloaded [ 66625 / 68416]
I (82478) http_client_support.c: HTTP_EVENT_DISCONNECTED
Stm bin downloaded [ 67650 / 68416]
I (83598) http_client_support.c: HTTP_EVENT_DISCONNECTED
Stm bin downloaded [ 68416 / 68416]
I (83948) http_client_support.c: total_read_len = 68416 to file /spiffs/0.0.20.sfb
I (83948) http_client_support.c: HTTP_EVENT_DISCONNECTED

assert failed: tlsf_free heap_tlsf.c:872 (!block_is_free(block) && "block already marked as free")

Backtrace:0x40379ffe:0x3fcf49900x403821cd:0x3fcf49b0 0x40388c0e:0x3fcf49d0 0x40387c06:0x3fcf4af0 0x403882da:0x3fcf4b10 0x4037a215:0x3fcf4b30 0x40388c5d:0x3fcf4b50 0x42042cf9:0x3fcf4b70 0x4204113e:0x3fcf4b90 0x420443ff:0x3fcf4bb0 0x4200d273:0x3fcf4bd0 0x42009f96:0x3fcf4e10 0x42008ad3:0x3fcf4e40 0x420a53a9:0x3fcf4ec0
0x40379ffe: panic_abort at C:/esp/Espressif/frameworks/esp-idf-v4.4/components/esp_system/panic.c:402

0x403821cd: esp_system_abort at C:/esp/Espressif/frameworks/esp-idf-v4.4/components/esp_system/esp_system.c:121

0x40388c0e: __assert_func at C:/esp/Espressif/frameworks/esp-idf-v4.4/components/newlib/assert.c:85

0x40387c06: tlsf_free at C:/esp/Espressif/frameworks/esp-idf-v4.4/components/heap/heap_tlsf.c:872 (discriminator 1)

0x403882da: multi_heap_free_impl at C:/esp/Espressif/frameworks/esp-idf-v4.4/components/heap/multi_heap.c:222
 (inlined by) multi_heap_free_impl at C:/esp/Espressif/frameworks/esp-idf-v4.4/components/heap/multi_heap.c:210

0x4037a215: heap_caps_free at C:/esp/Espressif/frameworks/esp-idf-v4.4/components/heap/heap_caps.c:305

0x40388c5d: free at C:/esp/Espressif/frameworks/esp-idf-v4.4/components/newlib/heap.c:39

0x42042cf9: esp_transport_esp_tls_destroy at C:/esp/Espressif/frameworks/esp-idf-v4.4/components/tcp_transport/transport_ssl.c:458

0x4204113e: esp_transport_destroy_foundation_transport at C:/esp/Espressif/frameworks/esp-idf-v4.4/components/tcp_transport/transport.c:65
 (inlined by) esp_transport_list_destroy at C:/esp/Espressif/frameworks/esp-idf-v4.4/components/tcp_transport/transport.c:124

0x420443ff: esp_http_client_cleanup at C:/esp/Espressif/frameworks/esp-idf-v4.4/components/esp_http_client/esp_http_client.c:735

0x4200d273: get_stm_image_from_cloud at d:\git_codes\espressif_idf_s3\build/../main/src/http_client_support.c:777 (discriminator 13)

0x42009f96: perform_update_stm at d:\git_codes\espressif_idf_s3\build/../main/src/ota_common.c:120

0x42008ad3: app_main at d:\git_codes\espressif_idf_s3\build/../main/src/main.c:115

0x420a53a9: main_task at C:/esp/Espressif/frameworks/esp-idf-v4.4/components/freertos/port/port_common.c:129
```

### More Information.

I have already gone through https://github.com/espressif/esp-idf/issues/8346 , i have made changes as per issue# 8346 but still facing the issue

KonssnoK, 2023-04-03:
likely duplicate of 
https://github.com/espressif/esp-idf/issues/11006

Ramakrishna247, 2023-04-04:
Hi Thanks ,
I will test this [wifi_lib_0404.zip](https://github.com/espressif/esp-idf/files/11144493/wifi_lib_0404.zip) .
But our issue is as soon as http event disconnected , if we are calling free to deallocate any memory or if we calling esp_http_client_cleanup it is crashing the ESP32 S3. I don't think issue https://github.com/espressif/esp-idf/issues/11006 and this issue are same it is more likely  some issue to deallocate memory in heap section.

hmalpani, 2023-04-05:
Hello @Ramakrishna247 
Thanks for reporting the issue. As of now, I am unable to reproduce the issue. Can you provide a minimal application that can be used to reproduce the issue? That will help debug the issue. Thanks!

hmalpani, 2023-04-05:
Hello @Ramakrishna247 
sdkconfig file might be helpful as well. You can share that with the application to reproduce the issue.

Ramakrishna247, 2023-04-06:
Ok,
will share the code  to reproduce issue and sdk file

Ramakrishna247, 2023-04-06:
Hi due to some safety restrictions  i was not able to share example code immediately  , but i'am sharing part of the code where iam facing issue ,  may be you can look in to it once.  

This code will download the .bin file and write it in to spiff part.  this part of code is working fine , once download is complete  when esp_http_client_cleanup(client); is called this issue is happening. 

Also  I have one doubt ,  Iam using dynamically memory allocated buffers for "esp_http_client_config_t config" members  .url, .user_data, do i manually need to free it or esp_http_client_cleanup(client); will free it .

Also one more thing that i have observed when ever  .user_data   as just a char array iam not facing any issues , but if i allocate memory dynamically for it iam facing this issues. 

[code.zip](https://github.com/espressif/esp-idf/files/11169994/code.zip)

hmalpani, 2023-04-10:
Hello @Ramakrishna247 
I tried with the example code you sent. It required a few modifications for compilation. But I was unable to reproduce the issue. The example was executed successfully on my end. It would be helpful if you could share a small code to reproduce the issue. Also, I tried the example with dynamically allocated memory for buffers and didn't get the crash so I don't think that using dynamic buffers can be a problem here.

Ramakrishna247, 2023-04-11:
Hi Hmalpani,
Thank you very much for trying out the example , We will try to send you the example code which can reproduce the issue, also can you share the example code you have tried , may be it might help us resolve issue on our end.

Ramakrishna247, 2023-04-14:
Hi Hmalpani,
Can you please tell me what all changes that you made on your end.

hmalpani, 2023-04-14:
Hello @Ramakrishna247 
I just removed the code which was responsible for writing the data in SPIFFS partition. I didn't change anything else in the code you sent. Also the URL I used was "http://httpbin.org/get"

Ramakrishna247, 2023-04-19:
Thanks for the reply, allow me sometime please, I will try to give you a sample code to re create the issue.

Ramakrishna247, 2023-04-21:
Hi Sorry to bother you , when we were trying to re-create issue with simple example , we realized that we haven't allocated enough memory to a  variable using malloc  in function just which was called before the OTA  function.  which was the cause of the issue , any how ESP32 S3 back trace was pointing to esp_http_client_cleanup  function, which created confusion we were focusing on this function but actually issue was in another function.  
 Thanks for the help.

closed 2023-04-21 by Ramakrishna247