espressif/esp-idf #11119: Custom brownout routine (Brownout ISR)

diablo1281, 2023-04-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am looking for an option to replace the brownout default ISR with my own implementation. I found solutions with IDF v4.x.x but none with v5.0.0+. Seems like a good amount of features were removed or hidden in this version. Can't find anything in the documentation also Can you give me any hints or directions for it?

mythbuster5, 2023-04-04:
Brownout is now only used internally. BTW, can you give some comparison on 4.x.x with 5.0? I don't remember any feature is removed or hidden on v5.0.

mythbuster5, 2025-07-07:
Thanks, we provide a callback naming `esp_brownout_register_callback`