espressif/esp-idf #11120: How to probe gpio pin can be accessed

waruqi, 2023-04-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

If some pins are inaccessible, continuing to read and write gpio will report LoadProhibited and cause a crash. Is there an interface to detect that a given pin is accessible and I can then avoid reading or writing to it in my program?

suda-morris, 2023-04-04:
> continuing to read and write gpio will report LoadProhibited and cause a crash

I'm afraid you're referring to some IO used by SPI flash. Please read the table in https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/gpio.html#gpio-summary for those special GPIOs.

In the code, there's a WIP API in [esp_gpio_reserve.h](https://github.com/espressif/esp-idf/blob/master/components/esp_hw_support/include/esp_private/esp_gpio_reserve.h), which can tell you if a GPIO is reserved by a special purpose.

waruqi, 2023-04-04:
I don't mean that, I'm reading a valid gpio io pin, which should be read correctly if I wire it up. But sometimes, I may unplug the cable that connects it. At this time, I want to be able to detect that the gpio pin is not connected, and not continue to read it. Otherwise the program will trigger a crash.

songruo, 2023-04-04:
I can't think of why LoadProhibited would happen. Can you give us an example, on which ESP chip, and which pin are you continuously trying to read, and through which API? Do you mean the read GPIO pin action itself will cause the crash?

waruqi, 2023-04-04:
I can't provide a complete example, I'm trying to use the pwm interface to read and write 6 servos at the same time.

When I only connect one servo gpio, other gpio can be successfully initialized by mcpwm_init.

When I call mcpwm_set_duty_in_us to write these gpio data. It doesn't crash immediately. But the program will crash afterward.

But if I connect all servo gpio, everything works fine.

suda-morris, 2023-04-06:
For such crash issue, please summary your hardware connection, what GPIO is used for what purpose? The crash log? The minimal project can help us reproduce the issue？

suda-morris, 2023-04-06:
> use the pwm interface to read and write 6 servos

As far as I can tell, the PWM peripheral doesn't have the ability to **read**. What do you mean by **reading 6 servos**?

waruqi, 2023-04-06:
It's my mistake, I misconfigured some gpio pin numbers.

closed 2023-04-06 by waruqi