espressif/esp-idf #11121: Interrupt allocation failure after upgrade from esp-idf v4.4.2 to v5.0.1

FTNMotiondevs, 2023-04-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.1

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESPS2-S3 custom board

### Power Supply used.

External 3.3V

### What is the expected behavior?

The twai driver initialised successfully with ESP-IDF v4.4.2. Code is 
`    twai_general_config_t g_config = TWAI_GENERAL_CONFIG_DEFAULT(txpin_, rxpin_, TWAI_MODE_NORMAL);
    ESP_ERROR_CHECK_WITHOUT_ABORT(err = twai_driver_install(&g_config, &t_config, &f_config));
`

### What is the actual behavior?

I now get:
`ESP_ERROR_CHECK failed: esp_err_t 0x105 (ESP_ERR_NOT_FOUND) at 0x42026b5a
file: "/IDF/components/driver/twai.c" line 449
func: twai_driver_install
expression: esp_intr_alloc(ETS_TWAI_INTR_SOURCE, g_config->intr_flags, twai_intr_handler_main, NULL, &p_twai_obj->isr_handle)
`
Which from my reading suggests interrupt allocation failure. The interrupt assignment prior to this failure is:
`
Rebooting...
I (194) cpu_start: Pro cpu start user code
I (194) cpu_start: cpu freq: 160000000 Hz
I (197) cpu_start: ESP-IDF:          v5.0.1
I (198) cpu_start: Min chip rev:     v0.0
I (198) cpu_start: Max chip rev:     v0.99
I (199) cpu_start: Chip rev:         v0.1
D (219) intr_alloc: Connected src 39 to int 2 (cpu 0)
D (232) intr_alloc: Connected src 42 to int 3 (cpu 0)
D (240) intr_alloc: Connected src 59 to int 9 (cpu 0)
D (244) intr_alloc: Connected src 79 to int 12 (cpu 0)
D (245) intr_alloc: Connected src 57 to int 13 (cpu 0)
D (245) intr_alloc: Connected src 80 to int 2 (cpu 1)
D (245) intr_alloc: Connected src 58 to int 3 (cpu 1)
D (245) intr_alloc: Connected src 52 to int 17 (cpu 0)
`

### Steps to reproduce.

TODO

suda-morris, 2023-04-04:
you can try with the `intr_flags = ESP_INTR_FLAG_LOWMED`, currently it's set with `ESP_INTR_FLAG_LEVEL1` by default.

FTNMotiondevs, 2023-04-04:
@suda-morris thanks, that fixed it!

closed 2023-04-04 by FTNMotiondevs