espressif/esp-idf #11123: BLE notification/Indicate callback (ESP_GATTC_NOTIFY_EVT) not triggered

AbiotsLabs, 2023-04-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

1. I have read the similar issues and documentations and have done following steps: 
      1. ` esp_ble_gattc_register_for_notify(gattc_if, p_data->connect.remote_bda, gl_profile_tab[PROFILE_A_APP_ID].char_handle);`
      2. After this got the response ESP_GATTC_REG_FOR_NOTIFY_EVT, in this wrote the following code
` if (param->reg_for_notify.status == ESP_GATT_OK)
{
            ESP_LOGI(HUB_TAG, "Registered for notifications!");
            esp_gattc_descr_elem_t *descr_elem_result = NULL;
            uint16_t count = 1;
            uint16_t notify_en = 2;
            esp_err_t ret_status;
            descr_elem_result = (esp_gattc_descr_elem_t *)malloc(sizeof(esp_gattc_descr_elem_t) * count);
            if (!descr_elem_result)
            {
                ESP_LOGE(HUB_TAG, "malloc error, gattc no mem");
            }
            else
            {
                ret_status = esp_ble_gattc_get_descr_by_char_handle(
                    gattc_if,
                    gl_profile_tab[PROFILE_A_APP_ID].conn_id,
                    gl_profile_tab[PROFILE_A_APP_ID].char_handle,
                    notify_descr_uuid,
                    descr_elem_result, &count);
                if (ret_status != ESP_GATT_OK)
                {
                    ESP_LOGE(HUB_TAG, "esp_ble_gattc_get_descr_by_char_handle error");
                }
                /* Every char has only one descriptor in our 'ESP_GATTS_DEMO' demo, so we used first 'descr_elem_result' */
                if (count > 0 && descr_elem_result[0].uuid.len == ESP_UUID_LEN_16 && descr_elem_result[0].uuid.uuid.uuid16 == ESP_GATT_UUID_CHAR_CLIENT_CONFIG)
                {
                    ESP_LOGD(HUB_TAG,"Enabling the notifications");
                    ret_status = esp_ble_gattc_write_char_descr(gattc_if,
                                                                gl_profile_tab[PROFILE_A_APP_ID].conn_id,
                                                                descr_elem_result[0].handle,
                                                                sizeof(notify_en),
                                                                (uint8_t *)&notify_en,
                                                                ESP_GATT_WRITE_TYPE_RSP,
                                                                ESP_GATT_AUTH_REQ_NONE);
                }
                if (ret_status != ESP_GATT_OK)
                {
                    ESP_LOGE(HUB_TAG, "esp_ble_gattc_write_char_descr error");
                }
                /* free descr_elem_result */
  free(descr_elem_result);`
3. Now when i send the indication message from gatt server i am not getting any callback, but i can see the BT_GATT getting the notification updates but call back is not happening.
4. I (6481) BT_GATT: GATT_GetConnectionInfor conn_id=3
I (6481) BT_GATT: GATTC_SendHandleValueConfirm conn_id=3 handle=0xd
D (6491) BT_GATT: notif_count=3 
D (6541) BT_GATT: gatt_get_ch_state: ch_state=4
D (6541) BT_GATT: gatt_process_notification 
D (6541) BT_GATT: gatt_get_link_encrypt_status status=0x8e
D (6551) BT_GATT: gatt_cl_op_cmpl_cback
I (6551) BT_GATT: GATT_GetConnectionInfor conn_id=3
I (6561) BT_GATT: GATTC_SendHandleValueConfirm conn_id=3 handle=0xd
D (6561) BT_GATT: notif_count=3 
D (6621) BT_GATT: gatt_get_ch_state: ch_state=4
D (6621) BT_GATT: gatt_process_notification 
D (6621) BT_GATT: gatt_get_link_encrypt_status status=0x8e
D (6621) BT_GATT: gatt_cl_op_cmpl_cback
I (6631) BT_GATT: GATT_GetConnectionInfor conn_id=3
I (6631) BT_GATT: GATTC_SendHandleValueConfirm conn_id=3 handle=0xd
D (6641) BT_GATT: notif_count=3 
D (6691) BT_GATT: gatt_get_ch_state: ch_state=4
D (6691) BT_GATT: gatt_process_notification 
D (6691) BT_GATT: gatt_get_link_encrypt_status status=0x8e
D (6701) BT_GATT: gatt_cl_op_cmpl_cback
I (6701) BT_GATT: GATT_GetConnectionInfor conn_id=3
I (6711) BT_GATT: GATTC_SendHandleValueConfirm conn_id=3 handle=0xd
D (6711) BT_GATT: notif_count=3 
D (6771) BT_GATT: gatt_get_ch_state: ch_s

Alvin1Zhang, 2023-04-06:
Thanks for reporting.

tiibb, 2024-03-24:
Hello,
Same issue for me. Any news ?

zhp0406, 2024-03-26:
@AbiotsLabs 
hi,Have your issues been resolved now?

tiibb, 2024-03-26:
Hi,
It seems I had the same issue as OP.
My mistake was that I assumed p_data->connect.remote_bda was defined when calling esp_ble_gattc_register_for_notify within ESP_GATTC_SEARCH_CMPL_EVT event.

closed 2024-12-18 by Alvin1Zhang