espressif/esp-idf #11125: BLE suddenly stop when Wi-Fi Mesh is connected

yuki-sato, 2023-04-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.3.5

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C3-MINI

### Power Supply used.

USB

### What is the expected behavior?

ESP32 is connected to the internet using Wi-Fi MESH. We use two or more devices.
One is connected to an access point and others are connected to the one.

We transmit and receive BLE HCI from via the internet and mesh network then all device start BLE scanning.

### What is the actual behavior?

Some devices went to stop collecting BLE advertisement suddenly. Never responds to HCI command.
That happen around 3 times in 24 hours.

### Steps to reproduce.

1. All device connect to same Wi-Fi MESH network
2. start ble on devices and send hci command from the internet then device start ble scanning
3. every 30 seconds stop scan and start scan 
4. after hours, some device went to stop responding ble advertisement and ble scan start command.

Alvin1Zhang, 2023-04-06:
Thanks for reporting.

xiewenxiang, 2023-04-07:
Could you please provide the complete HCI log? All of comand and event.

yuki-sato, 2023-04-12:
Sorry for delaying. We are collecting HCI logs.

yuki-sato, 2023-06-12:
Sorry for no response.
We realized normal Wi-Fi situation, not Wi-Fi MESH, could face same BLE issue.
We keep investigating.

zhp0406, 2023-11-17:
@yuki-sato 
Have your issue been resolved?

yuki-sato, 2023-11-18:
@zhp0406 
No. But our compony gave up using Wi-Fi MESH. So investigating and collecting debug data is stopping now.

yuki-sato, 2023-11-18:
I'm going to close this issue.

closed 2023-11-18 by yuki-sato