espressif/esp-idf #11127: Simultaneous operation of USB and i2s.

EfanovIvan, 2023-04-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.1

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32-S3-WROOM-1-N16R8

### Power Supply used.

USB

### What is the expected behavior?

Simultaneous operation of USB and i2s.

### What is the actual behavior?

 I am getting the following error when initializing i2s
```
E (11406) gdma: gdma_install_rx_interrupt(773): alloc interrupt failed
E (11416) gdma: gdma_register_rx_event_callbacks(435): install interrupt service failed
```

Initialization code
```
Microphone()
    {
        i2s_chan_config_t chanCfg = I2S_CHANNEL_DEFAULT_CONFIG(I2S_NUM_AUTO, I2S_ROLE_MASTER);
        ESP_ERROR_CHECK(i2s_new_channel(&chanCfg, nullptr, &rxHandle));

        i2s_pdm_rx_config_t pdmRxCfg = {
            .clk_cfg = I2S_PDM_RX_CLK_DEFAULT_CONFIG(SAMPLE_RATE),
            /* The default mono slot is the left slot (whose 'select pin' of the PDM microphone is pulled down) */
            .slot_cfg = I2S_PDM_RX_SLOT_DEFAULT_CONFIG(I2S_DATA_BIT_WIDTH_16BIT, I2S_SLOT_MODE_MONO),
            .gpio_cfg = {
                .clk = static_cast<gpio_num_t>(CONFIG_I2S_CLK_PIN),
                .din = static_cast<gpio_num_t>(CONFIG_I2S_DATA_PIN),
                .invert_flags = {
                    .clk_inv = false,
                },
            },
        };
        ESP_ERROR_CHECK(i2s_channel_init_pdm_rx_mode(rxHandle, &pdmRxCfg));
        ESP_ERROR_CHECK(i2s_channel_enable(rxHandle));
        m_buff.resize(SAMPLE_SIZE);
    }
```

### Steps to reproduce.

Initialize usb

```
CliHandlersStorage::CliHandlersStorage()
{
    esp_console_repl_config_t replCfg{};
    replCfg.max_history_len = CONFIG_CLI_HISTORY_LEN;
    replCfg.task_stack_size = CONFIG_CLI_TASK_STACK_SIZE;
    replCfg.task_priority   = CONFIG_CLI_TASK_PRIORITY;
    esp_console_repl_t* pRepl{};

    esp_console_dev_usb_serial_jtag_config_t usbjtag_config =
        ESP_CONSOLE_DEV_USB_SERIAL_JTAG_CONFIG_DEFAULT();
    if (esp_console_new_repl_usb_serial_jtag(&usbjtag_config, &replCfg, &pRepl))
    {
        throw std::runtime_error{ "Failed to create a REPL environment" };
    }

    if (esp_console_start_repl(pRepl) != ESP_OK)
        throw std::runtime_error{ "Failed to start a REPL environment" };
}

```
After that i2s, the code is shown above

EfanovIvan, 2023-04-04:
Thanks for the quick response!
I updated to the last commit but unfortunately the fix did not help
`I (977) cpu_start: ESP-IDF:          v5.0.1-464-gef4b1b7704-dirty`
I also looked at the pullquest and did not find the function
` esp_err_t function there i2s_channel_init_pdm_rx_mode(i2s_chan_handle_t handle, const i2s_pdm_rx_config_t *pdm_rx_cfg)`
This function also sets the interrupt flag
`ESP_GOTO_ON_ERROR(i2s_init_dma_intr(handle, ESP_INTR_FLAG_LEVEL1), err, TAG, "initialize dma interrupt failed");`
I tried changing `I2S_INTR_ALLOC_FLAGS` to this value but it didn't help

L-KAYA, 2023-04-06:
Hi @EfanovIvan, I can't reproduce the issue only by initializing I2S and USB console, not sure if I missed anything, here is my test codes, could you give more details to reproduce this issue?
```c
#include <stdio.h>
#include "esp_check.h"
#include "driver/i2s_pdm.h"
#include "driver/usb_serial_jtag.h"
#include "esp_console.h"

void init_i2s()
{
    i2s_chan_handle_t rxHandle = NULL;
    i2s_chan_config_t chanCfg = I2S_CHANNEL_DEFAULT_CONFIG(I2S_NUM_AUTO, I2S_ROLE_MASTER);
    ESP_ERROR_CHECK(i2s_new_channel(&chanCfg, NULL, &rxHandle));

    i2s_pdm_rx_config_t pdmRxCfg = {
        .clk_cfg = I2S_PDM_RX_CLK_DEFAULT_CONFIG(16000),
        /* The default mono slot is the left slot (whose 'select pin' of the PDM microphone is pulled down) */
        .slot_cfg = I2S_PDM_RX_SLOT_DEFAULT_CONFIG(I2S_DATA_BIT_WIDTH_16BIT, I2S_SLOT_MODE_MONO),
        .gpio_cfg = {
            .clk = 1,
            .din = 2,
            .invert_flags = {
                .clk_inv = false,
            },
        },
    };
    ESP_ERROR_CHECK(i2s_channel_init_pdm_rx_mode(rxHandle, &pdmRxCfg));
    ESP_ERROR_CHECK(i2s_channel_enable(rxHandle));
}

void init_usb_serial_jtag()
{
    esp_console_repl_t* pRepl = NULL;
    esp_console_repl_config_t replCfg = ESP_CONSOLE_REPL_CONFIG_DEFAULT();
    esp_console_dev_usb_serial_jtag_config_t usbjtag_config = ESP_CONSOLE_DEV_USB_SERIAL_JTAG_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_console_new_repl_usb_serial_jtag(&usbjtag_config, &replCfg, &pRepl));
    ESP_ERROR_CHECK(esp_console_start_repl(pRepl));
}

void app_main(void)
{
    init_usb_serial_jtag();
    init_i2s();
}

```

EfanovIvan, 2023-04-06:
Hello @L-KAYA I will build a test project and upload it.

L-KAYA, 2025-01-13:
Hi @EfanovIvan @ctag-fh-kiel, does the issue still exist? And could you upload the test project to reproduce?

L-KAYA, 2025-02-24:
There is a new example [i2s_usb](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/i2s/i2s_advance/i2s_usb) available on the master, it doesn't have this error, it is possibly not the coexistence issue of USB and I2S. 

Going to close this issue since no further response, feel free to reopen it or open a new issue about the GDMA.

closed 2025-02-25 by Alvin1Zhang