espressif/esp-idf #11129: ESP_LCD Low Performance (SPI LCD)

windorey, 2023-04-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello. Trying out the [SPI LCD example](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/lcd/spi_lcd_touch) with LVGL I noticed that even if I follow the performance guide [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/performance/speed.html), I still get very low frame rates.

SPI frequency is set to the maximum, I am using ILI9341 display, but the same happens with ST7789V.

I have also benchmarked the performance using LVGL's own "music demo" and a slightly modified version of the example, which resulted in a disappointing 17 FPS.
Even a Arduino driver beats this performance. The Arduino driver gets 24 FPS, and is noticably much smoother.

suda-morris, 2023-04-06:
@windorey Recently we're drafting guidance about LVGL performance in https://github.com/espressif/esp-bsp/pull/146 , could you please take the opportunity to take a look and give your feedback there?

windorey, 2023-04-06:
@suda-morris Okay, I had a look at this. The performance is better, but when trying to increase my SPI frequency from 40MHz to 80MHz my display suddenly stopped working. Just a black screen.

I know my display works on 80MHz without problem with different drivers, why isn't it working now?

suda-morris, 2023-04-07:
@windorey Then you may need to read the programming guide about the SPI Master driver, especially this section: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/spi_master.html#gpio-matrix-and-io-mux

And for good signal quality, to achieve an 80MHz clock, I'm afraid you need to take more care in your PCB layout.

closed 2023-04-08 by windorey