espressif/esp-idf #11130: Unexpected behaviour of the provisioning example on ESP-IDF V5.0.1

idea--list, 2023-04-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

Release V5.0.1

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

AdafruitESP32S3 Feather

### Power Supply used.

USB

### What is the expected behavior?

I am expecting the bundled provisioning example work without any quirks.

### What is the actual behavior?

Since upgrading from IDF V4.4.3 to 5.0.1 i am experimenting with the provisioning example. Opted in to reset provisionong state on each system start. At the beginning could successfully provision my board using the ESP BLE Prov app on my Android phone.

After that i tried to modify a copy of the example in a way that i dynamically generate a pop instead of hardcoding a string in FW. Also that seemed to work several times.

However for some reason it began to behave flaky. By that i mean the mobile application fails at step 3/3 of the process and displays authentication failure. On the console i get a WiFi disconnect reason 2 and immediately after that reason 205. After ca. 5 seconds my ESP still succeeds to connect and also an IP address gets assigned.

I thought i messed up the code, so reverted back to the original version (double checked, no difference to the bundled version), also called `idf.py erase-flash`, `idf.py fullclean` and then rebuilt&flashed the board. Despite that i still observe the same symptoms: app tells the provisioning was not successful, get disconnect reason 2 then 205 on the console, but after that the board still connects and gets an IP.

Have no idea if the router might be throttling something or what else might be the cause. Right now it just puzzles me why it behaves that way.

### Steps to reproduce.

1. Provision an ESP32S3 using the bundled provisioning example (BLE transfer) multiple times in a short time.
2. If it is not a router specific behaviour but a possible bug, then the tester should face same symptoms 

However right now i am not sure if this is a bug in IDF or my router does not work as it should.

[image]

idea--list, 2023-04-05:
Update 1:
Just changed the board to an ESP32 V3, ran `idf.py set-target esp32` then rebuilt&flashed. The whole process works with this board just as it should. The AP does not disconnect and also the Android app reports successful provisioning.

idea--list, 2023-04-05:
Update 2:
Almost 24 hours later i tried again the same ESP32S3 board. Did not change anything in the code and also the whole scenario is the same. Interestingly i could now provision my ESP32S3 board just as expected, however 1 time only.

5 minutes later i resetted the board to reset its provisioning state. On my phone i closed and restarted the provisioning app... and i get face the very same symptoms i described yesterday.

I do not know about anything in my router's settings that might be causing this, but i am not an expert in router settings. For me all this is unexpected.

idea--list, 2023-04-05:
Update 3: Tried with a second Adafruit ESP32S3 Feather board. When running the example for the first time, it works as expected. Then after restarting the example (opted in to reset provisioning state on restart) the second time it fails also on the second board. So it is not just some HW issue of that particular board i was trying first. The ESP32 V3 still works fine.

I know my description is not as specific as it should be, but have no idea what might be causing all this.

idea--list, 2023-04-05:
Update 4:
Removed IDF V5.0.1, restarted my computer, and installed IDF V4.4.4 this time.
It seems that the code of the provisioning example is somewhat different (no option for security 2, etc).
I also find that the QR code in the console does not look like it should. When trying to provision an ESP32S3 board i still get the same reports on the console (disconnect reason 2 then 205), but the ESP BLE Prov application tells me the provisioning was successful (failed only 1 out of 10 trials). Unlike when trying the example bundled to IDF V5.0.1 where it fails 10 out 10 times.

idea--list, 2023-04-05:
Update 5:
Reverted back to IDF V4.4.3 The provisioning example bundled to that version also connects fine on the first trial. In that case no disconnection reasons are showed on the console. On consecutive trials the console shows reason 2 then 205 but the board still can get an IP address and the ESP BLE Prov app succeeds fine all the time.

So to sum it all up: Using an ESP32S3 only the first provisioning runs without any glitches. Successive trials will yield disconnection reasons 2 and 205 reported on the console but in case of IDF V4.4.3 and 4.4.4 the phone application succeeds wihout reporting an error and the ESP32S3 gets an IP assigned. When trying the same with V5.0.1 the phone application reports an error at the very last step, despite the board still gets an IP number assigned. However i am sure any enduser of a product would be confused by this.

In contrast my ESP32 V3 board seems to work just as expected on all three IDF versions.

Obviously the ESP32S3 needs a long interval between 2 consecutive provisioning events to work as expected, otherwise i can repeatedly face the issue on several ESP32S3 boards. Will try to experiment how long should one wait in case of the ESP32S3 until the feature works fine again.

idea--list, 2023-04-05:
Update 6:
It seems that waiting 20 minutes between trying to re-provision an ESP32S3 is enough to have the whole process running just as expected. No disconnection reasons reported on console or error messages within the phone application.

Tried to find anything relevant inside the documentation of ESP-IDF, the provisioning example but did not notice anything relevant. In menuconfig i found an option called `BT_NIMBLE_RPA_TIMEOUT` which is set 900 by default, however have no clue if that might be relevant.

idea--list, 2023-05-18:
@laukik-hase 
Just commented out line case WIFI_REASON_AUTH_EXPIRE: inside components/wifi_provisioning/src/manager.c of my 5.0.1 installation. Compiled the bundled example again. This time the mobile application does not fail at step 3/3.
However on the console i still get the same error reasons 2 and 205.

This way end users will not be confused any more by the phone application but the underlying cause seems to persist.

(Beyond the scope of this ticket: meanwhile i face my projects using wifi to behave flaky on the ESP32S3 when tested for several hours. By that i mean the FW runs fine at the beginning but tends to fail when run for several hours. Have no idea why as there is nothing on the console and until now i could not find any grasp that would help me narrowing down where it might come from.)

mhdong, 2023-11-01:
Hi @laukik-hase Fix on master branch is available at https://github.com/espressif/esp-idf/commit/d0a30c1e7b7e4f9b9925bd0e1d026831bc4987eb, we are also back porting to other release branches.
Will update further if any progress. Thanks.
cc @laukik-hase

mhdong, 2023-12-19:
Hi @idea--list Fix on v5.0 is available at https://github.com/espressif/esp-idf/commit/b2a4cf197e7186fc3751cd0e2a528c617b5cc350, please confirm.

closed 2023-12-19 by Alvin1Zhang