espressif/esp-idf #11131: esp mbedTLS crash when trying to cleanup after a server connection fails due to loss of network

hairyhoncho, 2023-04-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

4.4.4

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32 WROVER IE

### Power Supply used.

USB

### What is the expected behavior?

When the underlying network  connection fails and esp_tls_connection_write() returns an error, a call to esp_tls_conn_destroy() is expected to close the connection and release resources

### What is the actual behavior?

The call to esp_tls_conn_destroy() results in a crash.

### Steps to reproduce.

1.Setup a network connection, eg WIFI_STA to an access point
2.open an mbedtls connection to a server
3.Write strings to the server over the connection
4.Turn off the AP
5. When write returns an error call esp_tls_conn_destroy
6. The node will crash.

### Debug Logs.

```plain
E (31830) esp-tls-mbedtls: write error :-0x004E:
Failed to write data to server (-78)
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4008d7d6  PS      : 0x00060933  A0      : 0x8008de85  A1      : 0x3ffba020  
0x4008d7d6: insert_free_block at /home/nuc/esp/esp-idf/components/heap/heap_tlsf.c:237
 (inlined by) block_insert at /home/nuc/esp/esp-idf/components/heap/heap_tlsf.c:263
 (inlined by) tlsf_free at /home/nuc/esp/esp-idf/components/heap/heap_tlsf.c:968

A2      : 0x3ffb7b94  A3      : 0xa5a5a5a5  A4      : 0x3ffb9f54  A5      : 0x3ffaeb58  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x3ffc5e00  A9      : 0x00000018  
A10     : 0x0000000f  A11     : 0x3ffc5e08  A12     : 0x3ffb7bec  A13     : 0x00000004  
A14     : 0x00000026  A15     : 0x0000000f  SAR     : 0x0000001c  EXCCAUSE: 0x0000001d  
EXCVADDR: 0xa5a5a5b1  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace: 0x4008d7d3:0x3ffba020 0x4008de82:0x3ffba040 0x40081d9a:0x3ffba060 0x4008f641:0x3ffba080 0x400f800a:0x3ffba0a0 0x400d7074:0x3ffba0c0 0x400d6edb:0x3ffba0e0 0x40156578:0x3ffba140 0x4008b911:0x3ffba160
0x4008d7d3: insert_free_block at /home/nuc/esp/esp-idf/components/heap/heap_tlsf.c:236
 (inlined by) block_insert at /home/nuc/esp/esp-idf/components/heap/heap_tlsf.c:263
 (inlined by) tlsf_free at /home/nuc/esp/esp-idf/components/heap/heap_tlsf.c:968

0x4008de82: multi_heap_free_impl at /home/nuc/esp/esp-idf/components/heap/multi_heap.c:225

0x40081d9a: heap_caps_free at /home/nuc/esp/esp-idf/components/heap/heap_caps.c:361

0x4008f641: free at /home/nuc/esp/esp-idf/components/newlib/heap.c:39

0x400f800a: esp_tls_conn_destroy at /home/nuc/esp/esp-idf/components/esp-tls/esp_tls.c:110

0x400d7074: syslog_close at /home/nuc/git/syslog/components/syslog_client/syslog_client.c:140

0x400d6edb: app_main at /home/nuc/git/syslog/main/syslog.c:66 (discriminator 13)

0x40156578: main_task at /home/nuc/esp/esp-idf/components/freertos/port/port_common.c:141 (discriminator 2)

0x4008b911: vPortTaskWrapper at /home/nuc/esp/esp-idf/components/freertos/port/xtensa/port.c:142

ELF file SHA256: 771bfb9b071e0787

CPU halted.
```

mahavirj, 2023-04-05:
@hairyhoncho Can you please share your application code and sdkconfig?

hairyhoncho, 2023-04-07:
I can share privately, how do you want to do this ?

mahavirj, 2023-04-07:
@hairyhoncho 

I would recommend that you take any IDF provided protocol example and then share minimal patch on top of it that can recreate the issue. Going through your entire project might not be a feasible option.

Interim, just a hint from your backtrace, if you are doing the TLS operation from main task itself then please ensure that you have increase its stack size (and not using the default value), alternatively you can create an independent task with sufficiently larger stack size (please see IDF protocol examples for this).

Additionally, you can also check out our guide to debug the heap allocator related issues here: https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/system/heap_debug.html

hairyhoncho, 2023-04-07:
Will do.Thank you

AdityaHPatwardhan, 2023-04-17:
Hi @hairyhoncho any update on this?

mahavirj, 2023-04-24:
Closing due to lack of update. Please drop a comment if the suggestions provided above did not help and you are still facing the issue.

closed 2023-04-24 by mahavirj