espressif/esp-idf #11133: esp_timer_get_time hangs

QuentinFarizon, 2023-04-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.1

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32S3-DevkitM1

### Power Supply used.

USB

### What is the expected behavior?

I expect esp_timer_get_time to not hang

### What is the actual behavior?

I'm calling esp_timer_get_time() in my main function, and sometimes it completely hangs, I get no restart, no WDT 

### Steps to reproduce.

Here is my code : 
```
    ESP_LOGI(TAG, "log1");
    uint32_t wifi_ok_time_in_sec = esp_timer_get_time() / 1000000;
     ESP_LOGI(TAG, "log2");
 ```

I reproduce randomly (very few times) to have :
```
I (10:41:55.722) MAIN: log1
```
And then nothing

CC-MASTER05, 2023-04-08:
I got both logs make sure you have install correct library "esp_timer.h" if you have still this issue verify onetime your code so if any 2 function access same time timer and modify operation like reset, pause, resume then it will stuck there and reboot your board

KonstantinKondrashov, 2023-05-19:
@QuentinFarizon 
On my side it works fine, no issue with it. Looks like the power supply issue.

closed 2023-05-19 by KonstantinKondrashov